tbnobody/OpenDTU #2299: Speichern der DTU Einstellungen nicht möglich.

DumpName, 2024-09-23:
### What happened?

Wenn ich die Einstellungen auf der DTU Config Seite speichere (ohne Änderung) erhalte ich "Erfolgreich gespeichert" zurück.
Sobald ich eine Änderung (z.B. ändern der Sendeleistung um 1db) vornehme erhalte ich "Benötigte Werte fehlen".
### Erfolgreiche Request Payload
```
-----------------------------297579537740323138651715974898
Content-Disposition: form-data; name="data"
{
    "cmt_chan_width": 250000,
    "cmt_country": 0,
    "cmt_enabled": true,
    "cmt_frequency": 865000000,
    "cmt_palevel": 0,
    "country_def": [
        {
            "freq_default": 865000000,
            "freq_legal_max": 870000000,
            "freq_legal_min": 863000000,
            "freq_max": 923500000,
            "freq_min": 860250000
        },
        {
            "freq_default": 918000000,
            "freq_legal_max": 925000000,
            "freq_legal_min": 905000000,
            "freq_max": 963500000,
            "freq_min": 900250000
        },
        {
            "freq_default": 918000000,
            "freq_legal_max": 928000000,
            "freq_legal_min": 915000000,
            "freq_max": 963500000,
            "freq_min": 900250000
        }
    ],
    "nrf_enabled": false,
    "nrf_palevel": 0,
    "pollinterval": 5,
    "serial": "199980142212"
}
-----------------------------297579537740323138651715974898--
```

### Fehlgeschlagene Request Payload
```
-----------------------------263990553329237855153048333242
Content-Disposition: form-data; name="data"
{
    "cmt_chan_width": 250000,
    "cmt_country": 0,
    "cmt_enabled": true,
    "cmt_frequency": 865000000,
    "cmt_palevel": "1",
    "country_def": [
        {
            "freq_default": 865000000,
            "freq_legal_max": 870000000,
            "freq_legal_min": 863000000,
            "freq_max": 923500000,
            "freq_min": 860250000
        },
        {
            "freq_default": 918000000,
            "freq_legal_max": 925000000,
            "freq_legal_min": 905000000,
            "freq_max": 963500000,
            "freq_min": 900250000
        },
        {
            "freq_default": 918000000,
            "freq_legal_max": 928000000,
            "freq_legal_min": 915000000,
            "freq_max": 963500000,
            "freq_min": 900250000
        }
    ],
    "nrf_enabled": false,
    "nrf_palevel": 0,
    "pollinterval": 5,
    "serial": "199980142212"
}
-----------------------------263990553329237855153048333242--
```

### To Reproduce Bug

### pre-Req:
- Vorhandener ESP32 mit CMT2003A verbunden und konfiguriert.
- SDK-Version	v4.4.7-dirty
- Konfigurationsversion	0.1.28
- Firmwareversion / git Hash	ab [v24.9.21](https://github.com/tbnobody/OpenDTU/releases/tag/v24.9.21)
- PIO Umgebung	generic_esp32
(Ich habe Version [v24.9.22](https://github.com/tbnobody/OpenDTU/releases/tag/v24.9.22) ebenso getestet, hier Tritt der Fehler auch auf, ältere Versionen habe ich noch nicht getestet)
### reproduce
- Navigiere zu Einstellugnen DTU
- Klicke auf Speichern
- Ändern eines Wertes / einer Einstellung
- Erneutes Speichern
 

### Expected Behavior

DAs Speichern von geänderten Werten sollte möglich sein

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

geteste auf v24.9.21 & v24.9.22

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Anything else?

Leider ist die C Entwicklung bisher an mir vorbei geangen ich würde aber vermuten dass die Änderung in `/src/WebApi_config.cpp:71` das Problem ist.

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

tbnobody, 2024-09-23:
> Leider ist die C Entwicklung bisher an mir vorbei geangen ich würde aber vermuten dass die Änderung in /src/WebApi_config.cpp:71 das Problem ist.

Das schonmal definitiv nicht. Methoden in der WebApi_config werden beim Download von JSON Files aufgerufen. Du beschreibst aber oben eine Speicherung von DTU Einstellungen

DumpName, 2024-09-23:
Der Fehler tritt in v24.8.5 noch nicht auf

tbnobody, 2024-09-23:
Bitte nochmal prüfen ob das Problem definitv mit der 24.9.22 besteht. An der Stelle hatte ich gestern nämlich was behoben (84e5c0821c25d4237071db14374b18721ae9b305).

Tritt der Fehler nur mit 0 db auf oder auch z.B. bei 2db?

DumpName, 2024-09-23:
Hi,
wie oben beschrieben hatte ich v24.9.22 ebenfalls bereits getestet leider ohne Erfolg.
Der Fehler tritt unabhängig davon welche einstellung ich ändere auf. Der erste Request enthält die Daten ohne Änderung (0db) für den 2. Request habe ich auf 1db erhöht. Wenn ich dann wieder auf 0 db wechsle erhalte ich weiterhin den Fehler.
Ich kann auch die Frequenz nicht anpassen. Es ist also vollkommen egal welche Einstellung ich ändere, sobald ich einmal etwas geändert habe kann ich bis zum neuladen der Seite keine Änderungen speichern. Nach dem neuladen dann auch wieder nur die "originalen" Einstellungen.

tbnobody, 2024-09-23:
Welche RF module sind bei euch konfiguriert?

MichaMEG, 2024-09-23:
Beide werden benutzt

tbnobody, 2024-09-23:
> 24.9.20 funktioniert noch, 24.9.22 leider nicht.

Ja, da ist das Upgrade auf ArduinoJson 7.2.0 dabei.

Hm der interessante Teil ist wohl dieser Unterschied:
```
"cmt_palevel": 0,
```
vs. 
```
"cmt_palevel": "1",
```
Also einmal ein String und einmal eine Zahl. Bin noch unsicher wo das entsteht. Aus irgendeinem Grund wird beim Ändern des Levels aus der Zahl ein String. Früher wurde nur geprüft ob der Schlüssel `cmt_palevel` im JSON Objekt enthalten ist. Die neue Methode prüft auch ob der Wert konvertierbar ist. Und hier scheint die ìs` Methode wohl etwas anderes zu machen als die `as` Methode. Muss ich aber noch genauer anschauen.

closed 2024-09-23 by tbnobody

MichaMEG, 2024-09-23:
Vielen Dank, sieht soweit gut aus :)

commit 5c460e26 (2024-09-23): Fix: Unable to CMT transmit power in WebApp

The pa_level was sent as string instead of a number.

fixes #2299