tbnobody/OpenDTU #2300: Statistics Reset algorithm

broth-itk, 2024-09-23:
### Is your feature request related to a problem? Please describe.

Statistics are a great new feature! **Thank you for the enhancement!**

Unfortunately the stats get "poisoned" by the attempts to reach the inverters in the early morning hours.

These failure counts makes it impossible to justify the link quality between DTU and the inverter.

I was helping myself by rebooting the DTU. Now the values look fine and allow to check if the reachability is fine.

### Describe the solution you'd like

Proposal:

The stats already get reset at midnight.
This will collect offline stats.

At the first successful response from the inverter, reset stats and set a flag.

Now we count fresh online stats.

----

First I like to collect feedback from other users and devs.

I'll be happy to code the required changes and issue a pull request on positive feedback :)

tbnobody, 2024-09-23:
Good idea. I am just not 100% sure what happens to people with batteries on their inverter. 
To be honest, the reset at midnight was just a poor first idea to prevent any kind of overflow of  these variables. 
I could also imagine that a button in the webapp or a mqtt command topic to reset the radio stats would make sense.

broth-itk, 2024-09-23:
I like the idea of some automatic mode which results in useful stats for 99% of users.
What I noticed this evening: It's gonna be more difficult to get useful stats when the sun goes down. When is a good moment/trigger to stop counting? No RX could be for any other reason, too :)

Button, MQTT (? are the values even published)... all possible.
Since the stats help to judge the link quality, maybe count only during a window of xx minutes. Then reset.

tbnobody, 2024-09-23:
> (? are the values even published)

Yes they are...

mrueg, 2024-10-11:
Would it be possible to store the counters in a way that you can select by a range, e.g. last 5 min, last hour and last 6 hours?

stefan123t, 2024-10-18:
@mrueg that would require to store them in several round-robin / ring registers for the times / intervals you have given.