tbnobody/OpenDTU #2312: Problème d'incrémentation du relevé de puissance

AlainVOU, 2024-09-28:
### What happened?

A l'affichage En Direct d'OpenDTU, il y a des valeurs positives et négatives disproportionnées, dans la bdd, je constate que le compteur de puissance indique une valeur très en écart avec la précédente et la suivante, sans incrémentation. Suivant le cas, je supprime l'enregistrement éronnée ou je lui donne une valeur, puis je réimporte la bdd, cela corrige l'affichage. Voici l'image de la bdd.
[image]

### To Reproduce Bug

Ce problème est aléatoire, je ne comprends pas ce qui déclenche l'enregistrement erronée. J'ai ce problème depuis le début de mon installation, en décembre 2022. ( 15 erreurs ). Je n'ai trouvé comment corriger les données que récemment. J'ai la dernière version d'OpenDTU bdd.

### Expected Behavior

Qu'il n'y ai pas d'erreur d'incrémentation des données..

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v24.6.10-1-g04881da

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AlainVOU, 2024-09-28:
Désolé, c'est pour https://github.com/RaBa64/OpenDTU-Database/issues

stefan123t, 2024-09-28:
Salut @AlainVOU poivez tu traduire ca en anglais ou allemande ? Les developpers du OpenDTU ne parlent si bien francais malheuresment, par exemple qu’est que tu veu dire avec bdd ?

Silverrider5, 2024-10-01:
'bdd' stands for database, if that can help....if you need more translation let me know.

stefan123t, 2024-12-06:
@AlainVOU est que tu veu fermer le issue içi ?

AlainVOU, 2024-12-06:
Oui, oui

closed 2024-12-06 by AlainVOU