tbnobody/OpenDTU #2316: Stack canary watchpoint triggered (async_tcp) when accessing Web UI

schlimmchen, 2024-09-29:
### What happened?

I am seeing this quite often recently. So often that I now think it is a third kind of new panic. And now that it happened in the upstream project as well, at least twice, I am leaving this here...

### To Reproduce Bug

As the backtraces show, this type of panic occurs when accessing the web interface, but it is not perfectly reproducible.

### Expected Behavior

(This field is stupid when reporting a panic :grin:)

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

b206cee8

### What firmware variant (PIO Environment) are you using?

generic_esp32s3_usb

### Anything else?

<details>
<summary>Log 1</summary>

```
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception).
Debug exception reason: Stack canary watchpoint triggered (async_tcp)
Core  1 register dump:
PC      : 0x420de8a9  PS      : 0x00060136  A0      : 0x820da000  A1      : 0x3fcb4730
A2      : 0x3fcb4ac8  A3      : 0x00000086  A4      : 0x00000001  A5      : 0x0000000c
A6      : 0x3fcb4740  A7      : 0x00000008  A8      : 0x00000001  A9      : 0x3fcb4740
A10     : 0x0000005a  A11     : 0x0000002a  A12     : 0x00000078  A13     : 0x3c161870
A14     : 0x00000078  A15     : 0xff000000  SAR     : 0x0000000b  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x420de8a6:0x3fcb4730 0x420d9ffd:0x3fcb4a40 0x4203eb52:0x3fcb4b00 0x420302e5:0x3fcb4b20 0x420303e6:0x3fcb4bb0 0x42030ada:0x3fcb4bf0 0x4202be33:0x3fcb4d20 0x42022846:0x3fcb4d50 0x4202c2c5:0x3fcb4d70 0x4202286d:0x3fcb4db0 0x4202c2c5:0x3fcb4dd0 0x4202286d:0x3fcb4e10 0x4202c2c5:0x3fcb4e30 0x4202286d:0x3fcb4e70 0x4202c2c5:0x3fcb4e90 0x4202286d:0x3fcb4ed0 0x4202c2c5:0x3fcb4ef0 0x4202286d:0x3fcb4f30 0x4202c2c5:0x3fcb4f50 0x4202286d:0x3fcb4f90 0x4202c2c5:0x3fcb4fb0 0x4202286d:0x3fcb4ff0 0x4202c2c5:0x3fcb5010 0x4202286d:0x3fcb5050 0x4202c2c5:0x3fcb5070 0x4202286d:0x3fcb50b0 0x4202c2c5:0x3fcb50d0 0x4202286d:0x3fcb5110 0x4202c2c5:0x3fcb5130 0x4202286d:0x3fcb5170 0x4202c2c5:0x3fcb5190 0x4202286d:0x3fcb51d0 0x4202c2c5:0x3fcb51f0 0x4202286d:0x3fcb5230 0x4202c2c5:0x3fcb5250 0x4202286d:0x3fcb5290 0x4202c2c5:0x3fcb52b0 0x4202286d:0x3fcb52f0 0x4202c2c5:0x3fcb5310 0x4202286d:0x3fcb5350 0x4202c2c5:0x3fcb5370 0x4202286d:0x3fcb53b0 0x4202c2c5:0x3fcb53d0 0x4202286d:0x3fcb5410 0x4202c2c5:0x3fcb5430 0x4202286d:0x3fcb5470 0x4202c2c5:0x3fcb5490 0x4202286d:0x3fcb54d0 0x4202c2c5:0x3fcb54f0 0x4202286d:0x3fcb5530 0x4202c2c5:0x3fcb5550 0x4202286d:0x3fcb5590 0x4202c2c5:0x3fcb55b0 0x4202286d:0x3fcb55f0 0x4202c2c5:0x3fcb5610 0x4202286d:0x3fcb5650 0x4202c2c5:0x3fcb5670 0x4202286d:0x3fcb56b0 0x4202c2c5:0x3fcb56d0 0x4202286d:0x3fcb5710 0x4202c2c5:0x3fcb5730 0x4202286d:0x3fcb5770 0x4202c2c5:0x3fcb5790 0x4202286d:0x3fcb57d0 0x4202c2c5:0x3fcb57f0 0x4202286d:0x3fcb5830 0x4202c2c5:0x3fcb5850 0x4202286d:0x3fcb5890 0x4202c2c5:0x3fcb58b0 0x4202286d:0x3fcb58f0 0x4202c2c5:0x3fcb5910 0x4202286d:0x3fcb5950 0x4202c2c5:0x3fcb5970 0x4202286d:0x3fcb59b0 0x4202c2c5:0x3fcb59d0 0x4202286d:0x3fcb5a10 0x4202c2c5:0x3fcb5a30 0x4202286d:0x3fcb5a70 0x4202c2c5:0x3fcb5a90 0x4202286d:0x3fcb5ad0 0x4202c2c5:0x3fcb5af0 0x4202286d:0x3fcb5b30 0x4202c2c5:0x3fcb5b50 0x4202286d:0x3fcb5b90 0x4202c2c5:0x3fcb5bb0 0x4202286d:0x3fcb5bf0 0x4202c2c5:0x3fcb5c10 0x4202286d:0x3fcb5c50 0x4202c2c5:0x3fcb5c70 0x4202286d:0x3fcb5cb0 0x4202c2c5:0x3fcb5cd0 0x4202286d:0x3fcb5d10 0x4202c2c5:0x3fcb5d30 0x4202286d:0x3fcb5d70 0x4202c2c5:0x3fcb5d90 0x4202286d:0x3fcb5dd0 0x4202c2c5:0x3fcb5df0 0x4202286d:0x3fcb5e30 0x4202c2c5:0x3fcb5e50 0x4202286d:0x3fcb5e90 0x4202c2c5:0x3fcb5eb0 |<-CONTINUES

ELF file SHA256: 9338ecf52ed17760

0x420de8a6: _svfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1543
0x420d9ffd: sprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/sprintf.c:618
0x4203eb52: MD5Builder::getChars(char*) at /home/schlimmchen/.platformio/packages/framework-arduinoespressif32/cores/esp32/MD5Builder.cpp:108 (discriminator 2)
0x420302e5: getMD5(unsigned char*, unsigned short, char*) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/WebAuthentication.cpp:75
0x420303e6: stringMD5(String const&) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/WebAuthentication.cpp:113
0x42030ada: checkDigestAuthentication(char const*, char const*, char const*, char const*, char const*, bool, char const*, char const*, char const*) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/WebAuthentication.cpp:238 (discriminator 11)
0x4202be33: AsyncWebServerRequest::authenticate(char const*, char const*, char const*, bool) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/WebRequest.cpp:742
0x42022846: AuthenticationMiddleware::allowed(AsyncWebServerRequest*) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/ESPAsyncWebServer.h:648
 (inlined by) AuthenticationMiddleware::allowed(AsyncWebServerRequest*) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/ESPAsyncWebServer.h:647
 (inlined by) AuthenticationMiddleware::run(AsyncWebServerRequest*, std::function<void ()>) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/ESPAsyncWebServer.h:652
0x4202c2c5: AsyncMiddlewareChain::_runChain(AsyncWebServerRequest*, std::function<void ()>)::{lambda()#1}::operator()() const at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/ESPAsyncWebServer.h:623
 (inlined by) std::_Function_handler<void (), AsyncMiddlewareChain::_runChain(AsyncWebServerRequest*, std::function<void ()>)::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:297
0x4202286d: std::function<void ()>::operator()() const at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:687
 (inlined by) AuthenticationMiddleware::run(AsyncWebServerRequest*, std::function<void ()>) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/ESPAsyncWebServer.h:652
0x4202c2c5: AsyncMiddlewareChain::_runChain(AsyncWebServerRequest*, std::function<void ()>)::{lambda()#1}::operator()() const at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/ESPAsyncWebServer.h:623
 (inlined by) std::_Function_handler<void (), AsyncMiddlewareChain::_runChain(AsyncWebServerRequest*, std::function<void ()>)::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:297
0x4202286d: std::function<void ()>::operator()() const at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:687
 (inlined by) AuthenticationMiddleware::run(AsyncWebServerRequest*, std::function<void ()>) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/ESPAsyncWebServer.h:652
0x4202c2c5: AsyncMiddlewareChain::_runChain(AsyncWebServerRequest*, std::function<void ()>)::{lambda()#1}::operator()() const at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/ESPAsyncWebServer.h:623
 (inlined by) std::_Function_handler<void (), AsyncMiddlewareChain::_runChain(AsyncWebServerRequest*, std::function<void ()>)::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:297
0x4202286d: std::function<void ()>::operator()() const at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:687
 (inlined by) AuthenticationMiddleware::run(AsyncWebServerRequest*, std::function<void ()>) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/ESPAsyncWebServer.h:652
[... cut ...] ecause of helgeerbe#1299 (see a screenshot there).

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2024-09-29:
<details>
<summary>Log 2 (notice the PS register is slightly different, so yes, this is actually a second instance of the panic :wink:)</summary>

```
Debug exception reason: Stack canary watchpoint triggered (async_tcp)
Core  1 register dump:
PC      : 0x420de8a9  PS      : 0x00060e36  A0      : 0x820da000  A1      : 0x3fcb4730
A2      : 0x3fcb4ac8  A3      : 0x00000086  A4      : 0x00000001  A5      : 0x0000000c
A6      : 0x3fcb4740  A7      : 0x00000008  A8      : 0x00000001  A9      : 0x3fcb4740
A10     : 0x0000005a  A11     : 0x0000002a  A12     : 0x00000078  A13     : 0x3c161870
A14     : 0x00000078  A15     : 0xff000000  SAR     : 0x0000000b  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x420de8a6:0x3fcb4730 0x420d9ffd:0x3fcb4a40 0x4203eb52:0x3fcb4b00 0x420302e5:0x3fcb4b20 0x420303e6:0x3fcb4bb0 0x42030ada:0x3fcb4bf0 0x4202be33:0x3fcb4d20 0x42022846:0x3fcb4d50 0x4202c2c5:0x3fcb4d70 0x4202286d:0x3fcb4db0 0x4202c2c5:0x3fcb4dd0 0x4202286d:0x3fcb4e10 0x4202c2c5:0x3fcb4e30 0x4202286d:0x3fcb4e70 0x4202c2c5:0x3fcb4e90 0x4202286d:0x3fcb4ed0 0x4202c2c5:0x3fcb4ef0 0x4202286d:0x3fcb4f30 0x4202c2c5:0x3fcb4f50 0x4202286d:0x3fcb4f90 0x4202c2c5:0x3fcb4fb0 0x4202286d:0x3fcb4ff0 0x4202c2c5:0x3fcb5010 0x4202286d:0x3fcb5050 0x4202c2c5:0x3fcb5070 0x4202286d:0x3fcb50b0 0x4202c2c5:0x3fcb50d0 0x4202286d:0x3fcb5110 0x4202c2c5:0x3fcb5130 0x4202286d:0x3fcb5170 0x4202c2c5:0x3fcb5190 0x4202286d:0x3fcb51d0 0x4202c2c5:0x3fcb51f0 0x4202286d:0x3fcb5230 0x4202c2c5:0x3fcb5250 0x4202286d:0x3fcb5290 0x4202c2c5:0x3fcb52b0 0x4202286d:0x3fcb52f0 0x4202c2c5:0x3fcb5310 0x4202286d:0x3fcb5350 0x4202c2c5:0x3fcb5370 0x4202286d:0x3fcb53b0 0x4202c2c5:0x3fcb53d0 0x4202286d:0x3fcb5410 0x4202c2c5:0x3fcb5430 0x4202286d:0x3fcb5470 0x4202c2c5:0x3fcb5490 0x4202286d:0x3fcb54d0 0x4202c2c5:0x3fcb54f0 0x4202286d:0x3fcb5530 0x4202c2c5:0x3fcb5550 0x4202286d:0x3fcb5590 0x4202c2c5:0x3fcb55b0 0x4202286d:0x3fcb55f0 0x4202c2c5:0x3fcb5610 0x4202286d:0x3fcb5650 0x4202c2c5:0x3fcb5670 0x4202286d:0x3fcb56b0 0x4202c2c5:0x3fcb56d0 0x4202286d:0x3fcb5710 0x4202c2c5:0x3fcb5730 0x4202286d:0x3fcb5770 0x4202c2c5:0x3fcb5790 0x4202286d:0x3fcb57d0 0x4202c2c5:0x3fcb57f0 0x4202286d:0x3fcb5830 0x4202c2c5:0x3fcb5850 0x4202286d:0x3fcb5890 0x4202c2c5:0x3fcb58b0 0x4202286d:0x3fcb58f0 0x4202c2c5:0x3fcb5910 0x4202286d:0x3fcb5950 0x4202c2c5:0x3fcb5970 0x4202286d:0x3fcb59b0 0x4202c2c5:0x3fcb59d0 0x4202286d:0x3fcb5a10 0x4202c2c5:0x3fcb5a30 0x4202286d:0x3fcb5a70 0x4202c2c5:0x3fcb5a90 0x4202286d:0x3fcb5ad0 0x4202c2c5:0x3fcb5af0 0x4202286d:0x3fcb5b30 0x4202c2c5:0x3fcb5b50 0x4202286d:0x3fcb5b90 0x4202c2c5:0x3fcb5bb0 0x4202286d:0x3fcb5bf0 0x4202c2c5:0x3fcb5c10 0x4202286d:0x3fcb5c50 0x4202c2c5:0x3fcb5c70 0x4202286d:0x3fcb5cb0 0x4202c2c5:0x3fcb5cd0 0x4202286d:0x3fcb5d10 0x4202c2c5:0x3fcb5d30 0x4202286d:0x3fcb5d70 0x4202c2c5:0x3fcb5d90 0x4202286d:0x3fcb5dd0 0x4202c2c5:0x3fcb5df0 0x4202286d:0x3fcb5e30 0x4202c2c5:0x3fcb5e50 0x4202286d:0x3fcb5e90 0x4202c2c5:0x3fcb5eb0 |<-CONTINUES

ELF file SHA256: 9338ecf52ed17760

0x420de8a6: _svfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1543
0x420d9ffd: sprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/sprintf.c:618
0x4203eb52: MD5Builder::getChars(char*) at /home/schlimmchen/.platformio/packages/framework-arduinoespressif32/cores/esp32/MD5Builder.cpp:108 (discriminator 2)
0x420302e5: getMD5(unsigned char*, unsigned short, char*) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/WebAuthentication.cpp:75
0x420303e6: stringMD5(String const&) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/WebAuthentication.cpp:113
0x42030ada: checkDigestAuthentication(char const*, char const*, char const*, char const*, char const*, bool, char const*, char const*, char const*) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/WebAuthentication.cpp:238 (discriminator 11)
0x4202be33: AsyncWebServerRequest::authenticate(char const*, char const*, char const*, bool) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/WebRequest.cpp:742
0x42022846: AuthenticationMiddleware::allowed(AsyncWebServerRequest*) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/ESPAsyncWebServer.h:648
 (inlined by) AuthenticationMiddleware::allowed(AsyncWebServerRequest*) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/ESPAsyncWebServer.h:647
 (inlined by) AuthenticationMiddleware::run(AsyncWebServerRequest*, std::function<void ()>) at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/ESPAsyncWebServer.h:652
0x4202c2c5: AsyncMiddlewareChain::_runChain(AsyncWebServerRequest*, std::function<void ()>)::{lambda()#1}::operator()() const at /home/schlimmchen/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/ESPAsyncWebServer/src/ESPAsyncWebServer.h:623
 (inlined by) std::_Function_handler<void (), AsyncMiddlewareChain::_runChain(AsyncWebServerRequest*, std::function<void ()>)::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:297
[... cut ...]
 (inlined by) std::_Function_handler<void (), AsyncMiddlewareChain::_runChain(AsyncWebServerRequest*, std::function<void ()>)::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:297
Error: '|<-CONTINUES' does not match the expected format.
```
</details>

Have to put this in a new comment, as otherwise the first comment (issue report) is longer than `uint32_t` chars, which GitHub does not like :wink:

stefan123t, 2024-09-30:
@schlimmchen yes the expected behaviour is a kind of silly question in case of a race condition or common bug. But we introduced this as part of the original github issue template, form follows function here.

What astonished me is the frequent occurrence of AuthenticationMiddleware, but then I havent peeked at a stack trace often, recently. Is this part of some SSL stack I am unaware of or is this the safety check for the AdminPassword for most of the requests ?

schlimmchen, 2024-09-30:
> the safety check for the AdminPassword for most of the requests?

Yes.

My best guess is that something (ESPAsyncWebServer) iterates this middleware chain, but the chain is corrupted, such that the loop does not stop where it is supposed to.

closed 2024-09-30 by tbnobody