tbnobody/OpenDTU #2325: OpenDTU bleibt auf großen Leistungen hängen

AlbPV, 2024-10-02:
### What happened?

Ich habe seit einigen Wochen das System des Anbieters Selbstbau PV installiert. Die DTU macht den ganzen Tag über genau was sie soll. Nun zum Problem, sobald größere Lasten entstehen (mehr als der Wechselrichter einspeisen soll) bleibt die DTU auf dem voreingestellten Wert hängen, d.h. auch wenn die Last wieder sinkt, bleibt das Limit stehen. 

Desweiteren ist ab diesem Moment der Wechselrichter nicht mehr erreichbar. 
Nachdem der Wechselrichter aus und wieder eingeschaltet wird funktioniert alles wieder perfekt. 
Eine Veränderung der Frequenz oder des max. Limit ergab keine Verbesserung.

### To Reproduce Bug

Last im Stromnetz überhalb des eingestellten Limit. Bsp. Max.Limit 1000w/ Last im Netz 2300W.

### Expected Behavior

OpenDTU soll nachdem die Last nicht mehr vorhanden ist wieder runter modulieren.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

V24.9.30

### What firmware variant (PIO Environment) are you using?

generic 

### Relevant log/trace output

```Shell
-
```

### Anything else?

-

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2024-10-02:
> OpenDTU soll nachdem die Last nicht mehr vorhanden ist wieder runter modulieren.

OpenDTU moduliert hier garnichts selbst. Das ist deine Aufgabe hier etwas zu implementieren. Man muss OpenDTU ein Limit schicken und diese schickt es zum  Wechselrichter. Bist du sicher, dass du nicht OpenDTU-OnBattery verwendest?

AlbPV, 2024-10-02:
Die OpenDTU bekommt ihre Werte von einem SmartMicroSolar. Es wird keine openDTU onBaterry verwendet.
Es ist nur aufgefallen, dass in der Funkstatistik, sobald dieser Fehler auftritt, die nicht empfangenen Anfragen ansteigen,bzw die Anfragen nicht mehr empfangen werden. 

Ich bin leider nicht noch nicht ganz so tief in der Materie Zuhause.

stefan123t, 2024-10-03:
@AlbPV es scheint da ist ein Problem auf Seite der SmartMucroSolar von Selbstbau PV. Vermutlich versucht die den WR auf einen absoluten Wert zu regeln, den der WR gar nicht kann. Dieses unmögliche / zu hohe sog. Active Power Limit wird dann vermutlich vom WR nicht quittiert werden und der WR ist dann vorrübergehend nicht erreichbar.

Bitte mit dem Hersteller Deiner Regelung besprechen oder ein Serial Log mit dem Problem auf Seite der OpenDTU mitschneiden und anhängen. Danke!

https://www.opendtu.solar/firmware/howto/serial_console/

closed 2025-04-18 by github-actions[bot]