tbnobody/OpenDTU #2345: Pressing F5 in Browser (reload) leeds to instant reboot of the Device

BrAiNeeBug, 2024-10-12:
### What happened?

Device restarts ESP32-D0WD-V3 (Software reset CPU)

### To Reproduce Bug

just press F5 in Browser (maybe two times)

### Expected Behavior

it should just reload the site or wait if the CPU is free again maybe drop such request or fix the WatchDog.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v24.10.6

### What firmware variant (PIO Environment) are you using?

opendtu-generic_esp32.bin

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2024-10-12:
Cannot reproduce. It also works if I press F5 ten times and more. What do you see on the serial console if the restart occours?

tbnobody, 2024-10-12:
And btw... there is a ongoing discussion here: #2326

BrAiNeeBug, 2024-10-13:
> Cannot reproduce. It also works if I press F5 ten times and more. What do you see on the serial console if the restart occours?

and if you hold the Button Down ? For me 2 Times refresh in short pressing the F5 it insta reboots, no Console Logs postet in these moment. i treid external power-source yesterday (no change) so its not power issiu or maybe ?

tbnobody, 2024-10-15:
The issue should be solved with version v24.10.15 please doublecheck and close the issue

closed 2024-10-16 by tbnobody

BrAiNeeBug, 2024-10-17:
i updated to v24.10.15 and the issue is resolved, thx you bro ♥