tbnobody/OpenDTU #2347: Ansicht bei mehreren Wechselrichter

sbernhard, 2024-10-13:
### What happened?

Bei mehreren Wechselrichter sieht der Umschalter komisch aus. Siehe Foto. Sieht nach einem CSS Fehler aus.

### To Reproduce Bug

Klick auf Live View

### Expected Behavior

Icon direkt neben Namen des Wechselrichters

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

24.10.6

### What firmware variant (PIO Environment) are you using?

24.10.6

### Relevant log/trace output

```Shell
Nicht relevant
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

sbernhard, 2024-10-13:
[image]

schneeer, 2024-11-03:
Hi.

Da schlägt der Zeilenumbruch zu. Insbesondere, wenn alles, bis auf die Buttons, in einer Spalte dargestellt wird.
Allerdings auch, wenn der Text nur ein Zeichen ist. Zudem sind die Buttons schmaler und nutzen nicht die gesamte zur Verfügung stehenden Breite.
[image]
[image]

schlimmchen, 2024-11-03:
In der Tat, das ist mir auch schon störend aufgefallen, allerdings hatte ich in #2388 bezüglich dieser Buttons bisher nur das Aneinanderkleben der Buttons korrigiert.

Vorhin habe ich commit b21e077154b8ff21f55e67337d48512e62600dea in #2388 geschoben, der das hier beschriebene Problemchen lösen sollte.

[image]

[image]

tbnobody, 2025-01-14:
Thank you @schlimmchen.. Going to close this issue then..