tbnobody/OpenDTU #2349: [Request] Anfrage-Häufigkeit in der Nach reduzieren

sbernhard, 2024-10-13:
### Is your feature request related to a problem? Please describe.

Nicht wirklich, außer dass es viele "Fehler" bei den Radio-Statistics gibt

### Describe the solution you'd like

Während der Nacht, bzw. wenn der HMS mehrmals nicht erreichbar war, die Häufigkeit der Anfragen von der DTU zum HMS reduzieren. 

### Describe alternatives you've considered

- Einstellungen "Nachtzeiten wo keine Anfragen geschickt werden"

tbnobody, 2024-10-13:
> Einstellungen "Nachtzeiten wo keine Anfragen geschickt werden"

Was spricht gegen diese Einstellung?

tbnobody, 2024-10-13:
außerdem, 

Duplicate of #547

closed 2024-10-13 by tbnobody

sbernhard, 2024-10-13:
Oh, sorry. Natürlich, dass ist die Option die ich gesucht habe. Ganz vergessen 😃