tbnobody/OpenDTU #2357: "Funkstatistik zurücksetzen" bei openDTU Fusion funktioniert nur wenige Male

Fokker-1, 2024-10-16:
### What happened?

Hallo,

ich habe eine openDTU Fusion, an der ein HM-800 und ein HMS-1000 angeschlossen sind. Das funktioniert prima.

Heute habe ich des FW-Updtae auf 24.10.15 gemacht. Dabei entdeckte ich, dass es schon seit einigen FW-Versionen die Funktion "Funkstatistik" gibt. Das hatte ich wohl zuvor übersehen.

Aber super, vielen Dank, das ist genau ein Feature, das ich mir schon lange gewünscht hatte!!
Toll wäre es auch, wenn das noch in openDTU on Battery implementiert würde!

Bei der Nutzung fiel mir dann auf:

Nach Einschlten der DTU funktioniert das Zurücksetzen, aber nur 1x (vielleicht war es auch 2x).
Danach ist der Button deaktiviert:

[image]

Eine Aktivierung galang bei mir erst wieder, nachdem ich manuell  "Neustart" durchgeführt hatte. 
Danach wiederholte sich der gleiche Ablauf immer wieder.

Gruß
Werner

### To Reproduce Bug

Unter "Funkstatistik" die Funktion "Statistiken zurücksetzen" mehrfach aufrufen. Der Button wird dann deaktiviert.
Aktivierung erst wieder durch manuellen "Neustart" möglich.

### Expected Behavior

Der Button "Statistiken zurücksetzen" soll nach Nutzung nicht dektiviert werden.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

24.10.15

### What firmware variant (PIO Environment) are you using?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
n.a.
```

### Anything else?

Danke für den tollen Support!

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2024-10-16:
The button gets disabled during the reset process (if you see the spinning wheel within the button). Afterwards the button gets enabled again (after the API request returns). What browser with which version are you using? Are you logged in when pressing the reset button?

ms49434, 2024-10-16:
Ich bin dagegen daß an OpenDTU etwas geändert werden kann ohne vorherige Authentifizierung. Auch wenn es nur der Button ist, um Statistikdaten zu löschen.

Fokker-1, 2024-10-16:
Nun denn, so hat jeder seine persönliche Meinung, ist ja auch ok. so.

Nur am Rande:
Man kann auch die laufenden Daten in der Konsole leeren - ohne Login.
Ich sehe da keinen Unterschied zum Leeren der Funkstatistik, bei beiden kann man auch nichts kaputt machen.

Gruß
Werner

closed 2024-10-16 by tbnobody

Fokker-1, 2024-10-17:
Guten Morgen,

danke für diese sachlichen Informationen, das klärt bei mir als Nicht-Insider bei diesem Thema das Verständnis.
Ok., dann ist jetzt alles klar.

Gruß
Werner