tbnobody/OpenDTU #2359: [Request] Integration (via HACS) for Home Assistant

devdems, 2024-10-16:
### Is your feature request related to a problem? Please describe.

PlugPlay solution to get the data from OpenDTU to Home Assistant.

### Describe the solution you'd like

Integration in HA should be installed via HACS and should be able to connect via MQTT/Rest API to OpenDTU.

tbnobody, 2024-10-16:
Will not implement. You can use MQTT and enable the MQTT auto discovery feature. No need for a separate custom integration which has to be written, maintained etc...

closed 2024-10-16 by tbnobody

stefan123t, 2024-10-16:
@devdems as noted by tbnobody and me in your respective discussion #2358 you should consult the fine manual and use the MQTT integration which works just fine, no need for additional HACS plugin development and configuration.

MQTT Settings
https://www.opendtu.solar/firmware/configuration/mqtt_settings/

Enable Home Assistant MQTT Auto Discovery 

Enables or disables the [Home Assistant MQTT Auto Discovery](https://www.home-assistant.io/integrations/mqtt/#mqtt-discovery) functionality.