tbnobody/OpenDTU #2362: [Request] Support for new function "Power Distribution Logic“

AlexJacu, 2024-10-18:
### Is your feature request related to a problem? Please describe.

Unterstützung für die neue umsetzung der drosselung. 

Mitteilung von Hoymiles: 
Die Reihe HMS-1600/1800/2000-4T unterstützt ab sofort die „Power Distribution Logic“, die die Ausgangsleistung des Wechselrichters und nicht den DC-Eingang begrenzt: Wenn der HMS1600-4T beispielsweise auf 800 W begrenzt ist und nur zwei Eingänge verwendet werden, können die aktiven Eingänge 400W+400 W statt 200W+200 W erreichen.

Sie benötigen dafür die Wechselrichter-Firmware 01.01.12 (für HMS-1600/1800/2000-4T )

Ticket Service für die Beantragung (Serialnummer des WR immer eingeben): Submit a request – Hoymiles

Das wird auch für die kommenden Generation (with Wifi+Bluetooth) gelten.

HMS-600/700/800/900/1000-2WB (unlisted)
HMS-1600/1800/2000-4WB (unlisted)

### Describe the solution you'd like

Support für "Power Distribution Logic“

stefan123t, 2024-10-20:
@AlexJacu danke für den Feature Request und den Hinweis, dass die neuere Hoymiles Firmware ein Eingang unabhängiges Active Power Limit kann. 

Wir können leider immer noch kein [Grid Profile Update](https://github.com/tbnobody/OpenDTU/issues/900#issuecomment-1899406249) geschweige denn ein [Firmware Update](https://github.com/tbnobody/OpenDTU/issues/1421) einspielen.

Abgesehen davon müsste man sich die Kommandos auf DTU Seite ansehen / mitschneiden, ob und wie sich das Active Power Limit Command geändert hat, um diese Funktion auch in OpenDTU zu unterstützen. 

Hast Du eine original Hoymiles DTU - die Du mit einem Logic Analyser / [zwei USB Serial Konvertern verbinden](https://github.com/tbnobody/OpenDTU/issues/1650#issuecomment-1900330064) kannst - um das Firmware Update **und** das neue Active Power Limit Command mitzuschneiden ?

@broth-itk und nivadis haben bereits ihre DTUs angepaßt / umgebaut, damit das geht, vielleicht könnten die dies ja zum Anlaß nehmen und das Firmware Update und das neue Active Power Limit Command mitschneiden ?

broth-itk, 2024-10-21:
Aktuelles Traces von Firmware- und Grid Profile Update kann ich gerne ziehen.
Die Scripte, mit denen ich die Daten abrufe, werden noch optimiert.
Dazu auch meine Frage in #1650 zum Framing.

broth-itk, 2024-10-22:
Hier sind aktuelle Traces: https://github.com/tbnobody/OpenDTU/issues/1650#issuecomment-2428650265

Im Bereich "Plant" lässt sich momentan nur "Active Power" auswählen

[image]

"Active Power Control (APC) means users can adjust the percentage of maximum output power and rated output power. For example, if the percentage is set to 70%, the maximum output power will be only 70% of the rated output power."

Habe das noch nicht ausprobiert da meine Wechselrichter momentan über OpenDTU fest limitiert sind.
Ist bekannt was die Funktion genau bewirkt?

Bei Interesse kann ich mal mitschneiden was genau an die Wechselrichter geschickt wird.

stefan123t, 2024-10-22:
@broth-itk auf S-Miles Cloud Seite sieht das immer noch gleich aus. Es ist dann wohl wirklich eine reine Firmware Verbesserung.
Bisher war es zB beim HM-600 so dass bei 40% Active Power Limit, jeder der beiden DC Eingänge ein eigenes Limit von 20% hatte, dh das APL des WR wurde durch die Anzahl der Eingänge geteilt. Wenn der eine Eingang mehr gebracht hat und der andere weniger dann wurde das nicht genutzt.

schlimmchen, 2024-10-27:
Was ist denn hierzu in OpenDTU zu tun, wenn ich so einen WR habe, der das kann? Ich sende ein Limit, wie bisher, und freue mich, dass der WR die gewünschte Ausgangsleistung auch erreicht. Fertig?

stefan123t, 2024-10-27:
@schlimmchen ja ist m.E. nichts anzupassen, das ist eine interne Verteilung / Berechnung der aktuellen Hoymiles Firmware. Speziell in den Traces von @broth-itk habe ich immer noch die selben ActivePowerLimit 0x0B Commands gesehen.
Interessant wäre es natürlich das von @broth-itk mitgeschnittene Firmware Update auch mit OpenDTU auf die WR flashen zu können, dann könnten auch Nutzer ohne ein Hoymiles DTU in den Vorzug dieser Firmware Verbesserung kommen.
Dazu wäre aber m.E. erstmal die Implementierung des weitaus kürzeren GridProfile Updates sinnvoll.

AlexJacu, 2024-10-28:
Hallo, glücklicherweise habe ich von einem Lieferanten eine original DTU zugeschickt bekommen obwohl ich sie nicht bestellt habe die steht auch nirgendwo auf der Rechnung drauf. Die habe ich mal programmiert und werde dann mal die Firmware aktualisieren.

Gruß von unterwegs,
A. Jacubowsky
Bitte entschuldigen Sie evtl. Schreibfehler und die Kürze der Nachricht.

AlexJacu, 2024-10-28:
@broth-itk
Ich habe den Wechselrichter nun aktualisiert auf die neue Version allerdings bekommt jetzt meine open DTU keine Verbindung mehr mit dem Wechselrichter alles bereits neu gestartet trotzdem keine Antwort vom Wechselrichter. Muss ich eventuell die Open DTU noch aktualisieren bevor das funktioniert? 
OpenDTU

Systeminformationen
Firmwareinformationen
Hostname	OpenDTU
SDK-Version	v4.4.6-dirty
Konfigurationsversion	0.1.28
Firmwareversion / git Hash	v24.4.24
PIO Umgebung	generic
Firmware-Aktualisierung	
Reset Grund CPU 0	Software reset CPU
Reset Grund CPU 1	Software reset CPU
Anzahl der Konfigurationsspeicherungen	41
Betriebszeit	0 Tage 00:55:54
Hardwareinformationen
Chip-Modell	ESP32-D0WD-V3

schlimmchen, 2024-10-28:
Schau mal hier: #2374 Zeigt dein WR auch Hardwareteilenummer 124097 nach dem Update?

AlexJacu, 2024-10-28:
Ich habe vorhin ein Update auf einem HMS 1600 - 4 t durchgeführt und die Funktion getestet es funktioniert einwandfrei über die vorhandene Der Limitierung habe ich Die Funktionalität getestet und tatsächlich es wird der Ausgang limitiert und nicht mehr der Eingang der module. Die Module werden aber nach Möglichkeit gleichmäßig ausgelastet. Ich hatte bei der Limitierung auf 400 Watt Werte bei den Modulen der Leistung von 90 bis 130 Watt

spcqike, 2024-11-15:
@AlexJacu 
Hast du das Limit mit der openDTU übertragen, oder mit der originalen DTU? 
in #2153 wird davon berichtet, dass Hoymiles mit V01.01.12 wohl die Kommunikation der Wechselrichter ändert? Dass man sie zwar noch auslesen, aber nicht mehr steuern kann. Kannst du das bestätigen?

AndreasBoehm, 2025-01-10:
Ich habe auf zwei HMS-2000 und einem HMS-1600 die V01.01.12 installiert und alle drei funktionieren einwandfrei mit OpenDTU, gab keinerlei Probleme und es musste auch nichts umgestellt werden.

Ich denke hier kann zu.

Kroki0815, 2025-01-22:
Weis zufällig jemand ob die HMT-Serie auch die Power-Distribution-Logic mit der Firmware 1.2.27 unterstützt ?
Hatte ein Changelog bei HM angefordert aber nix bekommen.

schlimmchen, 2025-01-22:
> Ich denke hier kann zu.

Sieht eher so aus, als sei doch noch etwas (vermutlich eine Kleinigkeit) zu tun: https://github.com/tbnobody/OpenDTU/discussions/2153#discussioncomment-11914884

Vielleicht magst du das einmal nachvollziehen, @AndreasBoehm Ich hab meinen WR ja noch nicht aktualsiert 🙊

AndreasBoehm, 2025-01-22:
Werde ich die Tage mal ausprobieren.

bytesinmotion, 2025-01-31:
GIbt es eine Möglichkeit das Firmware update zu installieren ohne offizieller Hoymiles DTU?

bytesinmotion, 2025-01-31:
> 
> Bestimmt hat da jemand schon eine Möglichkeit gefunden. OpenDTU kann es jedenfalls beim heutigen Stand nicht.

d.h. günstigste Möglichkeit aktuell DTU-wlite zum flashen und wieder ausstecken danach :)

ms49434, 2025-01-31:
> > Bestimmt hat da jemand schon eine Möglichkeit gefunden. OpenDTU kann es jedenfalls beim heutigen Stand nicht.
> 
> d.h. günstigste Möglichkeit aktuell DTU-wlite zum flashen und wieder ausstecken danach :)

Das ist die falsche DTU. Wenn schon, dann die DTU-WLite-S

AndreasBoehm, 2025-03-27:
> > Ich denke hier kann zu.
> 
> Sieht eher so aus, als sei doch noch etwas (vermutlich eine Kleinigkeit) zu tun: [#2153 (reply in thread)](https://github.com/tbnobody/OpenDTU/discussions/2153#discussioncomment-11914884)
> 
> Vielleicht magst du das einmal nachvollziehen, [@AndreasBoehm](https://github.com/AndreasBoehm) Ich hab meinen WR ja noch nicht aktualsiert 🙊

Ich hab vergessen hier zu Antworten.
Das neue Limit Command hab ich ausprobiert, konnte aber keinen unterschied feststellen.

https://github.com/hoylabs/OpenDTU-OnBattery/pull/1571

ThoMah61, 2025-05-15:
> > > Bestimmt hat da jemand schon eine Möglichkeit gefunden. OpenDTU kann es jedenfalls beim heutigen Stand nicht.
> > 
> > 
> > d.h. günstigste Möglichkeit aktuell DTU-wlite zum flashen und wieder ausstecken danach :)
> 
> Das ist die falsche DTU. Wenn schon, dann die DTU-WLite-S für HMS/HMT Modelle mit Sub-1G Funkmodul

Hallo zusammen, kann man alle DTU-WLite-S benutzen, wenn ich beim großen A schaue steht da bei den etwas günstigeren immer das die nur für die HMS 800 oder die HMS 1600 sind? Danke

stefan123t, 2025-05-18:
Die neue Firmware mit Power Distribution Logic PDL verwendet Version 3 des SystemConfigPara response https://github.com/tbnobody/OpenDTU/issues/2641#issuecomment-2886433841

OutdoorRob, 2025-07-26:
> Die neue Firmware mit Power Distribution Logic PDL verwendet Version 3 des SystemConfigPara response [#2641 (comment)](https://github.com/tbnobody/OpenDTU/issues/2641#issuecomment-2886433841)

Ich klinke mich in die Thematik mal mit ein. @stefan123t, dass kann ich so nicht bestätigen.

OpenDTU 25.5.10 (auf Fusion Board)

Wechselrichter: 
- Modell: HMS-1600-4T
- **Firmware-Version: 2.0.4**
- Firmware-Erstellungsdatum: 2024-12-19 17:49:00
- 'Power Distribution Logic' unterstützt: Ja

Siehe mein (formatierter) Log: auch mit der Firmware 2.0.4 sendet der Wechselrichter immer noch das bekannte `SystemConfigPara` Version 1 (0x0001).
```
[14:48:21.982] D (62510118) hoymiles: TX SystemConfigPara 865.00 MHz 
  --> 15 A00CA84B 80106080 80 0500 6884CE8F  0000 0000 0000 0000 5637 63
[14:48:21.982] D (62510205) hoymiles: RX 865.00 MHz 
  --> 95 A00CA84B 80106080 81 0001 0258 0000 03E8 0000 0000 0000 3D53 F5
```

In den [Logs von broth-itk](https://github.com/tbnobody/OpenDTU/issues/2641#issuecomment-2882981936) steckt denke ich ein Fehler. Siehe dazu [meinen Beitrag in #2641](https://github.com/tbnobody/OpenDTU/issues/2641#issuecomment-3122166008).

stefan123t, 2025-07-27:
Ja Dein Wechselrichter antwortet mit "version" (?) 0x0001 wie vorherige Firmware Versionen auch.

Ich kann aber bei den Logs von Bernhard keinen Fehler erkennen. Dort steht eindeutig 0x0003 wobei ich zugegeben die CRC16 und CRC8 Checksummen nicht kontrolliere das macht die OpenDTU mW schon selbst.

OutdoorRob, 2025-07-27:
@stefan123t 
Ich wollte nur der **allgemeinen** Aussage widersprechen, dass ein Firmware-Update welches PDL unterstützt, automatisch auch eine andere `SystemConfigPara`-Response (0x0003) nach sich zieht.

Der 'Fehler' in den Logs von Bernhard ist, dass diese zu zwei unterschiedlichen WR gehören und somit die Botschaften nicht gemeinsam betrachtet werden können / wir diese nicht vermischen dürfen.
Ist keine Kritik - nur ein Hinweis, dass wir hier nicht in der falschen Richtung weitersuchen :-)