tbnobody/OpenDTU #2363: Victron MPPT Status Lastausgang

hoschiking, 2024-10-19:
### What happened?

ich habe schon mal davon berichtet, ohne Erfolg, daher ein Issue.
Wenn ein VICTRON MPPT an die OpenDTUoB angeschlossen, wird auch der Status des **LAST**AUSGANGS angezeigt.
Fälschlicher Weise wird dieser jedoch als **LADE**AUSGANG angezeigt und das ist falsch.
Auch wenn es eine Kleinigkeit ist, sollte es korrigiert werden, da es dadurch einen falschen Status suggeriert.
Siehe Screenshot. Wenn der Laderegler sich im Modus "Bulk" befeindet, kann der Ladeausgang unmöglich OFF sein.
Dies betrifft alle MPPT, die einen Lastausgang haben, also bis 20A soweit ich weiß.
[image]

### To Reproduce Bug

Einfach den Lastaugang per APP ausschalten, schon ist der Fehler erkennbar.

### Expected Behavior

Umbennenung in der Webansicht

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

2024.09.11

### What firmware variant (PIO Environment) are you using?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
nicht relevant
```

### Anything else?

es betrifft alle mir bekannten Releases.

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2024-10-19:
Magst du bitte den bug hier schließen und im opendtu on battery Projekt aufmachen? Hier at das eigentlich falsch am Platz da opendtu Ansich keine Victron Unterstützung usw hat.

hoschiking, 2024-10-19:
Ich frage mich, wie ich ob diesem Projekt gelandet bin. Sorry dafür. Ich kann es mir nicht erklären.

closed 2024-10-19 by tbnobody