tbnobody/OpenDTU #2366: OpenDTU-Wifi does not reconnect after a network WiFi failure

dg3feh, 2024-10-21:
### What happened?

When e.g. rebooting my wifi accesspoints (Mikrotik Devices)  in the network the OpenDTU-Esp32 does not reconnect to the network. All esphome devices are fine with that. 

### To Reproduce Bug

Just reboot the Mikrotik Devices and it will happen again

### Expected Behavior

After a reboot the OpenDTU-esp32 connect to the Wifi and requets an IP using DHCP.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v24.10.15

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2024-10-21:
Did you consider this?
[image]
https://www.opendtu.solar/firmware/faq/#debugging

dg3feh, 2024-10-21:
AccessPointTimeout is set to 30min.  The Device reboot takes about 1...2min, thus it should reconnect after 10min, at least after 20min, but it doesn't.

tbnobody, 2024-10-21:
In this case, please provide a output of the serial console. 
But I think it's the same as #2202

mhin76, 2024-11-11:
Same issue here since upgrade to newer version. I am turning off my wifi on purpose over night. That worked just fine on version v24.6.29 using prebuild generic-ESP32 version. But on later version I had to reboot the DTU every time. I am back on version v24.6.29 again.

stinkmop, 2024-11-25:
not exactly the same, but very similar. my wifi is switched off every night for security reasons. During the day it happens again and again that the OpenDTU cannot be pinged. A few minutes later it works again without me having changed anything. The OpenDTU is only 1 metre away from the WLAN router. This means that the GUI is sometimes not accessible. This problem has been going on for a few weeks. I have regularly updated the firmware of the OpenDTU. When I bought the OpenDTU in May, this problem did not exist. It only appeared in the months that followed. Unfortunately, I don't know how to install an older firmware to test when this problem occurs.

mhin76, 2024-11-25:
> Not exactly the same, but very similar. My wifi is switched off every night for security reasons. During the day it happens again and again that the OpenDTU cannot be pinged. A few minutes later it works again without me having changed anything. The OpenDTU is only 1 meter away from the WLAN router. This means that the GUI is sometimes not accessible. This problem has been going on for a few weeks. I have regularly updated the firmware of the OpenDTU. When I bought the OpenDTU in May, this problem did not exist. It only appeared in the months that followed. Unfortunately, I don't know how to install an older firmware to test when this problem occurs.

try to downgrade to v24.6.29. you can download prebuild firmware from github repository. how do you upgrade currently to new version? in case you also use a file based firmware, you can just use (overwrite it with) older version.

stinkmop, 2024-11-25:
try to downgrade to v24.6.29. you can download prebuild firmware from github repository. how do you upgrade currently to new version? in case you also use a file based firmware, you can just use (overwrite it with) older version.

I currently upgrade in the GUI of the OpenDTU. I download it from here: https://github.com/tbnobody/OpenDTU/releases

I tried a downgrade to an older version in the GUI as well, but this failed. As I am not an IT expert, I have no idea how to downgrade other than with the GUI.

mhin76, 2024-11-25:
> I currently upgrade in the GUI of the OpenDTU. I download it from here: https://github.com/tbnobody/OpenDTU/releases
> 
> I tried a downgrade to an older version in the GUI as well, but this failed. As I am not an IT expert, I have no idea how to downgrade other than with the GUI.

I use a special upgrader, but that is because I have bought a prebuild OpenDTU hardware: https://www.diamex.de/dxshop/DX-DTU-Klassik-mit-kleinem-OLED-Display-096-242-cm under download there is this DX-DTU uploader. But if you don't use a DX version, I would not use it. Cannot further help you on downgrading. Maybe someone else in the community has an idea or you find it by googling. Good luck!

FranzToepfer, 2025-01-14:
Same here - WiFi on AVM router disabled over night, on next day Open DTU does not reconnect to re-enabled WiFi

Need to fix it by using a Zigbee plug, automatically switched Off/On early in the morning, when WiFi is assumed to be active again. Kind of hack working for the moment, appreciate if another solution could be found within DTU itself, to periodically scan WiFi networks while in disconnected mode, to re-connect in case _the known one_ will be back online.

mhin76, 2025-01-18:
> Same here - WiFi on AVM router disabled over night, on next day Open DTU does not reconnect to re-enabled WiFi
>
> Need to fix it by using a Zigbee plug, automatically switched Off/On early in the [...]

I used same workaround. Hope to have soon a bug fix on this on a new firmware. Happy to provide any logging required to support :)

SoniXtrA, 2025-02-21:
Hi,
I recently upgrade to  25.2.3 because I have the same Issue as discussed here.
Just a precision, maybe it could change the behavior : I use IP static setup
As I'm not using DHCP, maybe the problem come with using IP Static ?
But as described here, If I reboot my Internet Router, OPenDTU lost the network connection and never recover it :( 
So, I have to restart OpdenDTU and If I'm not able to reboot it, I lost measurments :( 

thx for help
Rgds, SoniX

mhin76, 2025-02-21:
Hi SoniX
thanks for your input. I am using static IP set on my router. I will try DHCP over the weekend. On old version v24.6.29 is running just fine. Just a pity as I am missing on all nice new feature of latest firmware.
Kind regards, Martin

FranzToepfer, 2025-02-24:
Just to keep the topic warm - problem still not solved, even not with recent firmware.

I setup Grafana alerting, to get an information when DTU not in status ONLINE for longer than 10min. So I will be notified, when DTU for whatever reason not accessible anymore via WiFi in local network. 

Based on the poor monitoring approach I can tell you, my DTU goes offline multiple times a day. Sometimes coming back some minutes later, but especially since using latest firmware, communication stops ultimately. Need to remove power to enforce reboot, which directly brings DTU back into WiFi.

Bit annoying, didn't expect it to be that unstable.

dg3feh, 2025-04-03:
What kind of data or logs do u need?

SoniXtrA, 2025-04-03:
Hello

Yes because behavior still there !
Yesterday evening, I had to reboot my router and I forgot to reboot  OpenDTU ... 
Result : this morning the OpenDTU is still not wifi connected to the Router :(
I had to power Off / power On the OpenDTU and everything is Right
In other terms, when network is lost, there is no watchdog to restart the network layer of the OPenDTU.... :(

tbnobody, 2025-04-03:
I wrote it some time ago in this issue...

https://github.com/tbnobody/OpenDTU/issues/2366#issuecomment-2426486688

SoniXtrA, 2025-04-04:
Ok, I'll try to paste what's happen on serial console...
ATM, I just can provide a "testing life" graph of the OpentDTU...

[image]

Moments where green line is interrupted (= where orange line *result code* = 3) indicate a failed http status between grafana and the OpenDTU. They are in the same LAN.
But how do you manage a Wifi Failure ? Do you restart network layer when a network failure happens ?
What sort of watchdog is used ? which condition ?

Rgds :)

SoniXtrA, 2025-04-04:
Fetch inverter: 114491837774
13:55:17.513 > Queue size - NRF: 0 CMT: 1
13:55:17.533 > TX RealTimeRunData 865.00 MHz --> 15 91 83 77 74 80 19 37 56 80 0B 00 67 EF C8 A5 00 00 00 00 00 00 00 00 D7 F0 B5 
13:55:17.543 > Interrupt received
13:55:17.554 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 01 00 01 01 55 01 52 04 38 04 40 0E 65 0E 5B 00 03 3E | -57 dBm
13:55:17.574 > Interrupt received
13:55:17.583 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 02 99 B2 00 03 9F 12 01 FD 02 00 09 73 13 88 1B 2B F4 | -57 dBm
13:55:17.624 > Interrupt received
13:55:17.634 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 83 00 05 01 1F 03 E7 01 7C 00 01 5F C8 EB | -57 dBm
13:55:18.023 > RX Period End
13:55:18.023 > Success
13:55:22.432 > Fetch inverter: 114491837774
13:55:22.432 > Queue size - NRF: 0 CMT: 1
13:55:22.513 > TX RealTimeRunData 865.00 MHz --> 15 91 83 77 74 80 19 37 56 80 0B 00 67 EF C8 AA 00 00 00 00 00 00 00 00 27 B1 0B 
13:55:22.528 > Interrupt received
13:55:22.538 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 01 00 01 01 55 01 51 04 36 04 3E 0E 5F 0E 54 00 03 78 | -56 dBm
13:55:22.574 > Interrupt received
13:55:22.583 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 02 99 B3 00 03 9F 12 01 FE 02 00 09 6D 13 88 1B 1F DC | -57 dBm
13:55:22.621 > Interrupt received
13:55:22.629 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 83 00 05 01 1F 03 E7 01 7C 00 01 EC A4 34 | -56 dBm
13:55:22.980 > RX Period End
13:55:22.980 > Success
13:55:37.482 > Fetch inverter: 114491837774
13:55:37.482 > Queue size - NRF: 0 CMT: 1
13:55:37.493 > TX RealTimeRunData 865.00 MHz --> 15 91 83 77 74 80 19 37 56 80 0B 00 67 EF C8 B9 00 00 00 00 00 00 00 00 17 68 F1 
13:55:37.580 > Interrupt received
13:55:37.591 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 01 00 01 01 54 01 51 04 3C 04 43 0E 6D 0E 62 00 03 0A | -57 dBm
13:55:37.602 > Interrupt received
13:55:37.612 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 02 99 B4 00 03 9F 14 01 FF 02 02 09 68 13 87 1B 3A F1 | -57 dBm
13:55:37.626 > Interrupt received
13:55:37.635 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 83 00 05 01 21 03 E7 01 7C 00 01 FE 27 9B | -58 dBm
13:55:37.986 > RX Period End
13:55:37.986 > Success
13:55:42.498 > Fetch inverter: 114491837774
13:55:42.498 > Queue size - NRF: 0 CMT: 1
13:55:42.510 > TX RealTimeRunData 865.00 MHz --> 15 91 83 77 74 80 19 37 56 80 0B 00 67 EF C8 BE 00 00 00 00 00 00 00 00 27 4E E0 
13:55:42.533 > Interrupt received
13:55:42.543 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 01 00 01 01 54 01 51 04 3E 04 46 0E 75 0E 6A 00 03 1D | -57 dBm
13:55:42.606 > Interrupt received
13:55:42.618 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 02 99 B5 00 03 9F 14 02 00 02 02 09 67 13 87 1B 49 70 | -58 dBm
13:55:42.630 > Interrupt received
13:55:42.640 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 83 00 05 01 22 03 E7 01 7D 00 01 B7 9C 6B | -55 dBm
13:55:43.009 > RX Period End
13:55:43.009 > Success
13:55:47.515 > Fetch inverter: 114491837774
13:55:47.515 > Queue size - NRF: 0 CMT: 1

Websocket: [/livedata][1] disconnect
14:02:44.353 > Fetch inverter: 114491837774
14:02:44.353 > Queue size - NRF: 0 CMT: 1
14:02:44.436 > TX RealTimeRunData 865.00 MHz --> 15 91 83 77 74 80 19 37 56 80 0B 00 67 EF CA 64 00 00 00 00 00 00 00 00 88 0C D5 
14:02:44.967 > RX Period End
14:02:44.967 > All missing
14:02:44.967 > Nothing received, resend whole request
14:02:44.967 > TX RealTimeRunData 865.00 MHz --> 15 91 83 77 74 80 19 37 56 80 0B 00 67 EF CA 64 00 00 00 00 00 00 00 00 88 0C D5 
14:02:44.988 > Interrupt received
14:02:45.071 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 01 00 01 01 54 01 51 02 C1 02 C7 09 60 09 60 00 03 7C | -65 dBm
14:02:45.088 > Interrupt received
14:02:45.116 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 02 99 D7 00 03 9F 36 02 22 02 24 09 4D 13 88 11 B9 EB | -67 dBm
14:02:45.131 > Interrupt received
14:02:45.155 > RX 865.00 MHz --> 95 91 83 77 74 80 19 37 56 83 00 00 00 BE 03 E8 01 8A 00 03 BF 05 98 | -66 dBm
14:02:45.481 > RX Period End
14:02:45.481 > Success

dg3feh, 2025-04-05:
As far as I know the  console needs to be active in the web-interface to get the data from it. If I loose wifi connection I also loose the connection to the esp32. So how should I get the data from the console via the web interface?

tbnobody, 2025-04-05:
Thats why I asked for a log of the [serial console](https://www.opendtu.solar/firmware/howto/serial_console/)

SoniXtrA, 2025-04-05:
Please tell me how to do that, I only know the Web gui way....

ms49434, 2025-04-05:
> Please tell me how to do that, I only know the Web gui way....

Could you please click on the link provided by tbnobody and follow the steps to collect the output of the serial console.
Link reminder: [https://www.opendtu.solar/firmware/howto/serial_console](https://www.opendtu.solar/firmware/howto/serial_console)

SoniXtrA, 2025-04-14:
I'm back :)

The log file :
[putty.log](https://github.com/user-attachments/files/19733383/putty.log)

So, at the beginning, until line 443 everything is right (right and left led green).
At line 443 Wifi is disconnected (left led blink in green, right led is off)
At line 1526, because the device was still unable to recover WiFi connection, I unplugged et plugged the device to force a cold restart. (left led blink in green, right led is off)
After connection is Ok and everything is ok (right and left led green). 

Thx for your help !
Rgds

tbnobody, 2025-04-14:
Which OpenDTU version did you use for the test?

You recived a disconnect code `2`. Based on the ESP-IDF source, that means:
`WIFI_REASON_AUTH_EXPIRE                        = 2,`

But the disconnect happens during the phase when the "Admin-AP" is open. 

Do you have a chance to connect to the OpenDTU Access point when the ESP is disconnected from your router and make a screenshot of the "Info" --> "Network" view?

SoniXtrA, 2025-04-14:
Version in use : V25.2.3
Just an information, at line 443, wifi is disconnected because I rebooted my FAI router and this is the device that bring wifi ...
Not a simple wifi disconnection... 

I Will try to take the screenshot

tbnobody, 2025-04-15:
I have currently no clue whats going on. I tested it several times now. Turned off the AP, changed the password etc.... At my setup I see a complete different behavior....

When looking in your log
```
Admin AP remaining seconds: 110 / 180
WiFi disconnected: 2
Try reconnecting
[122400][E][WiFiClient.cpp:429] write(): fail on fd 49, errno: 113, "Software caused connection abort"
Network lost connection
Disconnected from MQTT.
Disconnect reason:TCP_DISCONNECTED
Admin AP remaining seconds: 120 / 180
Connecting to MQTT...
Disconnected from MQTT.
Disconnect reason:TCP_DISCONNECTED
Connecting to MQTT...
Disconnected from MQTT.
Disconnect reason:TCP_DISCONNECTED
Admin AP remaining seconds: 130 / 180
```

The `Try reconnecting` occours only one time.. 
In my installation I see the following:
```
Try reconnecting
Network lost connection
WiFi disconnected: 201
Try reconnecting
Network lost connection
WiFi disconnected: 201
Try reconnecting
Network lost connection
WiFi disconnected: 201
Try reconnecting
Network lost connection
WiFi disconnected: 201
Try reconnecting
Network lost connection
WiFi disconnected
```

So a reconnect every couple of seconds. This reconnect is triggered by the ESP Framework itself. There is no Code in OpenDTU which handles this reconnect. 

What's also strange in your setup is the Admin AP remaining seconds counting down:
```
Admin AP remaining seconds: 10 / 180
Admin AP remaining seconds: 20 / 180
Admin AP remaining seconds: 30 / 180
Admin AP remaining seconds: 40 / 180
Admin AP remaining seconds: 50 / 180
Admin AP remaining seconds: 60 / 180
Admin AP remaining seconds: 70 / 180
Admin AP remaining seconds: 80 / 180
Admin AP remaining seconds: 90 / 180
Admin AP remaining seconds: 100 / 180
Admin AP remaining seconds: 110 / 180
```

Based on the code, the countdown should stop immediatly if no valid IP is assigned to the interface. If the countdown goes upwards, it means that for some reason there is still a IP assigned. (In this case the screenshot would be very helpfull to check whether the IP is really assigned)

tbnobody, 2025-04-15:
@SoniXtrA which PIO Environment (Info --> System) are you using? I would then provide a debug firmware.

SoniXtrA, 2025-04-16:
So:
I use "generic_esp32" as PIO environment.
I just reboot my internet router (router provided by my Internet Access Provider) and during the reboot ;
1 - left led has started to blink in green
2 - My wifi@home has disappeared from available SSID
3 - I refreshed the list of available Wi-Fi networks on my recent smartphone (6 months old) and I never saw the AP Wifi network from the OpenDTU. So no screenshot.
4 - Router has finished to reboot and wifi has came back and OpenDTY never recover its network connection.

screenshot from my actual OpenDTU

[image]

[image]

[image]

[image]

Thanks for you help

tbnobody, 2025-04-16:
Thank you for the info...
I already saw yesterday that the integrated AP only gets online on a restart or when network settings are changed. If during the normal operation the wifi drops, no AP is opened (for security reasons).

Could you maybe try this debug firmware? You can just flash it using the web interface.. (A log of the serial console would be required and don't wonder, there is more output).
If the connection is not established and no reconnection attempt is done within 60 seconds the wifi interface will be reset and should begin with a proper connection again (hopefully).
So it should take a maximum of 90 seconds until the connection should working again.....

[generic_esp32.zip](https://github.com/user-attachments/files/19780992/generic_esp32.zip)

SoniXtrA, 2025-04-18:
Hi folks :)

So to summarize what I've done :
1 - I flashed the ESP32 Generic debug firmware (Static IP)
2 - I launched kitty on my COM3 / 155200 with saving everything in session to disk
3 - I rebooted my Internet Router
4 - I waited for Wifi (during the reboot of my internet router, I saw OpenDTU SSID)
5 - I noticed that OpenDTU couldn't reconnect to Wi-Fi, so I restarted OpenDTU, which then connected automatically to my Wi-Fi.

[kitty.log](https://github.com/user-attachments/files/19812516/kitty.log)

Hope this helps you 👍 
Rgds
SoniX

tbnobody, 2025-04-18:
Thanks for the log!!

This is the interessting part
```
con: 1, _connectTimeoutTimer: 1, _connectRedoTimer: 1
con: 1, _connectTimeoutTimer: 1, _connectRedoTimer: 1
Admin AP remaining seconds: 40 / 180
con: 1, _connectTimeoutTimer: 1, _connectRedoTimer: 1
WiFi disconnected: 2
Try reconnecting
[ 50203][E][WiFiClient.cpp:429] write(): fail on fd 49, errno: 113, "Software caused connection abort"
Network lost connection
Disconnected from MQTT. Reason: TCP_DISCONNECTED
con: 0, _connectTimeoutTimer: 1, _connectRedoTimer: 1
con: 0, _connectTimeoutTimer: 2, _connectRedoTimer: 2
con: 0, _connectTimeoutTimer: 3, _connectRedoTimer: 3
con: 0, _connectTimeoutTimer: 4, _connectRedoTimer: 4
con: 0, _connectTimeoutTimer: 5, _connectRedoTimer: 5
..
..
..
con: 0, _connectTimeoutTimer: 57, _connectRedoTimer: 57
con: 0, _connectTimeoutTimer: 58, _connectRedoTimer: 58
con: 0, _connectTimeoutTimer: 59, _connectRedoTimer: 59
con: 0, _connectTimeoutTimer: 60, _connectRedoTimer: 60
Wifi reconnect watchdog triggered... Resetting Wifi hardware
Start setting hostname...
Setting hostname done
Start configuring WiFi STA using new credentials
Configuring WiFi done
Start configuring WiFi static IP...
Configure IP done
con: 0, _connectTimeoutTimer: 61, _connectRedoTimer: 61
Disabling search for AP...
```

Problem now is, that the search for AP is disabled immediatly... Will doublecheck and parepare a new firmware

SoniXtrA, 2025-04-18:
I'm ready ... in starting blocs :)

tbnobody, 2025-04-18:
Lets try this one :)

[generic_esp32.zip](https://github.com/user-attachments/files/19812723/generic_esp32.zip)

mhin76, 2025-04-18:
happy to test it as well, but I think I need the generic_esp32.factory.bin (not sure what the difference is). Or can I try this one as well? I am runnig it on a Diamex solution.

tbnobody, 2025-04-18:
> not sure what the difference is
Please see https://www.opendtu.solar/firmware/#factory-binaries

You need the `.factory` file if you setup  a completly new device (or if something completly terrible went wrong and the ESP does not boot any more). Those files can only be uploaded using a USB cable and a flash tool.

mhin76, 2025-04-18:
> > not sure what the difference is
> Please see https://www.opendtu.solar/firmware/#factory-binaries
> 
> You need the `.factory` file if you setup  a completly new device (or if something completly terrible went wrong and the ESP does not boot any more). Those files can only be uploaded using a USB cable and a flash tool. 
> 
> If you have such a old firmware version, make sure to make a backup of all config files first.

Thanks for that info. That is how I got instructed to upload new firmware :-)

SoniXtrA, 2025-04-23:
Hi

So:
1 - I restarted my Internet Router
2 - Left green led is off
3 - I see the OpenDTU SSID on my smartphone wifi screen and my Home SSID is no more listed

[image]

4 - My internet router finish its restart procedure and my Home SSID appears in list but OpenDTU is still there

[image]

5 - The OpenDTU left led stays off... and I waited for 1 or 2 minutes but it stays in this state.
6 - I restart the OpenDTU Device and everything recover a normal state (2 leds green static)

The log of the test :
[kitty23.log](https://github.com/user-attachments/files/19863321/kitty23.log)

What I see :

### Synopsis of the Sequence

1. **Startup and Initialization:**
   - Starting OpenDTU.
   - Initializing the file system (FS) and reading the configuration.
   - DTU serial check: Using existing serial.

2. **Network Configuration:**
   - Network initialization successful.
   - Configuring WiFi in STA mode with new credentials.
   - WiFi configuration successful.

3. **Obtaining the IP Address:**
   - WiFi connection successful.
   - IP address obtained: `10.143.143.152`.
   - Network connected.

4. **MQTT Connection:**
   - Attempting to connect to MQTT.
   - MQTT connection successful.

5. **Access Point Status:**
   - Remaining time for the access point: 10 / 180 seconds.
   - Remaining time for the access point: 20 / 180 seconds.
   - Remaining time for the access point: 30 / 180 seconds.

6. **Network Interruption:**
   - WiFi disconnection reported.
   - Failed reconnection attempt, resulting in network loss.
   - Disconnected from MQTT with reason: `TCP_DISCONNECTED`.

7. **Device Reboot:**
   - Beginning the reboot process.
   - Resetting WiFi hardware.
   - Reconfiguring WiFi with new credentials.
   - New attempt to connect to MQTT after reboot.

### Relevant Log Excerpts

```
Starting OpenDTU
Initialize FS... done
Reading configuration... DTU serial check: Using existing serial done
Initialize Network... done
Start configuring WiFi STA using new credentials
Configuring WiFi done
WiFi connected
WiFi got ip: 10.143.143.152
Network connected
Connecting to MQTT...
Connected to MQTT.
Admin AP remaining seconds: 10 / 180
Admin AP remaining seconds: 20 / 180
Admin AP remaining seconds: 30 / 180
WiFi disconnected: 2
Try reconnecting
Network lost connection
Disconnected from MQTT. Reason: TCP_DISCONNECTED
Wifi reconnect watchdog triggered... Resetting Wifi hardware
Start setting hostname...
Setting hostname done
Start configuring WiFi STA using new credentials
Configuring WiFi done
```

Rgds :)

tbnobody, 2025-04-23:
Thank you @SoniXtrA for your work. (There is by the way no need for such a detailed analysis. Please don't put too much work into it. The console log is enough)

I wish I could reproduce this issue here more reliable then debuging would be much easier. Since this one time mentioned above I wasn't able to trigger this behavior again.

When looking at your log, this is the important part:
```
on: 0, _connectTimeoutTimer: 57, _connectRedoTimer: 57
con: 0, _connectTimeoutTimer: 58, _connectRedoTimer: 58
con: 0, _connectTimeoutTimer: 59, _connectRedoTimer: 59
con: 0, _connectTimeoutTimer: 60, _connectRedoTimer: 60
Wifi reconnect watchdog triggered... Resetting Wifi hardware
Start setting hostname...
Setting hostname done
Start configuring WiFi STA using new credentials
Configuring WiFi done
Start configuring WiFi static IP...
Configure IP done
con: 0, _connectTimeoutTimer: 1, _connectRedoTimer: 1
Wifi reconnect watchdog triggered... Resetting Wifi hardware
Start setting hostname...
Setting hostname done
Start configuring WiFi STA using new credentials
Configuring WiFi done
Start configuring WiFi static IP...
Configure IP done
con: 0, _connectTimeoutTimer: 1, _connectRedoTimer: 1
Wifi reconnect watchdog triggered... Resetting Wifi hardware
Start setting hostname...
Setting hostname done
Start configuring WiFi STA using new credentials
Configuring WiFi done
Start configuring WiFi static IP...
Configure IP done
con: 0, _connectTimeoutTimer: 1, _connectRedoTimer: 1
Wifi reconnect watchdog triggered... Resetting Wifi hardware
Start setting hostname...
Setting hostname done
Start configuring WiFi STA using new credentials
Configuring WiFi done
Start configuring WiFi static IP...
Configure IP done
con: 0, _connectTimeoutTimer: 1, _connectRedoTimer: 1
Wifi reconnect watchdog triggered... Resetting Wifi hardware
Start setting hostname...
Setting hostname done
Start configuring WiFi STA using new credentials
Configuring WiFi done
Start configuring WiFi static IP...
Configure IP done
```

One problem relies in my code. The `Wifi reconnect watchdog triggered... Resetting Wifi hardware` is triggered much to often. But on the other hand, somehow after `Configuring WiFi done` there should be a `Try reconnecting`. I absolutly don't understand why the ESP does not react on the `WiFi.begin()`.

So unfortunatly I have to prepare another firmware version with different behavior and also another approach to reset the wifi interface. Would be nice if you could test it again. (And again, the log is sufficient, please no further work)

[generic_esp32.zip](https://github.com/user-attachments/files/19874114/generic_esp32.zip)

SoniXtrA, 2025-04-25:
So , 
2 logs :
- 1st_log.txt where I rebooted the OpenDTU, rebooted my internet router and the left led has stayed OFF and when I closed the Putty terminal, the leds has started to blink ... so I didn't understand ! So I tried another sequence and catched a new log...

[1st_log.txt](https://github.com/user-attachments/files/19905569/1st_log.txt)

- 2nd_log.txt same scenario but I waited for a long time (more then 5 minutes) after WiFi has been recovered and finally, I saw the left led that has become ON with a static state..

[2nd_log.txt](https://github.com/user-attachments/files/19905570/2nd_log.txt)

So I don't think that the result is completly satisfying

Rgds
SoniX

mhin76, 2025-04-25:
Great!! That sounds very promising!! Looking forward to a new firmware! Thanks all for your effort and contribution! Much appreciated!

Magnus8513, 2025-04-25:
Awesome! Had this issue in my watch list and are happy to finally see it addressed.

Do you want more people to test this with the fw you provided in this thread before officially releasing it or are you confident to push this change to the official master already?

tbnobody, 2025-04-25:
If you like you can test of course.. But the serial log would be highly appreciated

SoniXtrA, 2025-05-12:
Hi 
Do you know when this issue Will be added into a futur firmware ?

About the "router takes longer then 120 seconds to reboot":  it is important to know that Europe hunts for devices power consumption... So, quickly, the manufacturers should serialize the différent start of embedded services to reduce this power consumption... So the devices Will take longer time to reboot...

Thx 👌👍

tbnobody, 2025-05-12:
The patch which you tested before is included in the release from yesterday.

ms1958, 2025-05-15:
Guten Morgen,
nun, in der Hoffnung, das es bei der v25.5.10 mit den Verbindungsabbrüchen behoben ist,
wenn über nacht mal das Wlan aus geht, habe ich diese Version mal testweise eingespielt.

Leider war die DTU Heute Morgen wieder nicht erreichbar. Ich musste sie Rebooten.
Und Ich hab in der OpenDTU schon DHCP abgeschaltet und auf feste IP gestellt ...

Somit wieder zurück zur v24.6.29.....
Mit dieser bleibt die verbindung bestehen,bzw wird wieder aufgebaut,
auch wenn der Router nachts mal neu bootet.
Gruss, Manfred
-----------------------
Good morning,
Well, in the hope that v25.5.10 will fix the connection drops when the Wi-Fi goes down overnight, I installed this version for testing.

Unfortunately, the DTU was unavailable again this morning. I had to reboot it.
And I've already disabled DHCP in OpenDTU and set it to a static IP...

So back to v24.6.29...
With this version, the connection remains, or rather, is re-established,
even if the router reboots at night.

Regards, Manfred

ms1958, 2025-05-16:
Good morning,
How do I get the log if I can no longer access OpenDTU in the morning and have to reboot?

6 minutes?
The connection is lost at 2:00 a.m.,
and even at 9:00 a.m. I can't access OpenDTU.

Why don't I have these problems with version 25.4.10?

tbnobody, 2025-05-16:
>  How do I get the log if I can no longer access OpenDTU in the morning and have to reboot?
https://www.opendtu.solar/firmware/howto/serial_console/

> Why don't I have these problems with version 25.4.10?
I don't know. There was nothing changed what could influence something like this. On the contrary, the reconnect will be triggered again if the wifi is unavailable but no reconnect was was done within 60 seconds.

But without any log it's not possible to determine what happens.

ms1958, 2025-05-16:
Note, the WLAN (via repeater) is still in operation, only the DHCP server is briefly gone for 3 minutes.
That's why I set the IP addresses and the gateway in opendu to static addresses.

stefan123t, 2025-05-17:
@tbnobody I have the hunch that the "disconnect reason" `2 WIFI_REASON_AUTH_EXPIRE` Authentication expired may only be handled as a "retry" by the ESP32 Arduino library, instead of a common "reconnect" and therefor maybe some of the steps necessary to reset OpenDTU into the right state may be skipped.

See my reference to the upstream implementation here: https://github.com/tbnobody/OpenDTU/issues/2202#issuecomment-2888615142

closed 2025-07-01 by github-actions[bot]