tbnobody/OpenDTU #2375: DPL regelt HMS700 nicht auf angestrebten Netzbezug

Frohbi83, 2024-10-25:
### What happened?

Hallo, ich habe einen Hoymiles HMS700 der DC-seitig von einem Akku versorgt wird. Diesen möchte ich mittels OpenDTU on battery so regeln das der Hausverbrauch gedeckt ist. Den Stromzähler habe ich erfolgreich eingerichtet und er liefert auch ständig Werte. Der HMS700 ist auch eingerichtet und die Werte können ausgelesen werden. DPL ist aktiviert und die gewünschten Schwellwerte sind eingestellt. Der Wechselrichter regelt jedoch nicht auf den Hausverbrauch, sondern speist mit Maximalleistung ins Hausnetz ein. Wenn ich manuell die Leistung begrenzen, regelt der Wechselrichter sofort auf den gewünschten Wert herunter.

### To Reproduce Bug

Siehe oben

### Expected Behavior

DPL regelt den Wechselrichter auf den angestrebten Netzbezug

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

22.10.2024

### What firmware variant (PIO Environment) are you using?

generic_esp32_4mb_no_ota

### Relevant log/trace output

```Shell
Websocket: [/livedata][7] disconnect
22:43:16.497 > Fetch inverter: 114483767513
22:43:16.582 > TX RealTimeRunData 868.00 MHz --> 15 83 76 75 13 80 11 36 24 80 0B 00 67 1C 02 E4 00 00 00 00 00 00 00 00 DF E5 29 
22:43:16.599 > Interrupt received
22:43:16.617 > RX 868.00 MHz --> 95 83 76 75 13 80 11 36 24 01 00 01 01 08 01 08 00 9C 00 A4 01 9C 01 B1 00 00 90 | -80 dBm
22:43:16.639 > Interrupt received
22:43:16.652 > RX 868.00 MHz --> 95 83 76 75 13 80 11 36 24 02 55 10 00 00 01 BC 00 29 00 2B 09 11 13 8A 03 21 DE | -80 dBm
22:43:16.664 > Interrupt received
22:43:16.671 > RX 868.00 MHz --> 95 83 76 75 13 80 11 36 24 83 00 00 00 22 03 E8 01 20 00 01 27 04 CC | -81 dBm
22:43:17.009 > RX Period End
22:43:17.019 > Success
22:43:21.428 > Fetch inverter: 114483767513
22:43:21.468 > TX RealTimeRunData 868.00 MHz --> 15 83 76 75 13 80 11 36 24 80 0B 00 67 1C 02 E9 00 00 00 00 00 00 00 00 4F BD EC 
22:43:21.576 > Interrupt received
22:43:21.719 > RX 868.00 MHz --> 95 83 76 75 13 80 11 36 24 01 00 01 01 08 01 08 00 9B 00 A4 01 9C 01 B1 00 00 97 | -81 dBm
22:43:21.820 > Interrupt received
22:43:21.855 > RX 868.00 MHz --> 95 83 76 75 13 80 11 36 24 02 55 10 00 00 01 BC 00 29 00 2B 09 11 13 89 03 20 DC | -79 dBm
22:43:21.873 > Interrupt received
22:43:21.892 > RX 868.00 MHz --> 95 83 76 75 13 80 11 36 24 83 00 00 00 22 03 E8 01 20 00 01 D8 C7 F0 | -81 dBm
22:43:21.980 > RX Period End
22:43:21.992 > Success
22:43:22.007 > [DPL::announceStatus] waiting for inverter device information to be available
22:43:23.983 > [PowerMeterHttpJson] New total: 9.00
22:43:26.540 > Fetch inverter: 114483767513
22:43:26.738 > TX RealTimeRunData 868.00 MHz --> 15 83 76 75 13 80 11 36 24 80 0B 00 67 1C 02 EE 00 00 00 00 00 00 00 00 7F 9B FD 
22:43:26.772 > Interrupt received
22:43:26.788 > Frame kaputt
22:43:26.801 > Interrupt received
22:43:26.814 > Frame kaputt
22:43:26.824 > Interrupt received
22:43:26.835 > RX 868.00 MHz --> 95 83 76 75 13 80 11 36 24 83 00 00 00 21 03 E7 01 20 00 01 B0 2D 7E | -81 dBm
22:43:27.043 > RX Period End
22:43:27.064 > Middle missing
22:43:27.084 > Request retransmit: 1
22:43:27.093 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 81 84 
22:43:27.103 > Interrupt received
22:43:27.112 > RX 868.00 MHz --> 95 83 76 75 13 80 11 36 24 01 00 01 01 08 01 08 00 96 00 9E 01 8F 01 A3 00 00 A1 | -81 dBm
22:43:27.130 > RX Period End
22:43:27.143 > Middle missing
22:43:27.151 > Request retransmit: 2
22:43:27.172 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 82 87 
22:43:27.182 > Interrupt received
22:43:27.194 > Frame kaputt
[... 132 lines cut ...]
22:43:52.219 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 82 87 
22:43:52.235 > Interrupt received
22:43:52.242 > Frame kaputt
22:43:52.271 > RX Period End
22:43:52.283 > Middle missing
22:43:52.301 > Request retransmit: 2
22:43:52.329 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 82 87 
22:43:52.333 > Interrupt received
22:43:52.336 > Frame kaputt
22:43:52.437 > RX Period End
22:43:52.450 > Middle missing
22:43:52.475 > Request retransmit: 2
22:43:52.502 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 82 87 
22:43:52.523 > Interrupt received
22:43:52.526 > RX 868.00 MHz --> 95 83 76 75 13 80 11 36 24 02 55 10 00 00 01 BC 00 29 00 2B 09 11 13 88 03 07 FA | -81 dBm
22:43:52.527 > RX Period End
22:43:52.537 > Success
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2024-10-25:
Es wäre sinnvoll diesen Issue im Repository von OpenDTU-on-Battery zu erstellen.

closed 2024-10-25 by tbnobody

Frohbi83, 2024-10-26:
Hallo, ich habe das Problem im Repository von OpenDTU on battery geschildert und wurde wieder hierher verwiesen. Das Problem wird wahrscheinlich dadurch verursacht, dass in den Wechselrichterinformationen "unbekanntes Model steht". Ich habe das Problem im offenen Issue #2374 nochmal mit angebracht.

schlimmchen, 2024-10-28:
Der Vollständigkeit halber, hier ist das Downstream-Issue: https://github.com/hoylabs/OpenDTU-OnBattery/issues/1348