tbnobody/OpenDTU #2378: [Request] Auto-Detect uploaded file type

schlimmchen, 2024-10-27:
### Is your feature request related to a problem? Please describe.

It happend to me again today that I was uploading a pin_mapping.json but forgot to set the drop-down menu to "pin_mapping.json", so I killed the config.json instead.

### Describe the solution you'd like

Auto-detect the kind of file being uploaded and write the respective destination.

Parse the uploaded file as JSON using ArduinoJson, closing #1097 in the process. Then inspect the JSON to find a known property which allows to assume it is a config.json or a pin_mapping.json or garbage.

### Additional context

I have this on my radar for a long time now and I can implement it myself, however, I had to write this down now that it happened to me again...

tbnobody, 2024-10-27:
Duplicate of #1097

tbnobody, 2024-10-27:
No need for opening the same issue again

tbnobody, 2024-10-27:
But this can be done in the webapp itself again... otherwise we would have to save the file using a temporary file name on the littlefs, parse it, and then either rename it to the right file or drop it.

schlimmchen, 2024-10-27:
> No need for opening the same issue again.... or we are closing https://github.com/tbnobody/OpenDTU/issues/1097.... but not both

The requests are (slightly) different, as far as I can tell. That's why I opened a new issue. #1097 asks to check the uploaded file to be valid JSON. This issue is asking to drop the selection for the target file name from the UI and to detect automatically, which one should be overwritten, based on the uploaded file's contents.

> otherwise we would have to save the file using a temporary file name on the littlefs, parse it, and then either rename it to the right file or drop it.

That sounds tedious. I did not look at the code, yet. I was assuming that the contents of the uploaded files can be streamed into ArduinoJson, so that we could inspect it in-flight, i.e., while it is in RAM.

stefan123t, 2024-10-27:
@schlimmchen yes you are right this is distinct from the other issue. Here it is about uploading the right file to overwrite the pin_mapping.json and not the config.json. 

Could we simply use the filename prefix `pin_mapping`/`config` to distinguish the two ? 
I tend to have plenty of those files on my disk, but I try to keep the beginning unchanged.

Saving it in a new file does not sound too tedious from my perspective, we could store it and fall back in case we have to restart within several minutes (though not initiated from the Settings UI by saving). 
Something like two boot-/config-environments, then you can double check validity of JSON before rejecting / activating it for the next boot.

tbnobody, 2024-10-28:
in the end I think it's a easy one in the web app. will have a look at it as I already rewriting the whole config file management view.

schlimmchen, 2024-10-28:
> Could we simply use the filename prefix pin_mapping/config to distinguish the two ?

Please don't. Also, I am not sure that the file name is always preserved or accessible to JavaScript. At least the file path of upload fields are often obscured for privacy.

> in the end I think it's a easy one in the web app.

Yeah, that's probably true. Thanks for looking into it.

tbnobody, 2024-11-07:
Implemented in 8019eaf18218a4efb52995cb1bf861ba7e44450e

closed 2024-11-07 by tbnobody