tbnobody/OpenDTU #2400: ESP32 does not reconnect to Wifi

LPG-PG, 2024-11-09:
### What happened?

After an indefinite period of time, the ESP32 module loses the connection to the WiFi. No new DHCP lease is obtained, so neither web, API, MQTT nor ping works.

### To Reproduce Bug

It happens ramdomly between 1 and 5 days

### Expected Behavior

OpenDTU has to be reboot to fix it temporary 

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v24.10.15

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Anything else?

I checked some parts of the code and I'm not sure, if there is a bug in NetworkSettings.cpp -> NetworkSettingsClass::applyConfig()
line 288ff
<pre>
WiFi.begin(
            Configuration.get().WiFi.Ssid,
            Configuration.get().WiFi.Password,
            WIFI_ALL_CHANNEL_SCAN);
</pre>
I mean, that Wifi.begin is called with Channel 1 because the enum WIFI_ALL_CHANNEL_SCAN value is 1.

The declaration of the called function is:
<pre>
wl_status_t begin(char* ssid, char *passphrase = NULL, int32_t channel = 0, const uint8_t* bssid = NULL, bool connect = true);
</pre>
So the third paramter is the channel, which is in this case 1.

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Subnum12, 2024-12-18:
I have same issue. 
I did try wifi reconnect while console log was enabled. For whatever reason the console gets disconnected while wifi is disconnect. However, that's the last output even I could not recreate the issue with manual wifi reconnect.

```
[2094941][E][WiFiClient.cpp:429] write(): fail on fd 48, errno: 113, "Software caused connection abort"
Disconnected from MQTT.
Disconnect reason:TCP_DISCONNECTED
Network lost co˫▒ѥ▒▒5
                     Websocket: [/livedata][5] disconnect
```

stefan123t, 2024-12-19:
@tbnobody I think @LPG-PG is right, we maybe somehow using the wrong signature for the WiFi.begin() method:

https://github.com/espressif/arduino-esp32/blob/352705e6077de4f0575324efa234814e823b9fb8/libraries/WiFi/src/WiFiSTA.cpp#L232-L240

The actual begin() method obviously seems to have the signature as @LPG-PG reported:

https://github.com/espressif/arduino-esp32/blob/352705e6077de4f0575324efa234814e823b9fb8/libraries/WiFi/src/WiFiSTA.cpp#L83-L93

LPG-PG, 2024-12-21:
> You are right. at least the setting of `WIFI_ALL_CHANNEL_SCAN` is wrong at this place... did you try if it works better if you remove this parameter?

I haven't tried it yet.

pi-amp, 2025-01-07:
same here for me. Really anoying when you're away from the DTU... Hope it gets fixed !

karamo, 2025-01-09:
Ich habe vor 2 Wochen von der 24.5.6 auf 24.11.7 upgedatet.
Die Folge waren permanente Abbrüche ver WiFi-Verbindung.
Bin wieder zurück auf die bei mir funktionierende 24.5.6

I updated from 24.5.6 to 24.11.7 2 weeks ago.
The result was permanent interruptions in the WiFi connection.
I'm back to 24.5.6, which works for me.

ms1958, 2025-01-17:
Guten Tag,
jetzt ist ja die Version 25.1.14 draussen. ist das Problem damit behoben ?
Ich bin solange auf die v24.6.29 zurück gegangen, und hatte bis dato keine verbindungsabbrüche mehr.

Subnum12, 2025-01-17:
habe mit v25.1.14 seit Mittowch keine Abbrüche mehr. -> issue can be closed.

andreas-bulling, 2025-01-19:
I have upgraded to 25.1.14 yesterday and today OpenDTU doesn't connect to Wifi anymore. No changes otherwise, in particular I haven't moved the device or anyting.

Subnum12, 2025-01-19:
actually after almost a week running without any issue I got same error yesterday evening again.

andreas-bulling, 2025-01-19:
I think they must have changed something on the wifi stack - or maybe wireless transmission power? The OpenDTU device is located in the basement. No wifi connection problems whatsoever since more than a year now.

With the new firmware it didn't connect. I now brought it upstairs, closer to an AP and it connects without problems.

Back in the basement it also connects now for some reason.

ms1958, 2025-02-01:
Guten Morgen, 
bin ich nach einem Test von 2 Wochen mit [v25.1.14] wieder zurück zur [v24.6.29]
Auch mit [v25.1.14] habe ich schon 5x verbindungsverlust gehabt.
Dann muss ich jedesmal in die Garage die DTU Booten.
Nachts Bootet mein Router neu, dabei ist DHCP mal für 4min weg.
ich habe in der DTU schon eine feste IP vergeben, aber das nutzt nix.
Mit der [v24.6.29] habe ich jedenfalls diese Probleme nicht, Die verbindung bleibt über Monate bestehen.
Gruss, Manfred

Heiko-L, 2025-02-12:
My last working version is also v24.6.29.
In all versions after that the whole device freezes if WiFi is lost.
Last output before freeze:
> WiFi disconnected
> Try reconnecting
> [7163522][E][WiFiClient.cpp:429] write(): fail on fd 49, errno: 113, "Software caused connection abort"
> Disconnected from MQTT.
> Disconnect reason:TCP_DISCONNECTED
> Network lost connection

LPG-PG, 2025-02-16:
I changed the ESP32 Module and flashed the v25.2.3.
Since then i have a continuous and stable WiFi connection over 8 days. Additionally the RSSI value is 30dBm better (old -85dBm, now -55dBm).

I think a big part of my problem was probably caused by the ESP module itself.

Heiko-L, 2025-02-17:
Have you turned off your WLAN AP(s) to check what happens when OpenDTU gets disconnected?
I only have ESP32-D0WD-V3s, so I can't check if other hardware revisions cause this problem.

kopierschnitte, 2025-02-20:
Same observation here. Like @Heiko-L said, the ESP32 stalls if Wifi is lost. Happened last time when my Ubiquiti gear performed an update and switched off the APs for about 30sec.

Maybe a watchdog or simple reboot if there's no connection for a period of time would do the trick?

LPG-PG, 2025-02-22:
> Have you turned off your WLAN AP(s) to check what happens when OpenDTU gets disconnected? I only have ESP32-D0WD-V3s, so I can't check if other hardware revisions cause this problem.

I did some tests today. After turning the AP off and on again, the ESP module reconnected to the WiFi. I also tried using a Mash WiFi to turn off individual APs and all APs at the same time. The tests were repeated several times with the same result. I use Mikrotik devices.

kopierschnitte, 2025-03-16:
Unfortunately, my ESP32 isn't that "flexible" together with my Ubiquiti setup.
A regular maintenance of the network controller (which disconnects all WiFi clients for a few seconds) is sufficient to knock OpenDTU out of my network. Even if I just reboot the particular AP the ESP32 is connected with, OpenDTU fails to roam to a different one (which is clearly "visible" but on a different channel) and remains disconnected.

I suspect that it's a little bit tricky because other (not all!) ESP-based projects suffer the same problem.

kopierschnitte, 2025-04-16:
Bump ... problem still exists in v25.4.10

Last night, one of my APs did an update and went offline for about 5min. Other APs were clearly in range (other ESPs did roam) but OpenDTU went and stood offline until power cycle :-(

jstammi, 2025-04-16:
What sw are the successfully roaming esps running? Maybe comparison helps
improving...

Heiko-L, 2025-04-16:
The problem must be somewhere here:
https://github.com/tbnobody/OpenDTU/compare/v24.6.29...v24.8.1

tbnobody, 2025-04-16:
Please see #2366

jstammi, 2025-04-16:
> Please see [#2366](https://github.com/tbnobody/OpenDTU/issues/2366)

I am aware. But there the approach to track down by log/behaviour analysis.

I will give a try to check the code in AquaMQTT on how the wifi is setup there (said to be working with changing AP) compared to the code here.

kopierschnitte, 2025-05-11:
v25.5.10 seems to be more stable at first sight (tested by forcing a reconnect). Fingers crossed...

stefan123t, 2025-05-14:
@jstammi the code in AquaMQTT seems to be pretty straigth forward and based on an almost zero-trust model 😜 

https://github.com/tspopp/AquaMQTT/blob/main/AquaMQTT/src/handler/Wifi.cpp

<details><summary>Details</summary>
<p>

```c++
#include "handler/Wifi.h"

#include "config/Configuration.h"

namespace aquamqtt
{

bool WifiHandler::mConnectedToWifiWithValidIpAddress = false;

WifiHandler::WifiHandler() : mLastCheck(0)
{
}

void WifiHandler::setup()
{
    WiFiClass::mode(WIFI_STA);

    // we don't trust the auto reconnect routine, as it seems there are edge cases where it does not work
    WiFi.setAutoReconnect(false);

    // we trust the wifi callbacks to determine if we are properly connected or disconnected
    WiFi.onEvent(wifiCallback);

    // begin a single wifi session
    WiFi.begin(aquamqtt::config::ssid, aquamqtt::config::psk);

    // perform the next wifi check in config::WIFI_RECONNECT_CYCLE_S
    mLastCheck = millis();
}

void WifiHandler::loop()
{
    if ((millis() - mLastCheck) >= (config::WIFI_RECONNECT_CYCLE_S * 1000))
    {
        mLastCheck = millis();

        // we don't trust WiFi.isConnected() or WiFi.status() == WL_CONNECTED, since it is suspected to be unreliable
        if (!mConnectedToWifiWithValidIpAddress)
        {
            Serial.println("[wifi] attempting reconnect");
            WiFi.disconnect();
            WiFi.reconnect();
        }
    }
}

void WifiHandler::wifiCallback(WiFiEvent_t event)
{
    Serial.print("[wifi] event: ");
    Serial.println(WiFi.eventName(event));

    switch (event)
    {
        case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
        case ARDUINO_EVENT_WIFI_STA_LOST_IP:
            // if we lost connection or ip address, we wil enforce a reconnect within the next cycle
            mConnectedToWifiWithValidIpAddress = false;
            break;
        case ARDUINO_EVENT_WIFI_STA_GOT_IP:
        case ARDUINO_EVENT_WIFI_STA_GOT_IP6:
            // if we got connection and therefore a valid ip address we have a valid connection
            Serial.print("[wifi] ip address: ");
            Serial.println(WiFi.localIP().toString().c_str());
            mConnectedToWifiWithValidIpAddress = true;
            break;
        default:
            break;
    }
}

}  // namespace aquamqtt
``` 

</p>
</details>

ms1958, 2025-05-15:
Guten Morgen,
nun, in der Hoffnung, das es bei der v25.5.10 mit den Verbindungsabbrüchen behoben ist, 
wenn über nacht mal das Wlan aus geht, habe ich diese Version mal testweise eingespielt.

Leider war die DTU Heute Morgen wieder nicht erreichbar. Ich musste sie Rebooten.
Und Ich hab in der OpenDTU schon DHCP abgeschaltet und auf feste IP gestellt ...

Somit wieder zurück zur v24.6.29..... 
Mit dieser bleibt die verbindung bestehen,bzw wird wieder aufgebaut,
auch wenn der Router nachts mal neu bootet.

Gruss, Manfred
---------------------
Good morning,
Well, in the hope that v25.5.10 will fix the connection drops when the Wi-Fi goes down overnight, I installed this version for testing.

Unfortunately, the DTU was unavailable again this morning. I had to reboot it.
And I've already disabled DHCP in OpenDTU and set it to a static IP...

So back to v24.6.29...
With this version, the connection remains, or rather, is re-established,
even if the router reboots at night.

Regards, Manfred

stefan123t, 2025-06-11:
Yes see the WiFi callback handler here https://github.com/tbnobody/OpenDTU/issues/2400#issuecomment-2881784534

ms1958, 2025-06-18:
seit ich in openDTU mDSN deaktiviert habe, habe ich keine wiederverbindungsprobleme mehr. (seit einem Monat) 

Since I disabled mDSN in openDTU, I haven't had any reconnection issues (for a month now).
Manfred

universal-dilettant, 2025-07-17:
I have been experiencing this Wifi disconnct for several month, over several versions (sorry, no debug log, but >12 month minimum). Last event some minutes ago with release `generic-esp32 v25.4.10`. `mDNS` was disabled already.

Intervals between 2 disconnects to be counted in months, last FW upgrade to `25.4.10` was at May 6th, which makes it difficult to reproduce and/or debug (and, of course, the openDTU is not on my desk but in the basement). Now I have upgraded to `25.5.10`, hoping again that it will fix this quite annoying issue - hope dies last - and looking in parallel for a small timer to interrupt power supply from time to time.

stefan123t, 2025-07-25:
Please note that the first of these three WiFi reconnect issues #2202 has been closed as solved / fixed by the OP SteffenR72 recently / last week. The second #2366 by dg3feh has been closed by the github bot as not planned and only this #2400 is still open.

IMHO, we should ask the OP's of the two unresolved issues @LPG-PG and @dg3feh for their results with the current release version 2025.05.10

---

@tbnobody would you want to work around the shortcomings of the Arduino WiFi Reconnect implementation as in the AquaMQTT WiFi handler ?

https://github.com/tbnobody/OpenDTU/issues/2400#issuecomment-2881784534

>     // we don't trust the auto reconnect routine, as it seems there are edge cases where it does not work
>     // we trust the wifi callbacks to determine if we are properly connected or disconnected
>     // perform the next wifi check in config::WIFI_RECONNECT_CYCLE_S
>         // we don't trust WiFi.isConnected() or WiFi.status() == WL_CONNECTED, since it is suspected to be unreliable
>             // if we lost connection or ip address, we will enforce a reconnect within the next cycle
>             // if we got connection and therefore a valid ip address we have a valid connection

or do you prefer the issue to be fixed upstream in the Arduino / Espressif libraries we are using ?

> https://github.com/tbnobody/OpenDTU/issues/2366#issuecomment-2888643891
> I have the hunch that the "disconnect reason" `2 WIFI_REASON_AUTH_EXPIRE` Authentication expired may only be handled as a "retry" by the ESP32 Arduino library, instead of a common "reconnect" and therefor maybe some of the steps necessary to reset OpenDTU into the right state may be skipped.

See my reference to the upstream implementation here: 

> https://github.com/tbnobody/OpenDTU/issues/2202#issuecomment-2888615142
> We can see
> `WiFi disconnected: -47`
> `WiFi disconnected: 2`
> `WiFi disconnected: 209`
>
> This translates according to the esp-idf wifi [wifi_err_reason_t](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_wifi.html#_CPPv417wifi_err_reason_t) disconnect reason types defined in [esp_wifi_types_generic.h](https://github.com/espressif/esp-idf/blob/f9765d03162efd48feece14a4949de66356e00ae/components/esp_wifi/include/esp_wifi_types_generic.h#L107-L177) to the following:
> `2 WIFI_REASON_AUTH_EXPIRE Authentication expired`
> `209 WIFI_REASON_SA_QUERY_TIMEOUT SA query timeout`
> `47 WIFI_REASON_AP_INITIATED AP initiated disassociation`
> Wi-Fi disconnection reason codes.
> These reason codes are used to indicate the cause of disconnection.
> I do not know what negative sign in front of the 47 reason code may signify ?

stefan123t, 2025-08-25:
@Heiko-L can you contribute something more substantial to the issue instead if not solved only ?

I have given several indicators like disconnect reason and possible solutions for eventually handling these. I would appreciate it if you could provide some constructive feedback.