tbnobody/OpenDTU #2401: openDTU forgets the Inverter settings (HM1500) on reboot

wpoet, 2024-11-10:
### What happened?

The openDTU versions of 24.x all have this Problem: I can set my inverters HM1500. They are handled correctly and showing their data. Only after a reboot or power on, the settings are replaced by "Unknown". No Life data are shown. When i go to the inverter settings and start to edit an inverter, a simple "save'" will bring back the Inverter type and life data will be shown again. This happens with versions 24.x. I reverted to version 23.4.6 where this bug does not appear. The versions 24.x show the inverter with a "4T' at the end of the name.

### To Reproduce Bug

Set inverters. Life data are shown. Reboot. Inverters are Unknown.

### Expected Behavior

Settings should survive a reboot.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

24.x

### What firmware variant (PIO Environment) are you using?

Not available in my Info 

### Relevant log/trace output

```Shell
N/A
```

### Anything else?

N/A

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2024-11-10:
Versions 23.x and 24.x are in worst case one year apart. Please provide more detailed informations. Especially the last few version should not make any troubles here as they are commonly used.

wpoet, 2024-11-14:
Yes, I checked, indeed, it's the NRF24 in nodemcu_esp32.json. https://github.com/tbnobody/OpenDTU/blob/master/docs/DeviceProfiles/nodemcu_esp32.json. I was not aware that the pins specified in platformio.json are actually not transferred to the device. Thank you for the tip.

stefan123t, 2024-11-14:
@tbnobody is this `platformio.json` just a remnant of the old way of configuring it before `pin_mapping.json` 
or should this be packaged as a overridde next to the default `pin_mapping.json` into the LittleFS 
in case someone is building the firmware from scratch ?

tbnobody, 2024-11-14:
> I was not aware that the pins specified in platformio.json are actually not transferred to the device. Thank you for the tip.

There is no such file.

And of course the pins specified in the `platformio.ini` are still working. But you have to make sure to adjust and compile the correct environment

closed 2024-11-14 by tbnobody

wpoet, 2024-11-14:
Sorry, of course, it's platformio.ini, not .json. When i choose the vscode environment as in the documentation, pin settings are ignored. Anyway, it's working now by loading the pins through the WebUI. Thank you all for your input.