tbnobody/OpenDTU #2412: Kommunikationsproblem DTU - HMS-450

HaWy5611, 2024-11-16:
### What happened?

Ich habe 3 HMS-450 und habe diese mit der DTU verbunden im Menu Einstellungen, Wechselrichter.
2 Stk funktionieren, unter Typ kommt "HMS-500-1T v2", beim dritten kommt "Unknown"
Dabei ist auffällig dass bei den ersten beiden besteht die Serienummer nur aus Zahlen, beim dritten aus Zahlen und Buchstaben
Die Eingabe "1400A01FF23F" übernommen wird: "1400a01ff23f"
Bei der Live-Ansicht erscheinen nur die ersten beiden.

### To Reproduce Bug

Alle Wechselrichter sind am Netz 230V angeschlossen. Der fehlerhafte HMS-450 kann ins Netz einspeisen wenn man Gleichspannung ca. 48V anlegt.

### Expected Behavior

Ich erwarte, dass die Verbindung zwischen DTU und HMS-450 hergestellt wird bei allen drei Hoymiles HMS-450

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v24.11.7

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

spcqike, 2024-11-18:
ggf. eine neue, unbekannte Hardwareteilenummer?

was steht denn beim dritten in den Wechselrichter Informationen?
https://github.com/tbnobody/OpenDTU/issues/2374#issuecomment-2439530550

HaWy5611, 2024-11-18:
Auf dem Typenschild steht: sihe Bild
Model: HMS-450-1T
[image]

Vilen Dank

HaWy5611, 2024-11-18:
Guten Abend Ich sende Ihnen hier ein Bild vom Typenschild Ich weiss nicht ob ich mich sonst mit dem Umrichter verbinden kann Grüsse und Vielen Dank Hans Wyss

#2374 (comment)

stefan123t, 2024-11-19:
Hallo @HaWy5611,

Welche Version von OpenDTU verwendest Du ?
Hast Du die Version v24.3.15 oder neuer installiert ?

Deine Modellnummer 1400 ist mE bisher unbekannt. 
@tbnobody evtl muss diese noch zur Liste der bekannten Modellnummern #714 und zu einer künftigen OpenDTU version hinzugefügt werden ?

stefan123t, 2024-11-19:
@HaWy5611 kannst Du mal versuchen die Seriennummer als 1403A01FF23F einzugeben ?

stefan123t, 2024-11-19:
@HaWy5611 danke fürs Update und den erneuten Versuch.

Probiere bitte mal `1124A01FF23F` das Präfix `1124` bezeichnet einen HMS/HMT mit einem DC Eingang sein und sollte dann als Wechselrichter von der OpenDTU erkannt werden.

HaWy5611, 2024-11-19:
Guten Abend
Zur Zeit bin ich unterwegs, ich versuche dies gerne nächste Woche, so ab Dienstag 26.11
 
Vielen Dank und Grüsse
HaWy5611

Gesendet: Dienstag, 19. November 2024 um 06:33

stefan123t, 2024-11-21:
Laut der Dokumentation in der [lib/Hoymiles/src/inverters/README.md](https://github.com/tbnobody/OpenDTU/blob/3dc70ab40aade8b7eb9ed9a1c6605ca326299d18/lib/Hoymiles/src/inverters/README.md?plain=1) gibt es zwei gültige Präfixe für HMS-300..500-1T bzw. HMS-500-1T v2.

| Class         | Models                      | Serial range     |
| --------------| --------------------------- | ---------------- |
| HMS_1CH       | HMS-300/350/400/450/500-1T  | `1124`             |
| HMS_1CHv2     | HMS-500-1T v2               | `1125`             |

Ich vermute fast Du hast ein v2 oder gar ein v3 Modell und wir können den Source Code und die o.a. Tabelle mit Deinem Präfix `1400` ergänzen.

Hier der Code für die Methode `isValidSerial` in der v1

https://github.com/tbnobody/OpenDTU/blob/3dc70ab40aade8b7eb9ed9a1c6605ca326299d18/lib/Hoymiles/src/inverters/HMS_1CH.cpp#L34-L39

und hier der entsprechende Code für v2

https://github.com/tbnobody/OpenDTU/blob/3dc70ab40aade8b7eb9ed9a1c6605ca326299d18/lib/Hoymiles/src/inverters/HMS_1CHv2.cpp#L34-L39

stefan123t, 2024-11-28:
@HaWy5611 hast Du mal Deine Seriennummer mit 1124 oder 1125 anstelle der 1400 an den ersten vier Stellen probiert ?

HaWy5611, 2024-11-30:
Hallo stefan123t
Ich bin wieder zurück und habe die Serienummern mit 1124 und 1125 ausprobiert und siehe da die DTU hat den Wechselrichter erkann.
Mit 1124 war der genaue Typ mit HMS-300/350/400/450/500-1T angegeben und der Balken in der Live-Ansicht ist gelb.
Mit 1125 ist dere Wechselrichter Typ richtig angegeben mit HMS-500-1T v2 wie die andern zwei. 
Ich lasse somit die Seierienummer Einstellung bei 1125A01FF23F
 
Herzlichen Dank und Grüsse
HaWy5611

Gesendet: Donnerstag, 28. November 2024 um 23:27

stefan123t, 2024-11-30:
@HaWy5611 bitte nochmal kontrollieren ob die Werte alle plausibel sind. Die v1 und die v2 haben mW eine andere Reihenfolge und andere Multiplikatoren/Divisoren für die Werte.

Ggf auch mal einen Auszug aus der Serial Console posten damit @tbnobody den WR Typ mit 1400 korrekt zuordnen kann. Wir brauchen dazu neben dem Serial Log auch einen Screenshot / Kopie der Seite **Show Inverter Info** mit all den Details.

HaWy5611, 2024-11-30:
Hallo tbnobody
Beiliegend einige Screenshot
Der Wechselrichter mit der Serienr. 1125A01FF23F ist derjenige mit der Korrigierten Serienr. Uhrsprünglich: 1400A01FF23F
Frage wie komme ich zu einem Auszug aus der Serial Console
 
Grüsse
 

Gesendet: Samstag, 30. November 2024 um 16:53

schlimmchen, 2024-11-30:
@HaWy5611 Du hast vermutlich deine Screenshots an eine E-Mail angehängt? Du müsstest dir bitte die Mühe machen, dieses Issue im Browser zu öffnen um die Screenshots hochzuladen, da Anhänge in E-Mail-Antworten offensichtlich nicht in deinen Kommentar im Issue eingearbeitet werden.

stefan123t, 2024-11-30:
@HaWy5611 bitte tagsüber mal unter Info > Console einige erfolgreiche RealTimeRunDebug und evtl. auch ein paar SystemConfigPara Kommandos mit TX und RX Paketen abwarten und dann die Ausgabe auf der Github Webseite posten.

Das Symbol [/]code (Slash commands) erlaubt es den Quelltext aus dem Serial Log ordentlich zu formatieren. 
Das kann man sich auch ggf. mit Preview vorher ansehen. 
Deine Beiträge kannst Du auch im Nachhinein mit Edit nochmal überarbeiten bis es leserlich aussieht.

HaWy5611, 2024-12-01:
also hier die Screenshot
[image]
[image]
[image]
[image]
[image]
hoffe das geht so

Im weiteren habe ich die Kommando Ausgaben:
11:43:48.657 > RX 865.00 MHz --> 95 94 40 91 94 80 18 49 72 02 14 E1 00 00 00 00 07 C4 00 00 09 3F 13 86 00 7C 0C | -16 dBm
11:43:48.709 > Interrupt received
11:43:48.770 > RX 865.00 MHz --> 95 94 40 91 94 80 18 49 72 83 00 CD 00 05 02 06 00 EF 00 05 1D CF 90 | -16 dBm
11:43:48.904 > RX Period End
11:43:48.904 > Success
11:43:49.046 > Websocket: [/livedata][6] disconnect
11:43:49.189 > Websocket: [/livedata][7] connect
11:43:53.358 > Fetch inverter: 1125A01FF23F
11:43:53.495 > TX RealTimeRunData 865.00 MHz --> 15 A0 1F F2 3F 80 18 49 72 80 0B 00 67 4C 3D E8 00 00 00 00 00 00 00 00 11 81 21 
11:43:53.785 > Interrupt received
11:43:53.834 > RX 865.00 MHz --> 95 A0 1F F2 3F 80 18 49 72 01 00 01 01 FB 00 00 00 1C 00 00 00 8E 00 00 00 00 2C | -21 dBm
11:43:53.883 > Interrupt received
11:43:53.934 > RX 865.00 MHz --> 95 A0 1F F2 3F 80 18 49 72 02 13 9D 00 00 00 00 08 89 00 00 09 36 13 86 00 87 64 | -19 dBm
11:43:53.981 > Interrupt received
11:43:54.039 > RX 865.00 MHz --> 95 A0 1F F2 3F 80 18 49 72 83 00 00 00 05 03 E8 00 F5 00 1C 86 03 45 | -21 dBm
11:43:54.104 > RX Period End
11:43:54.104 > Success
11:43:54.150 > Admin AP remaining seconds: 80 / 180
11:43:58.359 > Fetch inverter: 112594409037
11:43:58.421 > TX RealTimeRunData 865.00 MHz --> 15 94 40 90 37 80 18 49 72 80 0B 00 67 4C 3D ED 00 00 00 00 00 00 00 00 41 BE 4A 
11:43:58.477 > Interrupt received
11:43:58.530 > RX 865.00 MHz --> 95 94 40 90 37 80 18 49 72 01 00 01 01 FB 00 00 00 21 00 00 00 A8 00 00 00 00 36 | -24 dBm
11:43:58.577 > Interrupt received
11:43:58.631 > RX 865.00 MHz --> 95 94 40 90 37 80 18 49 72 02 16 BD 00 00 00 00 07 CA 00 00 09 35 13 86 00 9F 17 | -24 dBm
11:43:58.700 > Interrupt received
11:43:58.764 > RX 865.00 MHz --> 95 94 40 90 37 80 18 49 72 83 00 CB 00 06 02 69 00 EE 00 06 D9 B0 E1 | -24 dBm
11:43:58.906 > RX Period End

und noch ein Editor-File
[Console_Debug_1_12_24.txt](https://github.com/user-attachments/files/17968667/Console_Debug_1_12_24.txt)
Vielen Dank und Grüsse

tbnobody, 2024-12-01:
Going to open it again until its implemented

closed 2025-04-18 by github-actions[bot]