tbnobody/OpenDTU #2419: OpenDTU reports Hoymiles HM400 as reachable although inverter is off

mibcat, 2024-11-18:
### What happened?

OpenDTU is sending **solar/serial-number/status/reachable** equal 1 each 10s although the inverter is not supplied via DC input.
[image]

### To Reproduce Bug

1. turn off inverter
2. start OpenDTU

### Expected Behavior

If the inverter is not powered the **reachable** signal should be zero.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

3dc70ab

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Relevant log/trace output

```Shell
16:56:17.691 > All missing
16:56:17.691 > Nothing received, resend whole request
16:56:17.691 > TX ActivePowerControl Channel: 3 --> 51 83 81 91 93 80 13 56 88 81 0B 00 00 00 00 00 60 01 F7 
16:56:18.556 > Received MQTT message on topic: solar/112183819193/cmd/limit_nonpersistent_absolute
16:56:18.556 > Limit Non-Persistent: 0.0 W
16:56:19.694 > RX Period End
16:56:19.694 > All missing
16:56:19.694 > Nothing received, resend count exeeded
16:56:19.811 > TX ActivePowerControl Channel: 23 --> 51 83 81 91 93 80 13 56 88 81 0B 00 00 00 00 00 60 01 F7 
16:56:21.861 > RX Period End
16:56:21.861 > All missing
16:56:21.861 > Nothing received, resend whole request
16:56:21.861 > TX ActivePowerControl Channel: 40 --> 51 83 81 91 93 80 13 56 88 81 0B 00 00 00 00 00 60 01 F7 
16:56:23.574 > Received MQTT message on topic: solar/112183819193/cmd/limit_nonpersistent_absolute
16:56:23.574 > Limit Non-Persistent: 0.0 W
16:56:23.908 > RX Period End
16:56:23.908 > All missing
16:56:23.908 > Nothing received, resend whole request
16:56:23.908 > TX ActivePowerControl Channel: 61 --> 51 83 81 91 93 80 13 56 88 81 0B 00 00 00 00 00 60 01 F7 
16:56:25.955 > RX Period End
16:56:25.955 > All missing
16:56:25.955 > Nothing received, resend whole request
16:56:25.955 > TX ActivePowerControl Channel: 75 --> 51 83 81 91 93 80 13 56 88 81 0B 00 00 00 00 00 60 01 F7 
16:56:28.002 > RX Period End
16:56:28.002 > All missing
16:56:28.002 > Nothing received, resend whole request
16:56:28.002 > TX ActivePowerControl Channel: 3 --> 51 83 81 91 93 80 13 56 88 81 0B 00 00 00 00 00 60 01 F7 
16:56:28.595 > Received MQTT message on topic: solar/112183819193/cmd/limit_nonpersistent_absolute
16:56:28.595 > Limit Non-Persistent: 0.0 W
16:56:30.004 > RX Period End
16:56:30.004 > All missing
16:56:30.004 > Nothing received, resend count exeeded
16:56:30.153 > TX ActivePowerControl Channel: 23 --> 51 83 81 91 93 80 13 56 88 81 0B 00 00 00 00 00 60 01 F7 
16:56:32.203 > RX Period End
16:56:32.203 > All missing
16:56:32.203 > Nothing received, resend whole request
16:56:32.203 > TX ActivePowerControl Channel: 40 --> 51 83 81 91 93 80 13 56 88 81 0B 00 00 00 00 00 60 01 F7 
16:56:33.508 > Received MQTT message on topic: solar/112183819193/cmd/limit_nonpersistent_absolute
16:56:33.508 > Limit Non-Persistent: 0.0 W
16:56:34.249 > RX Period End
16:56:34.249 > All missing
16:56:34.249 > Nothing received, resend whole request
16:56:34.249 > TX ActivePowerControl Channel: 61 --> 51 83 81 91 93 80 13 56 88 81 0B 00 00 00 00 00 60 01 F7
```

### Anything else?

This issue exists even if every OpenDTU relevant topics were deleted from the MQTT broker before OpenDTU is started.

some configurations which might be interesting:

[image]

[image]

[image]

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2024-11-18:
try stopping your power control commands and let the queue be cleared.... does it work then?

mibcat, 2024-12-06:
If this is the intended behavior I close this issue.

closed 2024-12-06 by mibcat

stefan123t, 2024-12-07:
@mibcat this is a common mistake that users tend to continue sending commands while the inverter is unavailable.
I think tbnobody is already working on a kind of kill-switch logic for not accepting or overwriting the last queued ActivePowerLimit command during night / absence of inverters.

@tbnobody you may want to look into the logs that we received from the Hoymiles DTU Pro S in #1650. I think that shows quite well how the vendor tries to reach/ping the inverter for up to 10 minutes using ChannelChangeCommands and only when he gets a reply continues with the other communication, ie RealTimeRunDebug, SystemConfigPara, etc.

Maybe we have to use the same procedure with the HMS/HMT models ?