tbnobody/OpenDTU #2434: [Request] Order of ok/cancel buttons, kWh for total yield day, total power in header

harvey637, 2024-11-27:
### Is your feature request related to a problem? Please describe.

no

### Describe the solution you'd like

- some buttons are "cancel - save", some are "save - cancel". eg settings-dtu versus modify-inverter
- total_yield_day is in Wh unlike total_yield_total, but total_yield_day is easyly more than kWh, with kWh all energy is of same unit magnitude (not kWh and Wh)
- write total_power in the header with autorefresh eg. every 30 s, nice overview with browser tabs without opening the tab
- Is "Total Yield Total" strange wording? (german is ok) -> "Alltime Total Yield"

### Describe alternatives you've considered

this is more or less cosmetical, but make the tool nicer, more consistent in buttons and units, not mixing thousands-serarator and decimals separator (with the same unit magnitude)

### Additional context

always confused about "Total Yield Total" = 222.999 kWh "Total Yieald Day" = 2,999 Wh (english) 
"Gesamtertrag Insgesamt" : 222,999 kWh "Gesamtertrag Heute" : 2.999 Wh (german)

I would prefer (using "," and "." in german/english, but as it is the same magnitude of unit they are not "mixed"
"Total Yield Total" : 222.999 kWh "Total Yield Today" : 2.999 kWh (english)
"Gesamtertrag Insgesamt" : 222,999 kWh "Gesamtertrag Heute" : 2,999 kWh (german)

stefan123t, 2024-12-06:
@harvey637 I agree some UX elements could be improved / harmonized:

* Button order should be save - cancel, everywhere you can overwrite data. When creating new data cancel - save.
* YieldDay vs YieldTotal and TotalYieldTotal as well as TotalYieldDay should be the same magnitude, i.e. kWh this is the most common unit, e.g. on your electricity bill too
* Updating the title tab with TotalPower or TotalYieldDay would be indeed a nice feature
* also see #2435 for a request on having the Power for each Inverter shown in the Tabs / Pills for each inverter
* "Alltime Yield Total" sounds better but actually the TotalYieldTotal is the sum of all Inverter YieldTotal's, whereas the Total YieldDay is the sum of all Inverter YieldDay's. 
  So this would require us to rename the Yield Total to Alltime Yield and Total Alltime Yield, I guess this might get more confusing, as this would have to match the REST API names too.
  -> YieldTotals Sum or YieldDay Sum would be more appropriate maybe, but I am unsure whether this is an easy change.

* I have a similar UX observation regarding button sizes I would like to add:

  * Buttons for editing / deleting the inverter have no backdrop / button but are just 16x16 icons 
    (this is too small for my thick mouse pointer :smile_cat: ).
    [image]
  * Inverter buttons are 42x34 and use 24x24 icons
    [image]
  * Padding for Tabs / Pills and Inverter headers are usually 8px or 16px padding.
    [image]
  * Text buttons are 38xWidth and use a 6px / 12px padding
    [image]

harvey637, 2024-12-07:
Hi, thank you for looking into this!
- this TotalYieldTotal might still be used in api (api changes are a sad story), but the UX Interface in different languages could reflect a more "User" centric language and does not use this irritating "Total Yield Total", where one "Total" meaning all Inverters, the other "Total" meaning all days.
- playing with words like "overall" and "alltime" and "sum" might (consistingly) create a "over all inverters" and "over all times" semantic.

Overall view is ok on small devices, but looks very large on 24" screen in 1920x1080 landscape, not even giving one single inverter without scrolling, very big blocks with large fonts (but this is my personal opinion, and scrolling is really not so nice)

harvey637, 2024-12-07:
I really prefer all energy units to be in "kWh", with three decimals.
Writing "Wh" in multiple (precicion) of WattHour with thousands-separator and no decimals reads more or less the same as "kWh" with tree decimals (so the same precision as above) and below 1000 kWh no thousands-separator. This also removes in most cases the mixture of decimals (english dot ".", german colon ",") and thousands-separator (english colon ",", german dot ".")

(english) 123,789 Wh ==> 123.789 kWh,     456 Wh ==> 0.456 kWh
(german) 123.789 Wh ==> 123,789 kWh,    456 Wh => 0,456 kWh

harvey637, 2024-12-07:
An the wording "Total Yeald Day" might be "Total Yeald Today", as there is only today an no other day  :-)
In german it is perfect!