tbnobody/OpenDTU #2435: [Request]

Hubseba, 2024-11-27:
### Is your feature request related to a problem? Please describe.

Performance overview of the individual inverters.

### Describe the solution you'd like

It would be great if the power of each inverter could be displayed in an overview. This would save you having to click on each inverter individually to get a quick overview.
This could be done directly next to the name, for example:
[image]

Hubseba, 2024-12-09:
> Please specify which Power value you would like to have in / next to the Pills ?

I would like to have the AC output power, this is the value that counts for me.

stefan123t, 2024-12-10:
@Hubseba Please edit / update the title of your feature request!

tbnobody, 2025-01-14:
Implemented in 8f60a3a12a14168b5f990fe74f36c326fb57121f

closed 2025-01-14 by tbnobody