tbnobody/OpenDTU #2438: [Request] ReInit after "TX SPI Timeout"

robsy, 2024-11-30:
### Is your feature request related to a problem? Please describe.

Sometimes my OpenDTU (CMT2300a) gets an infinitive "TX SPI Timeout", result is that there is no longer communication between OpenDTU and the inverter. So after every request to the inverter the message "TXI SPI Timeout" comes, and there is no automatic way out. I found a very simple way to get the communication working again:
Increase or Decrease  the TX power for 1 (eg. if set "normally" to 15: change to 14 or 16). Immediately afterwards the communication restart without any problems. Setting the TX power back to original value has no negative effect.

### Describe the solution you'd like

Just an idea:
- create an option like "step TX power to reinit SPI"
- count directly consecutive SPI TX Timeout failures (reset counter if an "normal" communication happens)
- if counter is above a e.g. 5 (or any other value) change the TX power by 1
- if SPI communication  is "reestablished" change back TX power to original value

### Describe alternatives you've considered

My first thought was that the power supply is not stable, but after putting some High-C MLCC and Low-Impedance Elcos to 5V and 3,3V line there is no improvement (also not by changing USB cable and USB supply). Monitoring 5V and 3,3V by an high-res Oscilloscope showed not abnormalities. 

### Additional context

Because of my limited experience of C++ (my daily-business are 8 bit controller with hardware-related programming C and assembler), I'm not sure if I could program my idea correctly. I will try and (maybe) come back with a pull request (or only a raw-patch to discuss).

schlimmchen, 2024-12-04:
You stabilized the power rails, which is good, but how does the rest of your setup look like? This sure does sound like a hardware issue, like loose cabling, or high capacitance on at least one SPI line, or serial resistance.

BTW: The thing that does the trick is probably not the increase or decrease of the TX power, but restarting the SPI "driver" when applying the new TX power setting. That's a guess, I have not looked at the code, but I am confident that there is some kind of re-initialization going on.

stefan123t, 2024-12-05:
@schlimmchen yes increasing / decreasing the Transmit power for the RF radio chip may require the code to re-initialize the radio module.

I am also unsure whether this works just by resetting the radio stack on the software side or it may also reinitialize the whole communication including the Channel Change Command to each and every imverter in the settings / configuration.

I think that @robsy may have found a nice hack / workaround to one of the most dreaded issues upstream. E.g. this https://github.com/tbnobody/OpenDTU/issues/2278 but also many other issues upstream come to my mind.

@tbnobody could you look at this too ?

khschmidt, 2024-12-12:
Hallo Stefan,

das hat mit dem von mir beschriebenen Verbindungsabbruch beim HMS 1600-4T nichts zu tun. Das war natürlich das erste was ich versucht habe, Power rauf und speichen, auch Neustart und gänzlich von der Spannungsversorgung nehmen und neu starten hat nichts gebracht. Ich habe auch eine 2. DTU aufgebaut und es mit der versucht, alles erfolglos. Nur der sofortige Frequenzwechsel auf das Nachbarband hat bei mir zum Erfolg der sofortigen Verbindungsaufnahme geführt, siehe https://github.com/tbnobody/OpenDTU/issues/2278#issuecomment-2537437801

pi-amp, 2025-01-30:
Hello,
same problem for me. Don't know if it's software or hardware related, but an option to reinit the radio module when infinite tx spi timout would be great !

robsy, 2025-02-06:
Hi,

sorry I hand no time in the meantime to write a patch as described. To get rid of the problem I made a very dirty workaround: I made a cron  job (every minute, or which time you like) setting the TX power to e.g. -9 and 30s later setting the TX power to e.g. -7 (cmt_palevel).  

Not nice, but working.

dtu-reset.sh
`#!/bin/bash

curl -d @/home/pi/dtu-reset/dtu1.out -u "admin:Password" http://DTU-IP/api/dtu/config
sleep 30
curl -d @/home/pi/dtu-reset/dtu2.out -u "admin:Password" http://DTU-IP/api/dtu/config
`
dtu1.out
`data={"serial":"199980166832","pollinterval":1,"nrf_enabled":false,"nrf_palevel":0,"cmt_enabled":true,"cmt_palevel":-7,"cmt_frequency":869500000,"cmt_country":0,"cmt_chan_width":250000,"country_def":[{"freq_default":865000000,"freq_min":860250000,"freq_max":923500000,"freq_legal_min":863000000,"freq_legal_max":870000000},{"freq_default":918000000,"freq_min":900250000,"freq_max":963500000,"freq_legal_min":905000000,"freq_legal_max":925000000},{"freq_default":918000000,"freq_min":900250000,"freq_max":963500000,"freq_legal_min":915000000,"freq_legal_max":928000000}]}`

dtu2.out
`data={"serial":"199980166832","pollinterval":1,"nrf_enabled":false,"nrf_palevel":0,"cmt_enabled":true,"cmt_palevel":-9,"cmt_frequency":869500000,"cmt_country":0,"cmt_chan_width":250000,"country_def":[{"freq_default":865000000,"freq_min":860250000,"freq_max":923500000,"freq_legal_min":863000000,"freq_legal_max":870000000},{"freq_default":918000000,"freq_min":900250000,"freq_max":963500000,"freq_legal_min":905000000,"freq_legal_max":925000000},{"freq_default":918000000,"freq_min":900250000,"freq_max":963500000,"freq_legal_min":915000000,"freq_legal_max":928000000}]}`

fr-fhe, 2025-05-15:
Hello, I was tired about TX SPI TIMEOUT problem.
I investigate, i found a simple source code resolution and to test my code, I FOUND THE ROOT OF PROBLEM : CPU saturation !

Test bench : 
- Working OpenDTU somewhere.
- From your PC, open a web page showing console with log level on error only (so nothing should appear).
- From another device (another brower tab, or my phone in my case), open a web page to System Information and angry-click on refresh button : this will push webapp refresh at its max and saturate CPU.
- And, voila!, TX SPI TIMEOUT appears....

If warriors of openDTU source code can investigate where can be the bottleneck, the TX SPI TIMEOUT should disappear.
I can also note that TX SPI TIMEOUT error is badnamed, because it's a general TAG for _radio->write() problem, without discriminate if the cause is OpenDTU, CMT2300A, SPI bus or whatsever.

If someone is really in need to have a workaround, you must recompile OpenDTU from scratch with these alterations : 
- in cmt2300wrapper.h file, add " void restart() { this._init_radio(); } " in public class section
- in HoymilesRadio_CMT.cpp, insert " _radio->restart(); " line just after " ESP_LOGE(TAG, "TX SPI Timeout"); " (currently line 285).
It's an ugly hack and i made something a little more sophisticated (i have a counter to 10 before restart, and i separate code in .h/.cpp). I works flawlessly at home since some hours. 

Again, i a source warrior, easy with git, can made a patch, it must : 
1/ Investigate more on error return conditions on _radio->write()
2/ restart radio only on the real problem (whatever it is)
3/ restart radio a better way ?

Keep in touch.
Fred.

schlimmchen, 2025-05-16:
Thanks for this info and a way to both reproduce the issue (did not yet try it myself) and to mitigate it.

I don't think that restart is very ugly. If that is what needs to be done to recover from this situation...

I am curious if maybe this is an interrupt (priority) issue?

fr-fhe, 2025-05-16:
In fact, tracing code execution is very delicate in this source code due to tasking.
Also, i don't know well hardware concurrencies in ESP32, so i talk about "CPU saturation" as a generic term (to move brains in community :; ) but it could be software task priority, cpu arbitration, IT issue -like you think, DMA concurrency, what else ?

I don't have enough knowledge on this hardware+software combo to continue investigation.

From my personal situation : 
- last 25.5.10 source code
- Wemos ESP32 module with type 3 (small oled) display
- Big CMT2300A module soldered in 4 wires mode (no GPIO2 nor 3).
- All wiring seems to be default configuration display + cmt2300 (i don't remember, i do it some months ago).
- 2 HMS-1600-T inverters on 4+2 panels

But I remember problems appears when i connected the second inverter to DTU (never had a problem with only one inverter).

Also, at every morning, OpenDTU appears offline and i was in obligation to "save DTU parameters" to restart radio communication.
NOW, it works flawlessly (first time OK this morning).
In need to test more, but it's very good for me now.

I've done my work on Visual Studio Code + PlatformIO.
I cloned repository from github from VSC (thanks documentation)
I made changes (10 to 15 lines at all) in 4 files.

But I'm not GIT fluent, so if someone can tell me how to cleanly push my modifications to github to help community, i'll do it.

LennartF22, 2025-05-16:
Do you know whether it‘s possible to reproduce this without beeing connected to a HMS/HMT inverter?

I could look into it, and I have a testing setup with a CMT2300A, but I currently do not have a HMS/HMT inverter, so the best I could do would be to configure a non-existent inverter.

fr-fhe, 2025-05-16:
Try fake serials like 116495000001 & 116495000002 ? (HMS-1600/1800/2000-4T)

fr-fhe, 2025-05-16:
Ok, there is a BIG PROBLEM with DATA management in OpenDTU.

I made a test : adding 116495000001 & 116495000002 inverters in my configuration, to test validity.
Then, when removing these fake inverters, saving page, my OpenDTU crashed and not responding then.
(last thing i see in an already opened and displaying openDTU console was error messages about a xxxTaskCb ... too bad i didn't make a screenshot !). 
No able to make a soft reboot, webapps not responding.
Hard reboot by removing power, but no display. Burned ? Bricked ?

Had to dismantle installation, install USB cable and serially see that configuration diseappeared ! 
(all is default so no display, no wifi, no nothing).

So bad to remember *after removing all the hardware* that openDTU has a default config AP mode ... but ok.

I reconfigure wifi access, load mapping configuration and found that i have never made (or lost) dtu configuration.
So i have to retrieve my inverter serials and reconfigure OpenDTU from scratch ,..

Something is bad in data management, and exchanges via webapps api....
" There are more things in heaven and earth, OpenDTU, than are dreamt of in your technology "

Good day, bad day.
Will be better tomorrow...

tbnobody, 2025-05-16:
> Ok, there is a BIG PROBLEM with DATA management in OpenDTU.

Its not a problem of data management but memory limitations.

Sometimes I saw the same behavior. Thats why I introduces this:
https://github.com/tbnobody/OpenDTU/commit/98675263b66b7925bea7ce5192be67188194b7db

tbnobody, 2025-05-16:
@fr-fhe How many inverters have you configured (incl. the two dummy ones?)

LennartF22, 2025-05-17:
Unfortunately I was not able to replicate the issue thus far. I'm using v25.5.10 with two dummy inverters and hammering the page refresh for some minutes did not produce the error message. I guess I'd need real inverters to reproduce it :/

fr-fhe, 2025-05-19:
@LennartF22 : So Strange. I Really think real inverters are not needed, because problems seems to occurs at main soc level.
A single click is sufficient enough for me to produce TX error... but i have 1 second DTU refresh rate, thanks to my patch ^^

Could there be some hardware issues ? 
I use an old'n'rusty nodeMCU like ESP32 board from 2017 (buyed on AliExpress as "Wemos Pro ESP32" at this time).
It's a model ESP32-D0WD-V3 (Rev3 soc) with 4MB flash. PIO environnement is generic_esp32 with my custom patched 25.5.10 firmware.

Except for this strange refresh-causing-TX behavior, my patched version is rock solid, online since 3 days.

jstammi, 2025-05-19:
Do you have 1s refresh rate **and** removed the tx-repetition-on-no-reponse
with your patch?

fr-fhe, 2025-05-19:
Once again, it's a less than 10 lines patch on 4 files (can also be a ugly 2 lines patch....).
If community thinks it's a good thing, tell me how-to make a git patch proposal with VSC.

pi-amp, 2025-05-21:
same error here. After somle time (1 hour, 30 minutes...) TX SPI timeout and no data. i have to open the "DTU" settings and save to make it work again. Don't even have to change settings !

fr-fhe, 2025-05-21:
> maybe we need a more modern esp32 with more cpu power ?

My patch works. 
ESP32 seems powerful enough if not stressed.

pi-amp, 2025-05-23:
i'm trying your code on opendtu-onbattery, wich i'm using. hope it works !

pi-amp, 2025-05-23:
it needs to be changed to 
void restart() { _init_radio(); }

to compile :)

gitisgreat2023, 2025-09-22:
> Hi,
> 
> Not a gorgeous patch, but it works flawlessly since May (OpenDTU v25.5.10) :
> 
> [TX_SPI_TIMEOUT.patch](https://github.com/user-attachments/files/22272782/TX_SPI_TIMEOUT.patch)
> 
> I was looking for a pull request, but no time anymore today. Will perhaps try later.

@fr-fhe did you find the time? I would love the feature as I currently have to do it manually…