tbnobody/OpenDTU #2445: [Request] Add opendtu update available on display

jaysee, 2024-12-05:
### Describe the solution you'd like

Adding a periodic check about openDTU update, and add a notification displayed on the screen (may also be a led indicator for those without screen)

schlimmchen, 2024-12-05:
In case it matters: I am strongly against OpenDTU calling home, even if it's opt-in and even if no error message is produced in case the connection fails (my OpenDTU cannot reach the Internet). Notice that the update-check in the web UI is implemented in the Browser. Also I think that users who are happy with their current setup shall only touch it if they are inclined to due to new features, not because of the device yelling "There is an update!" :wink:

tbnobody, 2024-12-05:
I agree with @schlimmchen. Will most likely not implement something like this.

closed 2024-12-05 by tbnobody