tbnobody/OpenDTU #2446: DTU update inverval is increasing over time (when inverter is unreachable)

mibcat, 2024-12-06:
### What happened?

In case of the inverter is unreachable over a longer duration (not powered for days) the update interval is increasing. It will become a problem, when the update interval is exceeding the time limit **exp_aft** (double value of the polling interval) stated in MQTT config topic for Home Assistant, like:

MQTT topic **homeassistant/sensor/dtu_1..........3/ch0_Power/config**

```yaml
{
  "dev": {
   ...
  },
  "unit_of_meas": "W",
  "dev_cla": "power",
  "stat_cla": "measurement",
  "name": "Power",
  "stat_t": "solar/1..........3/0/power",
  "uniq_id": "1..........3_ch0_Power",
  "exp_aft": 10
}
```

This will than lead to Nan sensor values in Home Assistant and interfere with my ESPhome application.

### To Reproduce Bug

Running OpenDTU with permanently turned-off inverter.

### Expected Behavior

The MQTT sending interval of all sensor values should not exceed the threshold **exp_aft**.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

3dc70ab

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Relevant log/trace output

```Shell
console log with correct MQTT sensor sending interval (5s)

Fetch inverter: 1..........3
19:31:26.671 > Request SystemConfigPara
19:31:26.716 > TX RealTimeRunData Channel: 3 --> 15 83 81 91 93 80 13 56 88 80 0B 00 67 50 9F FE 00 00 00 00 00 00 00 00 A9 AB 87 
19:31:27.270 > RX Period End
19:31:27.270 > All missing
19:31:27.270 > Nothing received, resend whole request
19:31:27.270 > TX RealTimeRunData Channel: 23 --> 15 83 81 91 93 80 13 56 88 80 0B 00 67 50 9F FE 00 00 00 00 00 00 00 00 A9 AB 87 
19:31:27.821 > RX Period End
19:31:27.821 > All missing
19:31:27.821 > Nothing received, resend whole request
19:31:27.821 > TX RealTimeRunData Channel: 40 --> 15 83 81 91 93 80 13 56 88 80 0B 00 67 50 9F FE 00 00 00 00 00 00 00 00 A9 AB 87 
19:31:28.373 > RX Period End
19:31:28.373 > All missing
19:31:28.373 > Nothing received, resend whole request
19:31:28.373 > TX RealTimeRunData Channel: 61 --> 15 83 81 91 93 80 13 56 88 80 0B 00 67 50 9F FE 00 00 00 00 00 00 00 00 A9 AB 87 
19:31:28.924 > RX Period End
19:31:28.924 > All missing
19:31:28.924 > Nothing received, resend whole request
19:31:28.924 > TX RealTimeRunData Channel: 75 --> 15 83 81 91 93 80 13 56 88 80 0B 00 67 50 9F FE 00 00 00 00 00 00 00 00 A9 AB 87 
19:31:29.446 > RX Period End
19:31:29.446 > All missing
19:31:29.446 > Nothing received, resend count exeeded
[... 281 lines cut ...]
16:42:08.254 > All missing
16:42:08.254 > Nothing received, resend whole request
16:42:08.254 > TX RealTimeRunData Channel: 75 --> 15 83 81 91 93 80 13 56 88 80 0B 00 67 53 1B 4E 00 00 00 00 00 00 00 00 17 D7 72 
16:42:08.802 > RX Period End
16:42:08.802 > All missing
16:42:08.802 > Nothing received, resend whole request
16:42:08.802 > TX RealTimeRunData Channel: 3 --> 15 83 81 91 93 80 13 56 88 80 0B 00 67 53 1B 4E 00 00 00 00 00 00 00 00 17 D7 72 

```
```

### Anything else?

Don't know when the update interval is rising - in this example I got a 14s instead of 5s

![image](https://github.com/user-attachments/assets/03118e6a-8e38-463b-a331-89efe9db70f0)

A reset of the OpenDTU device is fixing the problem until the next time.

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

mibcat, 2024-12-06:
hmm the image got lost after editing the post ...

bottom: update interval 5s right after start of OpenDTU
top: update interval 14s after operation time 	2 Tage 21:23:02

[image]

stefan123t, 2024-12-07:
According to your console logs there is no communication with your inverter what so ever.

```
19:31:26.716 > TX RealTimeRunData Channel: 3 --> 15 83 81 91 93 80 13 56 88 80 0B 00 67 50 9F FE 00 00 00 00 00 00 00 00 A9 AB 87 
19:31:27.270 > RX Period End
19:31:27.270 > All missing
19:31:27.270 > Nothing received, resend whole request
```

You may want to revisit / edit your triple backticks markdown formatting to show proper code.

mibcat, 2024-12-08:
you are right - the missing image was caused by wrong usage of the tripple backticks - just fixed it.

and yes: the inverter was permanently shut-down during the whole time when OpenDTU was started until the issue arise.

stefan123t, 2024-12-08:
Okay I think now I understand your issue:

The OpenDTU may choose tobuse a different communication interval with the inverter, depending on whether it has been able to reach the inverter before the Inverters downtime and how long ago the inverter has become unavailable, ie 5s vs ~14/15s interval.

This in itself is not the actual problem and the OpenDTU fowards the current values to your MQTT broker using a **fixed exp_aft validity** period of 10s. This is sufficient in the first place being double the interval of 5s.

But when the inverter has been unreachable for a longer period it is less than the query interval of 14/15s and therefor the MQTT recipient has short gaps of 4/5s. In fact the exp_aft should be increased to 30s being again the double of our query interval.

@tbnobody is this something that we could consider in our MQTT packets ?

mibcat, 2024-12-09:
To have two different update rates depending on the availability of the inverter is totally fine as long as the applied update rate is not exceeding the communicated **exp_aft** value.

From the user's perspective a possible solution could be a dynamic exp_aft value or a fixed value with the longest expected duration + some margin.

And many thanks for taking time to look into this issue !!!

stefan123t, 2024-12-10:
And what should the inverter do eg after dark?
Should the DTU repeat the last value and extend the value lifetime until eg midnight or even until a new dawn ? Should there be an expiry after some time of the inverter being unreachable or would you prefer that it simulates everything is fine until your defined reset at midnight / dawn ?

mibcat, 2024-12-10:
I think the best would be to set all life sensor values to zero/inactive if the inverter is detected as unreachable because these sensor values are simply not available if the inverter is unreachable **_regardless_** of the reason (no power due to light conditions or as in my case no battery voltage supply).

closed 2025-04-18 by github-actions[bot]