tbnobody/OpenDTU #2488: New serial number for HMS-2000-4T-NA inverter unknown

Azimuth0r, 2025-01-05:
### What happened?

[image]

I have a new inverter with serial number starting 1420A02 which is detected as Unknown and not added as the correct model.

### To Reproduce Bug

Add an inverter with this serial sequence

### Expected Behavior

Expect Inverter to be added as detected model HMS-1600/1800/2000-4T

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v24.11.7

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Azimuth0r, 2025-01-05:
[image]

photo of label

tbnobody, 2025-01-09:
Could you please replace the first 4 digits (1420) by 1164 and double check if its working?

Azimuth0r, 2025-01-09:
Yes, that did allow it to start responding.
[image]

tbnobody, 2025-01-09:
Thank you. Thats great! Are the production values also reasonable? Does the event log etc. also work? If this is the case I will add the prefix 1420 as the identification of HMS-2000 in the next release..

Azimuth0r, 2025-01-09:
I saw some low production numbers earlier (it was very overcast), and now both inverters are shut down.
I will know more in daylight tomorrow.

Azimuth0r, 2025-01-12:
Everything production-wise does look normal, however, there is somethign classified as unknown in the logs.
[image]

tbnobody, 2025-01-14:
>  there is something classified as unknown in the logs.

Thats ok as no one currently knows the meaning of id 223.

> Also the grid profile page does not show correctly

Please provide the data as mentioned in the "Support the development" section.

closed 2025-04-18 by github-actions[bot]