tbnobody/OpenDTU #2499: New serial number for MHS400 inverter unknown

geronimo-78, 2025-01-14:
### What happened?

When I try to add the new inverter :
[image]

### To Reproduce Bug

When I try to add the new inverter

### Expected Behavior

-

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

V24.11.7

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Relevant log/trace output

```Shell
-
```

### Anything else?

[image]
After change the model number
[image]

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-01-14:
Please test with the latest version as the prefix `1400` is already added there..

geronimo-78, 2025-01-15:
With your new firmware, the inverter is know.
Thank

tbnobody, 2025-01-15:
In this case I would close this issue.

closed 2025-01-15 by tbnobody