tbnobody/OpenDTU #2500: [Bug] Hint regarding missing device profile selection does not work as expected

schlimmchen, 2025-01-14:
### What happened?

The change to `WebApi_ws_live.cpp`

https://github.com/tbnobody/OpenDTU/blob/58382be16c733bead347f9ef21da6c9f11af994d/src/WebApi_ws_live.cpp#L129-L130

struck me as odd when reviewing Andreas' work to merge today's release into the downstream project OpenDTU-OnBattery.

### To Reproduce Bug

Build a non-generic firmware, like `opendtufusionv2`, flash and select the `(Default settings)` in the device manager. The hint to select a device profile will appear even though this is *not* a generic firmware image.

### Expected Behavior

My idea was to display the hint iff the PIO environment had "generic" in its name (and if no device profile was selected). Non-generic firmware images do have pins defined and are good to use without a pin mapping -- usually. At least users of those environments should know what they are doing, especially as they are self-compiling.

The way the feature is implemented in OpenDTU will show the hint if no device profile is selected, unless the PIO environment equals the legacy `generic` one. I assume you made a mistake?

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

v25.1.14

### What firmware variant (PIO Environment) are you using?

opendtufusionv2

### Anything else?

Maybe you meant this?

```c++
    bool isGeneric = String(PIOENV).startsWith("generic");
    hintObj["pin_mapping_issue"] = isGeneric && !PinMapping.isMappingSelected();
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-01-14:
> The way the feature is implemented in OpenDTU will show the hint if no device profile is selected, unless the PIO environment equals the legacy generic one.

You  are right. I've forgotten about the firmwares like `opendtufusionv2`. But just looking if the environment contains `generic` is also wrong because the `generic` only environment does not need any pin mapping.

tbnobody, 2025-01-14:
I am currently thinking about a define which could be set in an environment which defines whether a environment needs a pin mapping or not.

closed 2025-01-14 by tbnobody

schlimmchen, 2025-01-15:
I would still have tried to avoid a #define     

```c++
    bool needsPinMapping = String(PIOENV).startsWith("generic") && String(PIOENV) != "generic";
    hintObj["pin_mapping_issue"] = needsPinMapping && !PinMapping.isMappingSelected();
```

but I like this, too, as it gives explicit control, not relying on the particular name of the PIO environment. Thanks :+1:

commit 571ba2f3 (2025-01-14): Fix: Hint regarding required device profile is shown for profiles which don't need a device profile

This patch introduces a define which allows to specifiy for each environment if a device profile is absolutly required.

Fixes #2500