tbnobody/OpenDTU #2504: No wifi connection with 25.1.24

andreas-bulling, 2025-01-19:
### What happened?

I have upgraded to 25.1.14 yesterday and today OpenDTU suddenly doesn't connect to wifi anymore. Nothing else has changed. It just doesn't connect.

### To Reproduce Bug

uprade to the latest OpenDTU version

### Expected Behavior

that it connects as it did with the previous firmware version (some 24.x version)

### Install Method

Pre-Compiled binary from GitHub actions/pull-request

### What git-hash/version of OpenDTU?

can't tell - device doesn't connect to wifi

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-01-19:
> uprade to the latest OpenDTU version

Cannot reproduce, works perfectly

What was your previous version? What do you see on the [serial console](https://www.opendtu.solar/firmware/howto/serial_console/)?

schlimmchen, 2025-01-19:
Sounds to me like he [obsoleted](https://github.com/tbnobody/OpenDTU/issues/2400#issuecomment-2600846941) this issue already?

tbnobody, 2025-01-19:
You are right. Didn't look at the message author in the other thread. Going to close this issue..

closed 2025-01-19 by tbnobody