tbnobody/OpenDTU #2505: viele Zeitabgleiche mit v25.1.14

tm-107, 2025-01-19:
### What happened?

Seit v25.1.14 habe ich jeden morgen bis zu einer Stunde lang immer wieder Zeitabgleiche, teilweise im Minutentakt.
Das Problem betrifft alle 6 bei mir angeschlossenen Wechselrichter (1xHM-600, 1xHM-700, 4xHM-1500).
Während dieser Zeit klappt auch das Abfragen der JSON-Daten per http-request sehr oft nicht.

Ich habe das die letzten 4 Tage jeweils 2x getestet:
- mit v24.11.7: keine Probleme (lediglich ein Zeitabgleich bei einem Wechselrichter)
- mit v25.1.14: täglich bis zu >30 Zeitabgleiche bei allen Wechselrichtern (durchschnittlich 3-6 Zeitabgleiche)

### To Reproduce Bug

Just install v25.1.14

### Expected Behavior

Nur vereinzelte / sporadische Zeitabgleiche wie in den Versionen davor.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

653efb4 ??? (v25.1.14 eben)

### What firmware variant (PIO Environment) are you using?

generic

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Omega13x, 2025-01-20:
Bei mir mit dem ESP32S3 das gleiche Problem. Es sind ständige Reboots von OpenDTU. Das letzte stabile Image war das vom Oktober letzten Jahres.

tbnobody, 2025-01-20:
> Bei mir mit dem ESP32S3 das gleiche Problem. Es sind ständige Reboots von OpenDTU.

@tm-107 schreibt von Zeitabgleichen. @Omega13x von Reboots.... Ist für mich erstmal nicht das gleiche.

Unabhängig davon, was siehst du auf der seriellen Konsole bei einem Reboot?

schlimmchen, 2025-01-20:
Du musst, wohl der übel, einen [Rechner per USB anklemmen und beobachten/mitschneiden](https://www.opendtu.solar/firmware/howto/serial_console/), weil man anders nicht an diese Info herankommt. Da wird dann bei einem Reboot ein Backtrace stehen, der uns erlaubt zu vermuten, wo das Problem liegen könnte. Dazu schreibe bitte genau auf, mit welcher Version der Backtrace aufgetreten ist.

tm-107, 2025-01-20:
> Die Zeitabgleiche kommen nach dem Reboot. [@tm-107](https://github.com/tm-107) könnte es bestimmt bestätigen wenn er die Zeiten mal vergleicht. Leider sind sie sehr unregelmäßig. Kann man irgendwie ein Langzeitlog starten um es beobachten zu können? Bei mir waren es seit Mitternacht vier Reboots. Davor fast zwei Tage keine.

Heute morgen hatte ich auch wieder mehrere Zeitabgleiche, heute allerdings nur 2-3 (je nach Wechselrichter). Die Uptime liegt aber bei über einem Tag, also liegt es bei mir nicht an Reboots.
Die Zeitabgleiche finden auch nur in der Zeit zwischen "aufwachen" des Wechselrichters und Start der AC-Produktion statt. Wenn also morgens die Sonne scheint, geht das recht schnell (ca. 1-5 Zeitabgleiche). An bewölkten Tagen dauert es länger (bis zu >30 Zeitabgleiche, wobei OpenDTU [oder der Wechselrichter?] ja nur die letzten 15 Meldungen anzeigt - man muss also daneben sitzen und beobachten).

broth-itk, 2025-01-21:
Wird denn die Uptime immer wieder zurückgesetzt, was auf einen Reboot hindeutet?

tm-107, 2025-01-21:
> Wird denn die Uptime immer wieder zurückgesetzt, was auf einen Reboot hindeutet?

Ja, soweit ich das deuten kann ...
Dafür spricht auch, dass meine Wechselrichter zu völlig unterschiedlichen Zeiten den Zeitabgleich machen. Da liegen schon mal 30 Minuten dazwischen. Wenn es an einem Reboot liegen würde, müssten die Zeiten ja ähnlich sein.

Heute wieder "nur" 3 Zeitabgleiche ...

[image]

Uptime:

[image]

broth-itk, 2025-01-21:
Die DTU läuft aber seit 59 Stunden ohne Neustart durch... einen Neustart/Crash würde ich mal ausschließen.

tm-107, 2025-01-21:
> Die DTU läuft aber seit 59 Stunden ohne Neustart durch... einen Neustart/Crash würde ich mal ausschließen.

Ach so, Sorry ... habe Deine Frage falsch verstanden ...
Ich dachte Du meinst, ob die Uptime bei einem Neustart zurückgesetzt wird. Hatte mich schon gewundert ...

Omega13x, 2025-01-22:
Scheinen die Ursachen doch unterschiedlicher Natur zu sein. Bei mir fällt der letzte Zeitabgleich immer mit einem Reboot zusammen. Ich bin jetzt wieder zurück auf die v24.10.15. Da ich sowieso alles per Home Assistant überwache, ist es ziemlich egal von welcher Version die Daten kommen. ;)

kopierschnitte, 2025-02-01:
Komisch, also bei läuft die v25.1.14 seit knapp zwei Wochen ohne irgendwelche Neustarts/Hänger bzw. übermäßige Zeitabgleiche einfach durch.

Das einzige, was stört ist noch die Wifi-Problematik, wenn das Netz unvorhergesehen wegbricht. Aber das wird ja schon anderweitig diskutiert.

Nutze generic_esp32s3_usb als Basis.

tm-107, 2025-02-01:
Tja, einige Tage war es auch hier recht still, aber heute waren es bisher wieder 3-5 Zeitabgleiche (je nach Wechselrichter).

[image]

tm-107, 2025-02-01:
> Tja, einige Tage war es auch hier recht still, aber heute waren es bisher wieder 3-5 Zeitabgleiche (je nach Wechselrichter).
> 

Update: Vorhin nicht nachgeschaut, aber diesmal waren die Zeitabgleiche auch bei mir im Zusammenhang mit einem Reboot von OpenDTU (Reset Grund: Software reset CPU).
In meiner Statistik ist der PV-Ertrag von 10:50 Uhr bis 11:40 Uhr mit 0W und Fehlermeldung (keine JSON-Werte abrufbar) angegeben.

tm-107, 2025-02-03:
Heute auch wieder mehrere Zeitabgleiche, verbunden mit einem Neustart von OpenDTU (Software reset CPU).
Daten mitschneiden wird bei mir leider schwierig, da die OpenDTU im Arbeitszimmer keinen Empfang hat und ich kein 20m langes USB-Kabel habe um sie an ihrem eigentlichen Ort zu "belauschen".

Omega13x, 2025-02-04:
Habe Gestern mal die v25.2.3 drauf gemacht. Bis jetzt schon 15 Zeitabgleiche bei jedem der drei Wechselrichter und die laufen erst seit zwei Stunden. Also alles wieder zurück zur v24.10.15.

tm-107, 2025-02-04:
Gleiches Bild hier. Habe heute Nacht auch die v25.2.3 aufgespielt und aktuell je nach Wechselrichter 4-15 Zeitabgleiche. 
Diesmal aber wieder ohne reboot ...

broth-itk, 2025-02-04:
Bei meinen drei WRs sehe ich das auch bei zweien die jeweils am weitesten von der DTU entfernt sind.
Der WR in ca. 4m Luftlinie hat nur drei Eventlog-Einträge, kein einziger Timesync.

wittimagic, 2025-02-13:
Ich bin noch neu was DTU angeht, meine ist erst ein paar Tage im Betrieb mit 2 Wechselrichtern (HMS-1600-4T).
Ausgeliefert wurde diese mit 24.11.7 und dort hatte ich mit beiden WR sporadisch die Meldung bezüglich des Zeitabgleichs.
Nach Update auf 25.2.3 kamen die Meldungen bei beiden WR sehr häufig und auch das WLAN Netzwerk der DTU wurde sehr instabil, was dann natürlich auch Auswirkungen beim MQTT Dienst hatte. Auffällig war in meinen Augen, dass bereits ein Login auf der WebGUI offenbar zur Netzwerk Instabilität beigetragen hat bzw. diese verschlimmert hat, denn dann kamen die MQTT Daten kaum noch am Server an. Die WebGUI als solche brauchte auch mehrere Sekunden Bedenkzeit für Untermenüs.
Nach mehreren Anläufen konnte ich jetzt auf die 24.10.15 zurück und alles läuft stabil und flüssig. Es kommen auch keine Fehler bezüglich des Zeitabgleichs mehr hinzu.

gitisgreat2023, 2025-02-14:
> Du musst, wohl der übel, einen [Rechner per USB anklemmen und beobachten/mitschneiden](https://www.opendtu.solar/firmware/howto/serial_console/), weil man anders nicht an diese Info herankommt. Da [...]

Ich habe mal nachgeschaut, bei mir sind die Zeitabgleich auch nach einem reboot. Wäre es sinnvoll dass ich dann mal das mitschneiden mache? (OpenDTU on Battery natürlich ;-))

stefan123t, 2025-02-16:
Die Event ID 0x02 kommt i.d.R. im Event Log des Wechselrichters, wenn z.B. der in der RealTimeRunDataDebug 0x0B Abfrage von der OpenDTU übergebene UNIX Timestamp von dem des WR zu stark abweicht.

Zuerst hatten wir das als einen WR<->DTU Communication / unreachable Fehler interpretiert. Aber aus der warn_code.json der S-Miles Installer App wissen wir, dass es von Hoymiles als "Time calibration" oder zu deutsch "Zeitabgleich" bezeichnet wird.

Offenbar kommt es auch bei/nach einem Reboot der OpenDTU zu Kommunikation mit dem WR mit (noch) nicht synchronisierten Zeitstempeln ?

Oder könnte es sein dass der WR sich z.B. aufgrund fehlender Sonneneinstrahlung und/oder Temperaturen mehrfach neu startet und daher den Zeitstempel der OpenDTU schlicht vergisst ?

wittimagic, 2025-02-16:
@stefan123t bei mir war aber auch das WLAN extrem instabil und ich hatte den Eindruck, dass die Performance der DTU deutlich beeinträchtigt war.
Die Kommunikationsprobleme könnten aus meiner Sicht auch nur Symptome sein.

gitbreiti, 2025-02-18:
Ich hab die Zeitabgleiche auch seit ich auf die v25.2.3 aktualisiert habe. 
Hatte vorher schon länger kein Update mehr gemacht.

[image]

super-mertl, 2025-02-18:
ich kann das Thema bestätigen:

-Zeitabgleiche nehmen zu (gefühlt seit Okt/Nov, ich wechsle rel. regelmäßig auf die aktuelle Version)
-Zeitabgleiche sind überwiegend um den Ein- bzw Ausschaltzeitpunkt (Zeiten siehe nächster Punkt)
-Zeitabgleiche sind nicht immer bei allen WR (heute bei 2 WR gar nicht, bei 1 WR nur einmal 7:15, bei 1 WR 5 mal zwischen 7:05 und 7:15, aber auch um 12:15)

Habe 2x HMT-2250 und 2xHMS-1800 seit ca 1Jahr und sehe täglich nach dem System, daher kann  ich def. sagen, dass es mehr wurde und dass es aber weder einem Muster folgt noch jeden Tag vorkommt.

iomax, 2025-03-24:
Hi, I also have to confirm that with version 25.2.3 (never tried 25.1.14), the "2 - Time calibration" errors have returned:

- With version 24.11.7, there were little to no errors.
- I noticed that the error always occurs near, but not at every, MQTT request to set the production limit to 100% ("cmd/limit_nonpersistent_relative", 100). In my case, the limit update frequency is never less than 20 seconds, and in any case, with version 24.11.7, all requests were handled without errors.
- Also in my case (2 inverters: 1 HM-350 and 1 HM-400), the errors seem to occur randomly on one, the other, or both, while the limit requests are always sent in sequence and with the same values to both inverters.
- System Info does not report any CPU resets that could be associated with the errors.

Of course, switching back to 24.11.7 completely solves the issue.

iomax, 2025-04-07:
Quick&dirty update after quick try on 25.4.4 :

- With version 24.11.7, there were little to no errors.
- I noticed that the error always occurs near, but not at every, MQTT request to set the production limit ~~to 100% ("cmd/limit_nonpersistent_relative", 100).~~ In my case, the limit update frequency is never less than ~~20~~ 10 seconds, and in any case, <ins>with version 24.11.7, all requests were handled without errors</ins>.
- Also in my case (2 inverters: 1 HM-350 and 1 HM-400), the errors seem to occur randomly on one, the other, or both, ~~while the limit requests are always sent in sequence and with the same values to both inverters~~.
- System Info does not report any CPU resets that could be associated with the errors.
- <ins>Of course, switching back to 24.11.7 completely solves the issue.</ins>

wittimagic, 2025-04-07:
Feedback from my side running version 25.4.4 (compared to 25.2.3) so far:
- wlan connectivity is now stable
- web gui is now stable
- mqqt reports to home assistant are now stable

I will amend my experiance on the "time sync" errors tomorrow, when a little time has past after the boot of the inverters ( 2 x HMS-1600-4-T ).

wittimagic, 2025-04-09:
After 2 days testing with 25.4.4 the problem seems to be solved.
Everything is running rock solid, I have no stability problems any more and also no time sync errors.

Thank you for the excellent work and continuous commitment :-)

gitbreiti, 2025-04-09:
Also hier ist das Problem weiterhin vorhanden. Getestet mit v25.4.4

06:33:19 | 06:33:19 | 1 | Wechselrichter gestartet
08:55:14 | 08:55:14 | 2 | Zeitabgleich
08:55:18 | 08:55:18 | 2 | Zeitabgleich
08:57:14 | 08:57:14 | 2 | Zeitabgleich
08:57:18 | 08:57:18 | 2 | Zeitabgleich
08:57:43 | 08:57:43 | 2 | Zeitabgleich
08:57:48 | 08:57:48 | 2 | Zeitabgleich
09:21:54 | 09:21:54 | 2 | Zeitabgleich
09:22:59 | 09:22:59 | 2 | Zeitabgleich

Omega13x, 2025-04-09:
Bei mir auch. Habe zwei DTU's laufen. Eine mit zwei WR und eine mit einem WR. Habe mit beiden bis zu 15 Zeitabgleiche pro Tag.
Mit der v24.10.15  sind es NULL. 
Scheint zwar nicht relevant zu sein, sieht aber blöd aus.

gitisgreat2023, 2025-04-10:
> Bei mir auch. Habe zwei DTU's laufen. Eine mit zwei WR und eine mit einem WR. Habe mit beiden bis zu 15 Zeitabgleiche pro Tag. Mit der v24.10.15 sind es NULL. Scheint zwar nicht relevant zu sein, sieht aber blöd aus.

@gitbreiti @Omega13x 
Post bitte mal die Radio Statistics (ggf jetzt einen Reset und nach eine Stunde posten) und die Einstellungen von der DTU.

gitisgreat2023, 2025-04-10:
> > > Bei mir auch. Habe zwei DTU's laufen. Eine mit zwei WR und eine mit einem WR. Habe mit beiden bis zu 15 Zeitabgleiche pro Tag. Mit der v24.10.15 sind es NULL. Scheint zwar nicht relevant zu sein, sieht aber blöd aus.
> > 
> > 
> > [@gitbreiti](https://github.com/gitbreiti) [@Omega13x](https://github.com/Omega13x) Post bitte mal die Radio Statistics (ggf jetzt einen Reset und nach eine Stunde posten) und die Einstellungen von der DTU.
> 
> [@gitisgreat2023](https://github.com/gitisgreat2023) Du meinst die Funkstatistik, oder? Ich hab jetzt mal zurückgesetzt und poste dann nachher das Ergebnis. Welche Einstellungen willst du sehen?
> 
> Greez

Genau Funkstatistik. DTU transmitting power, (frequency for CMT2300A), poll interval.

gitbreiti, 2025-04-12:
> Genau Funkstatistik. DTU transmitting power, (frequency for CMT2300A), poll interval.

Hier ein Screenshot der Funkstatistik und der DTU Einstellungen.

[image]
[image]

Es sind tatsächlich jeden Tag exakt 14 Zeitabgleiche. In der Zahl dann 15 Meldung (inkl der WR gestartet Meldung). 

Greez

gitisgreat2023, 2025-04-15:
Das sind viele Wiederanforderungen... wie weit sind die WR und DTU aus einander?
Die Antenna in der DTU hat ja 6 grobe Orientierungen, (pro x,y,z Achse 2 Richtungen). Probier doch mal alle 6 aus (DTU einfach drehen). Vor Änderungen Statistik zurücksetzen. Auch wo genau die DTU steht macht bei mir viel aus, zwar vor dem Fenster aber mal nach links/rechts weiter zum Fenster weiter weg macht viel aus. In Summe macht die Orientierung sehr viel aus (10-20dBm) und der Platz noch mal 5dBm. Sendeleistung würde ich auch mal mit -6dBm probieren.

 Abfrageintervall mal auf 1 sec setzen, mit 6 Orientierungen und dann noch rumschieben und Sende 0 auf -6dBm sind viel Kombi's. Damit ist man man 15 Minuten beschäftigt, weiss anschließend dann aber etwa das globale Optimum.

gitbreiti, 2025-04-15:
> Das sind viele Wiederanforderungen... wie weit sind die WR und DTU aus einander? Die Antenna in der DTU hat ja 6 grobe Orientierungen, (pro x,y,z Achse 2 Richtungen). Probier doch mal alle 6 aus [...]

Aber was hat das mit den Zeitabfragen zu tun? 
Die treten ja bei v24.11.7 nicht auf. Sobald ich aber über diese Version hinaus gehe, sind es täglich 14 an der Zahl. 

Zu deiner Frage, ich schätze es sind vllt 4 Meter zum WR. Allerdings liegt eine Decke dazwischen. 
Ich werde mal die Position mal ändern und schauen ob sich die Wiederanforderungen damit verbessern oder beheben lassen.

gitisgreat2023, 2025-04-15:
> Aber was hat das mit den Zeitabfragen zu tun? Die treten ja bei v24.11.7 nicht auf. Sobald ich aber über diese Version hinaus gehe, sind es täglich 14 an der Zahl.
>
> Zu deiner Frage, ich schätze es [...]

Ich hatte genau das gleiche Problem, viele Zeitabfragen und auch kurze uptimes. Warum beide überhaupt mit der Qualität der radioverbinding zu tun hat; keine Ahnung. Seit die Verbindung optimiert ist, sind beide Probleme weg.

Josef1966, 2025-04-16:
> Ich hatte genau das gleiche Problem, viele Zeitabfragen und auch kurze uptimes. Warum beide überhaupt mit der Qualität der radioverbinding zu tun hat; keine Ahnung. Seit die Verbindung optimiert [...]

Habt ihr das schon mal probiert, hat bei mir voll geholfen (Schlauchschelle):

https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1519#discussioncomment-12102686

gitisgreat2023, 2025-04-16:
> > > > Das sind viele Wiederanforderungen... wie weit sind die WR und DTU aus einander? Die Antenna in der DTU hat ja 6 grobe Orientierungen, (pro x,y,z Achse 2 Richtungen). Probier doch mal alle 6 aus (DTU einfach drehen). Vor Änderungen Statistik zurücksetzen. Auch wo genau die DTU steht macht bei mir viel aus, zwar vor dem Fenster aber mal nach links/rechts weiter zum Fenster weiter weg macht viel aus. In Summe macht die Orientierung sehr viel aus (10-20dBm) und der Platz noch mal 5dBm. Sendeleistung würde ich auch mal mit -6dBm probieren.
> > > > Abfrageintervall mal auf 1 sec setzen, mit 6 Orientierungen und dann noch rumschieben und Sende 0 auf -6dBm sind viel Kombi's. Damit ist man man 15 Minuten beschäftigt, weiss anschließend dann aber etwa das globale Optimum.
> > > 
> > > 
> > > Aber was hat das mit den Zeitabfragen zu tun? Die treten ja bei v24.11.7 nicht auf. Sobald ich aber über diese Version hinaus gehe, sind es täglich 14 an der Zahl.
> > > Zu deiner Frage, ich schätze es sind vllt 4 Meter zum WR. Allerdings liegt eine Decke dazwischen. Ich werde mal die Position mal ändern und schauen ob sich die Wiederanforderungen damit verbessern oder beheben lassen.
> > 
> > 
> > Ich hatte genau das gleiche Problem, viele Zeitabfragen und auch kurze uptimes. Warum beide überhaupt mit der Qualität der radioverbinding zu tun hat; keine Ahnung. Seit die Verbindung optimiert ist, sind beide Probleme weg.
> 
> Habt ihr das schon mal probiert, hat bei mir bei der Verbindungsoptimierung voll geholfen (Schlauchschelle):
> 
> [hoylabs#1519 (reply in thread)](https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1519#discussioncomment-12102686)

Habe leider Flachantenne... :-(

gitbreiti, 2025-04-19:
> Habt ihr das schon mal probiert, hat bei mir bei der Verbindungsoptimierung voll geholfen (Schlauchschelle):
>
> [hoylabs#1519 (reply in [...]

Soooo, ich hab jetzt mal getestet.... 

Ich hatte die ganze Zeit dieses Modul hier im Einsatz: "Generic nRF24L01+ Module with PCB antenna"

Jetzt habe ich mit zum testen mal folgendes bestellt: "EBYTE ML01DP5 module"

Allerdings hat es keine Verbesserung gebracht (abgesehen davon, dass ich die Sendeleistung mit diesem Modul auf Minimun stellen konnte). Weder die Wiederanforderungen haben sich großartig verbessert, noch hat sich etwas an am Zeitabgleich Problem geändert. Ich glaube aber auch nicht an einen Zusammenhang, immerhin sind es wirklich jeden Tag genau 14 Zeitabgleiche zu völlig verschiedenen Zeiten. Heute Morgen waren es z.B. direkt 14 hintereinander und danach passierte dann nichts mehr.

Ich für meinen Teil, werde es halt jetzt einfach als "kosmetisches" Problem ansehen und damit leben.

iomax, 2025-05-14:
Quick & dirty update after about 3 days of testing on version 25.5.10 :

- With version 24.11.7, there were little to no errors (confirmed up to the 25.5.10 test).
- The first two sunny days showed no issues, but the problems returned today, likely due to a partly cloudy sky causing fluctuating production.
- Also in my case (2 inverters: 1 HM-350 and 1 HM-400), the errors seem to occur randomly on one, the other, or both, ~~while the limit requests are always sent in sequence and with the same values to both inverters~~ but the one slightly farther from the ESP (just a couple of meters more than the closer one) seems more prone to errors.
- my personal impression is that version 25.5.10 has reduced the odd behaviour (as of now, still running since this morning, 7 errors on the far inverter and only 2 on the close one), but hasn't completely resolved it.
- System Info does not report any CPU resets that could be associated with the errors.

stefan123t, 2025-05-14:
@iomax ja wenn der Wechselrichter (WR) nur schlecht mit Solarenergie versorgt wird, dann wird er mitunter vergesslich und dabei tritt dann der 2 - Time Calibration "Fehler" in der freien Wildbahn auch gehäuft auf. Das ist eben i.d.R. früh morgens oder spät abends, bzw. wenn starke Bewölkung und/oder Wind die Ertragssituation des WR sprunghaft ändern.

@gitbreiti der WR kann mW nur eine begrenzte Anzahl an Events speichern. Wenn zu viele 2 - Time Calibration Events auftreten, dann schmeisst er die ältesten dieses Event Typs 2 einfach aus dem Log. Die DTU kann die dann auch nicht mehr abfragen.

Es gibt zwar prinzipiell auch Commands um auch die zweite Bank mit weiteren Events abzufragen bzw. die abgefragten Events auf dem WR zu löschen / purgen, aber die hat m.W. noch niemand ausreichend reverse-engineered und in der OpenDTU Software implementiert.

iomax, 2025-05-21:
> [@iomax](https://github.com/iomax) ja wenn der Wechselrichter (WR) nur schlecht mit Solarenergie versorgt wird, dann wird er mitunter vergesslich und dabei tritt dann der 2 - Time Calibration [...]

@stefan123t I can't comment on it definitively, but after spending several days on 25.5.10 and getting lots of 'Time Calibration' errors/warnings, I switched back yesterday — once again — to 24.11.7. Even though the same annoying overcast sky is still there, the inverter hasn’t shown any more errors since. So, from my point of view, some kind of 'magic mix' must have happened after that release, which seems to help the inverter’s tendency to be "mitunter vergesslich".

closed 2025-07-05 by github-actions[bot]