tbnobody/OpenDTU #2511: Home Assistant Integration not displaying the same limit % as the website when over 100%

mathieucarbou, 2025-01-23:
### What happened?

I am playing with Hoymiles HM and limits.

Currently, my inverters are configured like this since they are rated at 400 Wp:

[image]

But I am dubious about this config... What really it is used for ? Because irradiation, if calculated based on it, won't be right and depends on the temperature too (each 25C plus or minus removed or add about 10% yield).

I have this question because I am playing with limitations:

[image]

Here, I have set a limit of 500W. The reason behind that is:

- setting 400 Wp in the inverter config for the nominal power and irradiation calculation, even if not accurate
- setting 500W as the output limit because in Winter an inverter can output more than its nominal power (there is a tolerance usually of about 10% if I remember the HM specs)

When doing that, OpenDTU displays 125% but Home Assistant 25%:

[image]

It kind of make sense to not be able to go over 100%, but if the 100% was the real inverter power limit. The one configured is just the nominal power.

But if I change the nominal power in the config to be 450 for example (which would match a reasonable 100% limit), then what is the effect of that elsewhere ?

So there is currently a bug regarding the percentages, but more than that, I find an incoherence between :
- The nominal power (Wp) configured
- The irradiation calculated based on output power and nominal power
- The limit percentage regarding the nominal power, instead of the hard limit
- Home assistant which is not going over 100%

Thanks a lot for your advices @tbnobody 👍 

Also, FYI, I don't know if you saw the announcement: https://github.com/ESP32Async/ESPAsyncWebServer/discussions/1
Discord: https://discord.gg/7MythzfGDp

### To Reproduce Bug

N/A

### Expected Behavior

N/A

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v24.11.7

### What firmware variant (PIO Environment) are you using?

esp32dev

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

mathieucarbou, 2025-01-23:
I found that after updating the hard limit, the % does not move. So I changed the hard limit back and forth to 450W:

In OpenDTU:

[image]

In HA:

[image]

schlimmchen, 2025-01-23:
> When doing that, OpenDTU displays 112.5% but Home Assistant 25%:

That might be a HomeAssistent issue. If Thomas has a hard time debugging this, I will gladly take a look.

> I am playing with limitation right now to test a future setup where I will have to limit the grid excess, so I will have to update the limits frequently based on the home consumption.

Are you aware of the [daughter project](https://github.com/hoylabs/OpenDTU-OnBattery), which has the Dynamic Power Limiter?

tbnobody, 2025-01-23:
> When doing that, OpenDTU displays 112.5% but Home Assistant 25%:

Please take a look at the raw values on the mqtt broker (e..g. using mqtt explorer) to see whether it's HA specific or if the values on the broker are already wrong.

Second, you cannot set percentage values > 100%. The command does not allow this. It's also disabled here:

https://github.com/tbnobody/OpenDTU/blob/653efb41a23aa19be375d3e0c2a9a58acb8671b5/lib/Hoymiles/src/inverters/HM_Abstract.cpp#L127-L129

But you can set absolut values which are higher then your max inverter value. It will be sent to the inverter (and updated via mqtt immediatly). But after a certain amount of time, the value is fetched from the inverter again and you can see what the inverter made with your command.

tbnobody, 2025-01-23:
> Also, FYI, I don't know if you saw the announcement: https://github.com/ESP32Async/ESPAsyncWebServer/discussions/1

Yes I already saw it and already updated the `platformio.ini` :-)

mathieucarbou, 2025-01-23:
> That might be a HomeAssistent issue. If Thomas has a hard time debugging this, I will gladly take a look.

No, much more the type of sensor : it is defined as percentage, so cannot be more than that. It is not defined as a number.

But I think that the question to answer is much more: should percentages more than 100% be allowed ? If yes, then in HA it has to be a number, not a percentage. If no, then OpenDTU needs to be fixed, but in that case, what is happening when a limit is set with a value more than the detected power of the inverter, like I do ? 

I saw that OpenDTU will detect the power value of the inverter, but does it use it also as a maximum limit ? Or we can actually send to inverters a limit higher than the detected value ?

> Please take a look at the raw values on the mqtt broker (e..g. using mqtt explorer) to see whether it's HA specific or if the values on the broker are already wrong.

yes I can do tomorrow! I completely forgot lol!

> But after a certain amount of time, the value is fetched from the inverter again and you can see what the inverter made with your command.

I will look! For now, I set a persistent limit, and the value of 450W is still kept both in HA and in OpenDTU. This is the night.

[image]

> Are you aware of the [daughter project](https://github.com/hoylabs/OpenDTU-OnBattery), which has the Dynamic Power Limiter?

I know about OpenDTU-OnBattery, but since it has a focus on battery usage and I don't / will not have batteries, I was wondering whether it would fit my need...

What I would need is to integrate OpenDTU with an MQTT topic to read the grid voltage, or, even better, with a JSY-163-T for example (https://github.com/mathieucarbou/MycilaJSY) which will read the grid power (in and out), and then, have a setting in OpenDTU that lets me chose the excess / import I wan to allow: 0 to simulate a zero-inject profile, etc.. 

And if I have a solar diverter (I designed some), we could set for example this number to -300W to allow some excess that will be redirected to the water heater by the solar diverter. If diverting, then opendtu would see that the grid is at 0, it will, if possible update the limit to let another 400W flow, etc. 

That's really the flow that I would need to make, either integrated within OpenDTU or on the side.

I am really not good at UX design, but this is definitely something I could contribute to OpenDTU: JSY support (in my library, I also support a remote mode with UDP in order to have multiple clients reading the grid metrics), coupled with some settings to control the on persistent limit of inverters based on a target.

Typically, this would be implemented like a PID algorithm, like I did for my solar diverters, to make sure the target can be reached better, since computing a difference (tested with my inverters today) does not exactly lead to 0 excess for example and limiting to for example 200W also does lead to more than 200W produced.  There are always about 10, 20W more than the limit that is produced. So this PID could then control the adjustment of the limits depending in the grid power read and the target (setpoint in PID terms).

schlimmchen, 2025-01-23:
> since it has a focus on battery usage

Meh, that's not true, despite the name. We explicitly support (possibly multiple) solar-powered inverters as well.

> What I would need is to integrate OpenDTU with an MQTT topic to read the grid voltage

✅ 

> even better, with a JSY-163-T for example

Possibly supported. I guess this is speaking SML? If not, integrating a new power meter interface should be straight-forward for you 😉 

> have a setting in OpenDTU that lets me chose the excess / import I wan to allow

✅ 

> Typically, this would be implemented like a PID algorithm

Hm, yeah, this is where OpenDTU-OnBattery might fall short on your requirements... I am no good at stuff like this so I did not even attempt something yet. This is my grid power response when enabling and disabling a water heater from yesterday. You are probably turned off by the significant feed into the grid when the heater is disabled. I am curious how you could do much better.

[image]

Know that you will have trouble speeding this up since communication to the inverter is inherently slow. The situation becomes worse when multiple inverters have to adjust their limit to achieve a particular target grid consumption value.

Thomas, please excuse all this advertisement for the daughter project and the off-topic discussion. Mathieu, if you are interested in discussing this further, maybe start a discussion over at OpenDTU-OnBattery 😊

mathieucarbou, 2025-01-23:
Ahah !! Good to know !!  I was searching in the doc but it really focuses on battery integration. I will have a look once everything will be in place.

For the PID, its use does not relate to the required accuracy, or the speed of the changes. It more relates to ability to react more properly to an inaccurate correction. For example, if I set a target excess to -300W, and I need to set the limit to 300W on 10 inverters to arrive to this target, in reality, one inverter will produce 280W, one 330W, one will be under some shade, etc. So the PID will be able to auto-correct such inaccurate adjustment by recalculating a new correction to apply, until the next measurement comes. A PID would also be more tolerant and able to correct the uneven production of inverters, caused by shade, difference of power, orientation, etc.

without PID, if you simply correct by computing a difference and spreading the new limit on inverters, the reality is that the real correction won't be as expected by the calculation because of too many unpredictable factors.

But I am pretty sure that in your code, to do that, you are using some kind of PID, but just more specific, with the Kp term only (which is a correction done based on a percentage of the error == target - current measurement).

The graph that you have is very good I think!
I know that it takes a few seconds for inverters to adhere to a new limit.  So doing that in 5 seconds is nice!

The Kp Ki Kd terms of a PID allows also to tweak its responsiveness.
In your graph example, you seem to have quite stable home consumption after the boiler is disabled.
But imagine that you had someone cooking in the kitchen ;-) You would get a lot of on/off with different appliances. So a PID is also able to "slowdown" to reach a target by smoothing its corrections to avoid these sort of abrupt correction (coming a few seconds later in effect) in the case of a sequence of lot of high / low consumptions. It does that thanks to his integral term and summations.

mathieucarbou, 2025-01-24:
@tbnobody : I can confirm that OpenDTU sends the right % amount to MQTT:

[image]

But HA is stuck at 100%:

[image]

[image]

I will try modify something and raise a PR it it works... An idea could be to remove this `min()` check, and if we see that an absolute value is set, and it is more than the nominal power of the inverter, then we can update the HA discovery topic also to set the max value for the slider to be the new max limit set by the user.

The problem is that the discovery topic is sent with a min set to 0-100%.

So this is not the fault of HA, but how OpenDTU sets the limits for the slider

[image]

tbnobody, 2025-01-24:
> because if right, then, the UI should never display a % more than 100% right ? And this is 100% relative to what ? The detected inverter power ? If yes, then the limit should be allowed to go passed that because an inverter is allowed to output more than its nominal power

The inverter will Never Return a value >100%. The percentage value has nothing to do with any value detected by opendtu. The value ist just sent to the inverter and it applies it in some way. You can send values > 100% but the inverter will never apply them and return something else if it is fetched the next time. 
So it makes sense to limit the current limit also to 100%

mathieucarbou, 2025-01-24:
Ah! And there is another confusion:

```c++
        publishInverterNumber(inv, "Limit NonPersistent Relative", "status/limit_relative", "cmd/limit_nonpersistent_relative", 0, 100, 0.1, "%", "mdi:speedometer", STATE_CLS_NONE, CATEGORY_CONFIG);
        publishInverterNumber(inv, "Limit Persistent Relative", "status/limit_relative", "cmd/limit_persistent_relative", 0, 100, 0.1, "%", "mdi:speedometer", STATE_CLS_NONE, CATEGORY_CONFIG);

        publishInverterNumber(inv, "Limit NonPersistent Absolute", "status/limit_absolute", "cmd/limit_nonpersistent_absolute", 0, MAX_INVERTER_LIMIT, 1, "W", "mdi:speedometer", STATE_CLS_NONE, CATEGORY_CONFIG);
        publishInverterNumber(inv, "Limit Persistent Absolute", "status/limit_absolute", "cmd/limit_persistent_absolute", 0, MAX_INVERTER_LIMIT, 1, "W", "mdi:speedometer", STATE_CLS_NONE, CATEGORY_CONFIG);
```

The UI made me understand that there are 2 different limits:
- the hard one
- the soft one

This is only true when setting a value. 
But when reading  limits, there is only 1 limit, which is the applied current limit (so the one in force now):
- `inv->SystemConfigPara()->getLimitPercent()` and it can be more than 100%

absolute value is computed:

```c++
            uint16_t maxpower = inv->DevInfo()->getMaxPower();
            if (maxpower > 0) {
                MqttSettings.publish(subtopic + "/status/limit_absolute", String(inv->SystemConfigPara()->getLimitPercent() * maxpower / 100));
            }
```

So when I send an absolute value to the inverter, the inverter allows it, and returns a limitPercent() > 100%, which displays a correct value. But when using % the code with the min prevents setting a % higher than 100. Why ? I don't know since it seems to be supported in the inverter I have (Hm400).

Is this constrain there to support inverters that do not allow > 100% ?

mathieucarbou, 2025-01-24:
> The inverter will Never Return a value >100%. The percentage value has nothing to do with any value detected by opendtu. The value ist just sent to the inverter and it applies it in some way. You [...]

That is not my understanding of the code:

- I set a n absolute limit to 450W for a HM400: the code goes to : `sendActivePowerControlRequest()` and sets this limit. I see it also in MQTT. The inverter takes it and the % and absolute value is correctly updated in the UI to 112.50%.

when publishing to MQTT, the code publishing the absolute value is:

```c++
            uint16_t maxpower = inv->DevInfo()->getMaxPower();
            if (maxpower > 0) {
                MqttSettings.publish(subtopic + "/status/limit_absolute", String(inv->SystemConfigPara()->getLimitPercent() * maxpower / 100));
            }
```

which means that the inverter took in consideration the absolute limit > nominal power and returned 112.5%.

tbnobody, 2025-01-24:
> I set an absolute limit to 450W for a HM400: the code goes to : sendActivePowerControlRequest() and sets this limit. I see it also in MQTT. The inverter takes it and the % and absolute value is correctly updated in the UI to 112.50%

That's not right. We are sending a new limit to the inverter and assume it applied it. After ~4-5min the current limit is polled from the inverter (its not possible earlier otherwise it would generate nasty errors in the event log)
That means, if you set 450W the limit goes first up to 112,5% internally in opendtu but will then corrected when the inverter is polled the next time

mathieucarbou, 2025-01-24:
> That means, if you set 450W the limit goes first up to 112,5% internally in opendtu but will then corrected when the inverter is polled the next time

No in my HM400: I set a hard limit to 450W yesterday, it remained and was there this morning...

mathieucarbou, 2025-01-24:
I tested 2 patches.

First one for HA Disovery:

```patch
diff --git a/src/MqttHandleHass.cpp b/src/MqttHandleHass.cpp
index c39485dd..c0481fab 100644
--- a/src/MqttHandleHass.cpp
+++ b/src/MqttHandleHass.cpp
@@ -83,8 +83,8 @@ void MqttHandleHassClass::publishConfig()
         publishInverterButton(inv, "Restart Inverter", "cmd/restart", "1", "", DEVICE_CLS_RESTART, STATE_CLS_NONE, CATEGORY_CONFIG);
         publishInverterButton(inv, "Reset Radio Statistics", "cmd/reset_rf_stats", "1", "", DEVICE_CLS_NONE, STATE_CLS_NONE, CATEGORY_CONFIG);
 
-        publishInverterNumber(inv, "Limit NonPersistent Relative", "status/limit_relative", "cmd/limit_nonpersistent_relative", 0, 100, 0.1, "%", "mdi:speedometer", STATE_CLS_NONE, CATEGORY_CONFIG);
-        publishInverterNumber(inv, "Limit Persistent Relative", "status/limit_relative", "cmd/limit_persistent_relative", 0, 100, 0.1, "%", "mdi:speedometer", STATE_CLS_NONE, CATEGORY_CONFIG);
+        publishInverterNumber(inv, "Limit NonPersistent Relative", "status/limit_relative", "cmd/limit_nonpersistent_relative", 0, 200, 0.1, "%", "mdi:speedometer", STATE_CLS_NONE, CATEGORY_CONFIG);
+        publishInverterNumber(inv, "Limit Persistent Relative", "status/limit_relative", "cmd/limit_persistent_relative", 0, 200, 0.1, "%", "mdi:speedometer", STATE_CLS_NONE, CATEGORY_CONFIG);
 
         publishInverterNumber(inv, "Limit NonPersistent Absolute", "status/limit_absolute", "cmd/limit_nonpersistent_absolute", 0, MAX_INVERTER_LIMIT, 1, "W", "mdi:speedometer", STATE_CLS_NONE, CATEGORY_CONFIG);
         publishInverterNumber(inv, "Limit Persistent Absolute", "status/limit_absolute", "cmd/limit_persistent_absolute", 0, MAX_INVERTER_LIMIT, 1, "W", "mdi:speedometer", STATE_CLS_NONE, CATEGORY_CONFIG);
@@ -232,6 +232,9 @@ void MqttHandleHassClass::publishInverterNumber(
     root["min"] = min;
     root["max"] = max;
     root["step"] = step;
+    if (max > 100 && unit_of_measure == "%") {
+        root["mode"] = "box"; // do not show a slider for possible % going more than 100%
+    }
 
     publish(configTopic, root);
 }
```

Which is basically saying: if we define a number where we have to enter a % and its max allowed value is more than 100%, then switch to a box, not a slider.

And it works.

After the DTU is flashed and once the radio got the inverters limit, I have some consistent values in HA:

[image]

[image]

-----

Then, I did another patch to test your theory about the % value sent to inverters:

```c++
diff --git a/lib/Hoymiles/src/inverters/HM_Abstract.cpp b/lib/Hoymiles/src/inverters/HM_Abstract.cpp
index d58324da..f77689cb 100644
--- a/lib/Hoymiles/src/inverters/HM_Abstract.cpp
+++ b/lib/Hoymiles/src/inverters/HM_Abstract.cpp
@@ -124,9 +124,9 @@ bool HM_Abstract::sendActivePowerControlRequest(float limit, const PowerLimitCon
         return false;
     }
 
-    if (type == PowerLimitControlType::RelativNonPersistent || type == PowerLimitControlType::RelativPersistent) {
-        limit = min<float>(100, limit);
-    }
+    // if (type == PowerLimitControlType::RelativNonPersistent || type == PowerLimitControlType::RelativPersistent) {
+    //     limit = min<float>(100, limit);
+    // }
 
     _activePowerControlLimit = limit;
     _activePowerControlType = type;
```

So from what I observe of all my HM400,  I can set a persistent absolute value of 450W, and it will stay in the inverters, and the inverters will return 112.5% for the % (`inv->SystemConfigPara()->getLimitPercent()`) that the UI computes as 450W (`inv->SystemConfigPara()->getLimitPercent() * maxpower / 100)`)

So with this patch, I am now testing if I can send 115% to the inverter.

So I am entering 115% in the HA box:

[image]

And the inverter switches to 115% / 460W

[image]

and both limits (absolute and relative) are displayed in MQTT:

[image]

So even if I agree with your sentence that the inverter could correct the input user value, I don't see any reason why OpenDTU should clamp the percentage (and weirdly, only the percentage) to 100%, when it could just let go all values and reads back the corrected one after.

If clamping to 100%, then the absolute power also should be clamped to the detected nominal power for consistency right ?

mathieucarbou, 2025-01-24:
@schlimmchen : FYI.
You never had such discussions in OpenDTU-Battery ?
 Using limits is really important to control the excess. So I guess you should have seen such inconsistencies ? How do you control the limits ? You never allow inverters to use their maximal output power and stick to the nominal one 0-100% ?

mathieucarbou, 2025-01-24:
I've opened PR https://github.com/tbnobody/OpenDTU/pull/2513

mathieucarbou, 2025-01-24:
Here is the board I am testing with in HA, with my current PnP installation, before going to a 5.5 kW rooftop one.

[image]

So what's important for me is to be able to globally and manually set a persistent absolute power limit for all inverters, which I do with the number at the top, which automatically updates in a few seconds the inverters and the stats on the left (% and absolute power) are updated in consequence.

That is the base of the panel  I will use to monitor the auto adjustment of the inverters power limit based on the grid power excess / home consumption read from a JSY (local or remote - do not know yet but both are easy to support), to always let some little excess going out as a buffer.

schlimmchen, 2025-01-24:
> You never allow inverters to use their maximal output power and stick to the nominal one 0-100% ?

I will give you the shallow answer for now: I was not aware that the inverter would accept a limit higher than 100% of its nominal output power. @AndreasBoehm probably also is not aware of this.

I just tried something:

[image]

[image]

The inverters don't actually produce more than their nominal power.

However, I am confused that the limit value is shown as it is, because I know specifically that the value shown there is read from the inverter's answer to the limit command. That would mean the inverter receives a new absolute limit higher than its capabilities and does **not** answer with the value capped to its nominal power, but with the value it received, like in an echo.

What I don't know is that other command which Thomas mentioned that is only "allowed" every four minutes or something, which would apparently update the limit shown by OpenDTU to the one actually in effect? I guess I need to restart OpenDTU to see that in action, as it seems the limit shown is not updated. So I initiated a reboot. The limit reads "0" now, as the limit is not known. Okay, so I did not send any new limit to the inverters, but both show the previous non-persistent limit I set earlier. That should mean that the inverter has internalized the actual value I sent, which is higher than its nominal power.

I set 2222W as the **persistent** limit of my HM-1500:
[image]

After restarting the inverter, it produces at 100% of its nominal power:
[image]

After restarting OpenDTU, the limit is unknown:
[image]

I shut down the inverter and wait for OpenDTU to fetch the current limit value:
[image]

So... What I make of this: The inverters accept and even save a limit higher than their nominal power, because they obviously just don't check the value for sanity. The part of the inverter that consumes the limit value, however, obviously is not tricked into trying to produce more than the nominal power.

I guess we should therefor try harder to avoid that the user ever sees a limit value greater than 100%, i.e., cap the limit value to the (eventually) known power output capability. That's the opposite direction of what Mathieu is suggesting in his PR 🤷‍♂ 

> So what's important for me is to be able to globally and manually set a persistent absolute power limit for all inverters

Are you asking me [how to do this in OpenDTU-OnBattery](https://opendtu-onbattery.net/firmware/mqtt_topics/#general_1)? Publish to MQTT topic `powerlimiter/cmd/upper_power_limit` (the docs is outdated, or at least not very accurate, as this topic is applied as the "Maximum Total Output" setting in the DPL).

mathieucarbou, 2025-01-24:
@schlimmchen an inverter can output more than the nominal power, there is a tolerance it seems. especially in winter, with good sun and low temp, It happened to me to produce 420W AC with an HM400 (400W nominal power). So I think the inverter accepts any limit, but will just make sure to clamp the production to this limit.

But for what is worth, at any rate, these spikes only happen for a few seconds since a rate inverter of 400W is not supposed to output more than that. 

I was just talking about the tolerance and the possibility to have more, and the fact that the % do not match the absolute value.

Why would an absolute value be allowed to be set (and kept) with a value being more than the detected power, if the percentage is clamped to 100%.

> That's the opposite direction of what Mathieu is suggesting in his PR 🤷‍♂

I also agree with that approach, because these spikes of production are not important and just last a few seconds. So they are irrelevant.

This ticket is much more about bringing consistency across the values.

schlimmchen, 2025-01-24:
I don't doubt that the inverter is able to produce these spikes of AC output power. Not even spikes: When fed from a battery the inverters produce slightly more power even for extended periods of time. Notice that the output power in my screenshots above is also slightly higher than the nominal power. However, that is also the case for a limit of exactly 100%:

[image]

That is to say: IMHO there is no need to be able to set a limit higher than the nominal power rating.

tbnobody, 2025-01-24:
> However, I am confused that the limit value is shown as it is, because I know specifically that the value shown there is read from the inverter's answer to the limit command.

Nope. The limit command only returns whether it was successful received. It does not return the currently set value. The SystemConfigPara command is used to fetch the the percentage value from the inverter (reading the absolute limit is not possible)

mathieucarbou, 2025-01-24:
So what should be the conclusion ?

- Have the UI and HA display and accept values similar to the inverters (so more than 100%), even if technically, most inverters wont go more than the nominal power for a long period of time ?

- Or clamp the UI and HA values to 100%, which also means adding some sanity checks in case the inverter limit % returned is more than 100%, to clamp it to 100%, to make sure the UI is consistant with the HA slider of 0-100% ?

schlimmchen, 2025-01-24:
> Nope. The limit command only returns whether it was successful received.

Ha... Thanks! I re-read the code with that in mind and I can see it now. Once the command succeeded, the value that was transported to the inverter using that command is then set in the inverter's SystemConfigPara.

> Or clamp the UI and HA values to 100%, which also means adding some sanity checks in case the inverter limit % returned is more than 100%, to clamp it to 100%, to make sure the UI is consistant with the HA slider of 0-100% ?

I prefer that option. There is a catch: The inverter's max power is not known right from the start, it has to be fetched first. In the DPL, inverters are not eligible for use until their max power is known. We'd have to do something similar when processing requests to set absolute limit values...

mathieucarbou, 2025-01-24:
> We'd have to do something similar when processing requests to set absolute limit values...

No, we can let them go as now, It is just the displayed value and the one sent to mqtt that will be corrected.

Anyone, even another system that opendtu could set a value more than the nominal power or a greater percebtage.

But right now, opendtu has different behavior in the display of both information. it clamps the setting of %, but does not clamp the reading.

mathieucarbou, 2025-01-24:
More a max instead of min ;-)

schlimmchen, 2025-01-24:
min(100, 42) = 42 => as expected
min(100, 112) = 100 => as expected

I assume you have the typical brain fuck when using min/max functions? You limit some value to a known max by using the min function.

mathieucarbou, 2025-01-24:
> I assume you have the typical brain fuck when using min/max functions? You limit some value to a known max by using the min function.

ahah! yeah exactly ;-)

So should I update my PR or you raise another one ?

mathieucarbou, 2025-01-24:
Updated https://github.com/tbnobody/OpenDTU/pull/2513

mathieucarbou, 2025-01-24:
I confirm it works! 

I have 125% set in my inverters. After a few minutes, I get 100% 400W

[image]

[image]

[image]

PR #2513 (2025-02-03): fix: Make sure limit values sre consistent and not above 100%
- Fix #2511 

fix: Make sure limit values sre consistent and not above 100%
Ref: https://github.com/tbnobody/OpenDTU/issues/25
closed 2025-02-03 by tbnobody