tbnobody/OpenDTU #2518: OpenDTU does not update summary accross several inverters in the headline automatically anymore

meystervie, 2025-01-27:
### What happened?

After updating the firmware to the latest 9280c14 the information for the individual inverter is automatically updated every 10 seconds. The total summary across all inverters (big headline) is only updated after a manual refresh of the page. Is this normal?
The behaviour was different in the previous firmware versions. 

### To Reproduce Bug

install latest opendtu firmware. Install more than 1 inverter. Wait for automatic refresh. 

### Expected Behavior

Headline summary of data should be updated automatically. 

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

9280c14 

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-01-27:
I can't reproduce this.

Which of these do you mean? All of them?

[image]

> the information for the individual inverter is automatically updated every 10 seconds

How long is your polling interval? Updates should arrive sooner than 10 seconds unless your polling interval is greater.

Are you sure this worked before you updated the firmware? Are you accessing OpenDTU through a proxy?

meystervie, 2025-01-27:
Yes, all three not updated automatically and I am sure, this was updated automatically before. 
Updates are done after 10 seconds. Not proxy, same network. 

Below the screenshot: blue is automatically updated every 10 seconds, red is not updated without manual refresh. 

[image]

schlimmchen, 2025-01-27:
Please try hitting `Ctrl+F5` (`Strg+F5`) in Chrome to force it to re-load the webapplication from the device.

closed 2025-04-18 by github-actions[bot]