tbnobody/OpenDTU #2523: WR Firmware 2.0.4 temporary limit not displaying constantly

agger83, 2025-01-29:
After updating the firmware to 2.0.4, a temporary limit can still be set and used by the inverter. But the display in openDTU is showing the permanent limit after roughly 5 minutes. 

Steps to reproduce:
- set a temporary limit with openDTU, e.g. 5%
- OpenDTU will display this current limit and the inverter will produce accordingly 
- after roughly 5 minutes openDTU will display the permanent limit instead, eg 100%
- the inverter will still product according to the temporary limit until the next restart

So actually this is a display bug in openDTU but must be somehow related to the new firmware of the inverter.

_Originally posted by @agger83 in https://github.com/tbnobody/OpenDTU/discussions/2153#discussioncomment-11994747_

MichaMEG, 2025-02-12:
A permanently set limit via the OpenDTU is also not retained, at least in the display.
After a short time, the limit previously set with the Hoymiles DTU is displayed. I cannot yet say which one is used.

agger83, 2025-02-14:
Interesting. I didn’t check that, but assume it is similarly only a display issue. Maybe you can check this next time when the sun is shining or by setting a real low permanent limit. 
I could see that the inverter is still producing according the maintained limit, even though openDTU is showing the wrong limit. 

Is this topic a bug? How is this flagged as a bug,  is this to be done by an admin? Sorry, I am new to GitHub

agger83, 2025-02-17:
Actually, I am using openDTU on battery and have activated DPL now. This is „healing“ the issue, as soon as openDTU falls back to display the permanent limit (800 W in my case), the DPL recognizes and calculates and sends a new limit. 
Thus this issue is not showing up for me, but is still persistent

chkbln, 2025-03-16:
I can set a temporary limit, but it disappears the next time I reboot. This has been the case since the firmware update to 2.04. Perhaps someone can look into what the problem is?
Best regards

Back-code, 2025-03-20:
I can confirm there is a bug.
WR HMS-2000-4T FW 2.0.4
every 5 min. the limit goes to 100%
It would be great if someone could look into it
Best regards

agger83, 2025-04-08:
Does anyone know how to get this categorized as a bug? @tbnobody  can you help?

herblecupra, 2025-05-13:
I updated also to 2.0.4 and see the same issue. Where this comes from. With that firmware it sets the default limit which was initially set by factory or dtu wlite. To overcome use dtu-lit to set the value you need. As this is then the permanent value. You can set still the temporary limit but after some time it is reset from WR to dtu set value.

stefan123t, 2025-05-18:
This is IMO related to version 3 of the SystemConfigPara response as detailled in https://github.com/tbnobody/OpenDTU/issues/2641#issuecomment-2886433841