tbnobody/OpenDTU #2526: Incorrect Power Readings in Complete Darkness (Phantom Power)

Biospargel, 2025-01-30:
### What happened?

I have noticed that OpenDTU continues to display a power input of 24W (it’s different every time) even in complete darkness. This issue persists despite the inverter being offline for several hours.

The last update from the inverter was at 15:10, yet OpenDTU still displays power input at 20:57. Could this be caused by a caching issue or a bug in power calculation?

### To Reproduce Bug

It’s only sometimes and not really common. But for me it’s annoying because I have charging automations that uses this value.
A restart solves this problem.

### Expected Behavior

That the input value is set to zero after the inverter is shut down.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

5A9F44

### What firmware variant (PIO Environment) are you using?

generic

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-01-30:
>  Could this be caused by a caching issue or a bug in power calculation?

The value is not calculated it is the same value as the inverter supplies..

> That the input value is set to zero after the inverter is shut down.

Its not possible to detect whether the inverter is shut down. It's just known that the inverter is no more reachable.. In this case, just enable the [Zero Runtime Data](https://www.opendtu.solar/firmware/configuration/inverter_settings/#zero-runtime-data) feature.

Elektron79, 2025-02-02:
I‘ve seen the issue also, for Hours the ghost value is shown.
Yes maybe it’s a compromise when to set it to zero, but after hours without update it’s very likely that the inverter is offline due to no power.
I had the issue on days with very low daylight…

[image]
[image]

tbnobody, 2025-02-02:
> Yes maybe it’s a compromise when to set it to zero, but after hours without update it’s very likely that the inverter is offline due to no power.

What else do you expect other then setting it to zero?
I mean if the inverter is really offline, the values would be zero anyways........

schneeer, 2025-02-06:
I've noticed it too.

> I mean if the inverter is really offline, the values would be zero anyways........
If the inverter is not reachable, openDTU should not invent phantom values. 
The inverter doesn't send anything.

schneeer, 2025-02-06:
It may be true that this corrects the ad. But even without this function there shouldn't be any nonsense values ​​after 3 hours of complete darkness.
At 9:00 p.m. the output for me is still 0.7 W DC. The inverter is definitely off. The software knows when the sun has set and you could automatically set the values ​​to 0 at some point. If the inverter is not producing anything, nothing should be displayed.
This is definitely a bug, albeit a very small one.

tbnobody, 2025-02-06:
> But even without this function there shouldn't be any nonsense values ​​after 3 hours of complete darkness.

After what amount of time do you think its the best to set it to zero? on every loss of connection? then it would accour very often...... it depends of the taste of every single user..... therefor there is a option to set this up.

> At 9:00 p.m. the output for me is still 0.7 W DC. The inverter is definitely off. The software knows when the sun has set and you could automatically set the values ​​to 0 at some point.

Why? What happens if someone connects a battery to the inverter?

>  If the inverter is not producing anything, nothing should be displayed.

If it is not producing anything does not means it's off..... you can turn off the AC production side while the inverter stays on..

I for my part like it how it is.... If you don't, there is off course the option to set it to zero :-)

closed 2025-02-06 by tbnobody

schneeer, 2025-02-06:
Einen Bug Report als gelöst zu markieren, weil einem das Thema nicht gefällt, behebt das Problem nicht.
Das ist der falsche Umgang mit dem Thema.

> > Könnte dies durch ein Caching-Problem oder einen Fehler bei der Leistungsberechnung verursacht werden?
> 
> Der Wert ist nicht berechnet, es handelt sich um den gleichen Wert, den der Wechselrichter liefert.

Das wiederspricht sich, wenn der Wechselrichter nicht erreichbar ist, kommen ja keine aktuellen Daten, somit liefert der auch keinen Wert. Daher sollte es 0 sein.
Und wenn er doch an ist, könntenes  ja auch mehr wie die angezeigten 0,2 W pro Eingang sein, und dann?? Schätz du das jetzt für alle?

> Nach welcher Zeitspanne ist es Ihrer Meinung nach am besten, es auf Null zu setzen? Bei jedem Verbindungsverlust? Dann würde es sehr oft vorkommen ... es hängt vom Geschmack jedes einzelnen Benutzers ab ... dafür gibt es eine Option, dies einzurichten.

Bei nicht Erreichbarkeit nach einer Stunde z.b.

>  Was passiert, wenn jemand eine Batterie an den Wechselrichter anschließt?

Dann ist er an und sendet Daten. Das merkt die DTU ja.
Wenn nicht, hilft die ein oder aus geschaltete Funktion "Zero runtime data" auch nicht, da ja nicht bekannt ist wieviel von der Batterie kommt. Dann sind 0,2 W pro Inverteranschluss immer sinnfrei.

> Mir persönlich gefällt es so wie es ist.... Falls nicht, besteht natürlich die Möglichkeit, es auf Null zu setzen :-)

Kann man überdenken. Falsche Daten sind falsche Daten.
Als zusätzliche Option pro String am Wechselrichter, hätte das mehr Sinn.

Eine Behandlung des Fehlers muss ja keine hohe Priorität haben, man kann sich auch stur stellen.

Elektron79, 2025-02-06:
Hallo zusammen,

Noch einmal ein Anlauf zu erklären wie eine mögliche Erkennung aussehen könnte…

Ich habe mehrere Wechselrichter HMS-2000-4t und auch HMS-450-1T in Betrieb.
Das Problem tritt immer dann auf wenn die Sonneneinstrahlung sehr gering ist und die Wechselrichter mangels Energie „absaufen“ und keine Daten mehr senden.

Mein Vorschlag wäre:
Wenn der Wechselrichter nicht wie erwartet erreichbar ist und die letzt bekannte Gesamtleistung des Wechselrichters unter - hier wäre mein Vorschlag - 3 Watt lag (nach meiner Beobachtung brauchen die ungefähr 2,4W damit es los geht), dann setzt die DTU den Wert auf Null.

Wir verlieren im „false positive“ Fall für den Zeitraum bis zu nächsten erfolgreichen Datenübertragung nur wenige Watt (die Energie in Wh wir ja „nachgetragen“) wir vermeiden aber über die gesamte Nacht eine „Erzeugung“ von wenigen Watt die - zumindest bei mir - die Plots komisch aussehen lassen.

Sollte es in einem „stabilen“ Betriebszustand (also richtig Leistung auf dem WR) zu einer Unterbrechung kommen, schlägt die Regel nicht zu.

Mit „Goldkante“ könnte man noch plausibilisieren ob überhaupt noch eine Erzeugung zu erwarten ist (da die Sonne schon längst untergegangen ist) und dann auf jeden Fall auf Null setzen. Das könnte entweder aus der Position und dem Datum errechnet werden oder aus dem „Produktionsende“ des Vortags +x Minuten.

Wir müssen einfach mit dem Leben was uns die Wechselrichter bieten, die Übertragung ist nicht 100%ig stabil, da braucht es nach meiner Meinung etwa Kreativität.
Wie oben angedeutet, sehe ich in meinem Vorschlag eine kleinere Ungenauigkeit als wir sie haute haben.

Viele Grüße Michael

tbnobody, 2025-02-07:
Das von dir gewünschte Verhalten ist schon implementiert. Du musst die Funktion nur aktivieren. Dann verhält sich die dtu genau so wie von dir beschrieben. 
Eine Implementierung ist nicht nötig. Nur ein einziger Klick.

schlimmchen, 2025-02-07:
> Das ist der falsche Umgang mit dem Thema.
> man kann sich auch stur stellen.

Mit dieser herablassenden, belehrenden, einfordernden Art und Weise wirst es schwer jemanden zu überzeugen, etwas für dich zu tun -- wenn es denn hier etwas zu tun gäbe. Den Schalter in den Einstellungen müsst ihr halt selbst umlegen.

schneeer, 2025-02-07:
Ohne Umlegen des Schalters Werte anzuzeigen, die nicht der Realität entsprechen, entbehrt jeglicher Logik.

tbnobody, 2025-02-07:
> Ohne Umlegen des Schalters Werte anzuzeigen, die nicht der Realität entsprechen

Klar entsprechen die der Realität. Man muss die Werte nur richtig lesen. Es steht das alter der Daten dabei. Zu diesem Zeitpunkt waren die Werte korrekt. Wenn dir diese Ansicht nicht zusagt darfst du sie gerne ändern. Ich mag sie wie sie ist. Ich sehe gerne den Stand der letzten Datenübertragung.

ms49434, 2025-02-07:
> Ohne Umlegen des Schalters Werte anzuzeigen, die nicht der Realität entsprechen, entbehrt jeglicher Logik.

Den zuletzt übermittelten Wert anzuzeigen ist weder falsch, unlogisch noch realitätsfern.

schneeer, 2025-02-07:
Wenn der letzte Wert Stunden her ist, ist es realitätsfern.
Wenn vom WR keine Daten kommen sollte die Anzeige 0 sein. Die Anzeigefelder sind ja für die aktuellen Werte.
Sobald wieder Daten eintreffen, wird die Ansicht aktualisiert. 
Die zuletzt übermittelten Werte könnten ja zusätzlich  oder per Button eingeblendet werden.
Vielleicht will ja der eine oder andere den Schalter nicht umlegen, damit der Tagesertrag zum Beispiel nicht zu früh gelöscht wird.

ms49434, 2025-02-07:
WoT, eigentlich willst du nur, daß die vorhandene Logik umgekehrt wird und die Standardeinstellung auf 2 Kontaktversuche eingestellt und "Nulle Laufzeit Daten" aktiviert ist.

Aktiviere es und gut is.

schlimmchen, 2025-02-07:
@tbnobody 

[image]

Ich hab diesen Knopf noch nicht benutzt... Zeig uns doch mal, wie das aussieht, wenn man den drückt :grin:

ms49434, 2025-02-07:
Kann OpenDTU auch Außenthermometer? Wie hast Du das hinbekommen? Wäre nett wenn du mal eine Kurzanleitung posten könntest.
BTW, lese dich bitte nicht in MQTT ein, eine Welt könnte für dich zusammenbrechen.

schneeer, 2025-02-07:
> Kann OpenDTU auch Außenthermometer? Wie hast Du das hinbekommen? Wäre nett wenn du mal eine Kurzanleitung posten könntest. BTW, lese dich bitte nicht in MQTT ein, eine Welt könnte für dich zusammenbrechen.

Ja es kann. Hier sind es 4.4°C. Da braucht's Anleitung, sieht man in der Übersicht...
Ist aber nicht das Thema. Es ging um die Anzeige.

Juergen2453, 2025-02-07:
> Ich vergleiche das mal mit einem Außenthermometer.

Was meinst du wohl wofür der Knopf in der Mitte ist.
Richtig **"zurücksetzen"**

[image]

schneeer, 2025-02-07:
Der ist gut. Und ich soll sachlich bleiben, oder was?

schneeer, 2025-02-07:
Oh man.
War ein Beispiel.
Dafür hätte ich auch ein Foto:
[image]

Vielleicht könnte die Anzeige der Werte irgendwie trotzdem modifiziert werden.
Schon klar das das Status/ Übersichtsfeld rot ist und hiermit deutlich gemacht wird, das der WR funktechnisch nicht erreichbar ist.
Trotzem ist es verwirrend, wenn der letzte empfangene Wert dauerhaft angezeigt wird. Durch das Löschen der Laufzeitdaten wäre der ja komplett weg. 
Vorstellen könnte ich mir eine Anzeige mit "0" als Wert und irgend etwas Richtung Button mit: letze empfangene Werte anzeigen. 
Das könnte auch nach einer Weile wieder ausgeblendet werden. Dann hätte man beides erstmal ohne Schallter. Und den zusätzlich benutzen kann wer will ja auch.

schneeer, 2025-02-07:
> Leider ist es in der Realität so, daß nur der letzte übertragene Wert der einzige verlässliche Wert ist. 

Nur zum Zeitpunkt wo er übertragen wurde.

> Alles andere sind Vermutungen, Annahmen oder Spekulationen.

Das ist bei  fehlenden neuen Daten generell so.

> Wenn es tagsüber stockfinster wird und der WR nicht mehr Solarenergie bekommt um Energie zu produzieren, bedeutet es dann implizit, daß es Nacht ist?

Natürlich nicht. Die DTU weiß doch welcher Tag ist und wie spät es ist. Wenn der Wechselrichter wieder Daten sendet soll es wieder da stehen. Die Haltezeit bis nichts mehr angezeigt wird könnte konfigurierbar sein. Am steigenden Tagesertrag sieht man ja auch, das produziert wird.

> "--" anzuzeigen macht Sinn, da der aktuelle Status unbekannt ist.
> Den Wert "0" anzuzeigen ist grob fahrlässig, solange du dir nicht 100% sicher bist.

Da gehe ich mit.

tbnobody, 2025-02-07:
> Die Haltezeit bis nichts mehr angezeigt wird könnte konfigurierbar sein.

Schaue dir doch bitte mal die Optionen an die es schon gibt. Auch diese Konfigurationsmöglichkeit ist schon vorhanden.
Ich werde diesen Issue nun sperren. Wenn jemand mit der aktuellen Situation (die genau euren Use-Case abbilden kann, man muss es nur machen!) nicht zufrieden ist, darf gerne einen Fork erstellen. Dafür ist Open Source da.