tbnobody/OpenDTU #2535: openDTU crash with 4x HM-1500 from releases after 24.9.30

plieven, 2025-02-09:
### What happened?

openDTU 25.2.3 crashes periodically (every 30-60 seconds) with 4x HM-1500 attached. With only 2 of the HM 1500 it seems to work.

I was on 24.5.6 before. That worked flawlessly for months. I have gone back from 25.2.3 through all minor releases back to 24.9.30. 24.9.30 is now working for a few hours.

I once obseverd very high heap fragmentation levels (>70%) shortly before the esp crashed. Afaik this is only estimated on ESP32, but might give a hint.

ESP ist ESP32-D0WD-V3.

### To Reproduce Bug

OpenDTU on ESP32-D0WD-V3 with more than 2x HM-1500 and software release newer than 24.9.30.

### Expected Behavior

ESP should not periodically crash.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

25.2.3

### What firmware variant (PIO Environment) are you using?

opendtu-generic.bin

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-02-09:
Please provide a log of the [serial console](https://www.opendtu.solar/firmware/howto/serial_console/) when the bug occours.. Otherwise it will not be possible to trace this issue.

plieven, 2025-02-11:
I will try to attach a serial console during the week. The DTU is installed outdoor so I have to dismount it first.

plieven, 2025-02-14:
It took a little longer than expected because I had to find out that the CH340 was broken. I first hat to order a new ESP board.

`RX Period End
All missing
Nothing received, resend whole request
TX AlarmData Channel: 23 --> 15 83 77 57 42 80 16 24 32 80 11 00 67 AE F8 5A 00 00 00 00 00 00 00 00 3B FF 4A 
RX Period End
All missing
Nothing received, resend whole request
TX AlarmData Channel: 40 --> 15 83 77 57 42 80 16 24 32 80 11 00 67 AE F8 5A 00 00 00 00 00 00 00 00 3B FF 4A 
RX Period End
All missing
Nothing received, resend whole request
TX AlarmData Channel: 61 --> 15 83 77 57 42 80 16 24 32 80 11 00 67 AE F8 5A 00 00 00 00 00 00 00 00 3B FF 4A 
RX Period End
All missing
Nothing received, resend whole request
TX AlarmData Channel: 75 --> 15 83 77 57 42 80 16 24 32 80 11 00 67 AE F8 5A 00 00 00 00 00 00 00 00 3B FF 4A 
Interrupt received
RX Channel: 75 --> 95 83 77 57 42 83 77 57 42 01 00 01 80 01 00 01 65 FA 65 FA 00 00 00 00 80 24 B1 | -80 dBm
Interrupt received
RX Channel: 75 --> 95 83 77 57 42 83 77 57 42 03 69 E9 FF FF FF F9 80 02 00 04 6A 0A 6A 0A 00 00 96 | -80 dBm
Interrupt received
RX Channel: 75 --> 95 83 77 57 42 83 77 57 42 04 00 07 80 02 00 05 6A 10 6A 10 FF FF FF FA 80 02 96 | -80 dBm
Fetch inverter: 116185122930
Request device info
Queue size - NRF: 9 CMT: 0
Interrupt received
RX Channel: 3 --> 95 83 77 57 42 83 77 57 42 05 00 06 6A 32 6A 32 00 00 00 06 80 02 00 07 6A 37 48 | -80 dBm
Interrupt received
RX Channel: 75 --> 95 83 77 57 42 83 77 57 42 06 6A 37 FF FF FF FB 80 02 00 08 6A 5A 6A 5A 00 00 40 | -80 dBm
E (145946) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (145946) task_wdt:  - async_tcp (CPU 1)
E (145946) task_wdt: Tasks currently running:
E (145946) task_wdt: CPU 0: IDLE0
E (145946) task_wdt: CPU 1: async_tcp
E (145946) task_wdt: Aborting.

abort() was called at PC 0x4012965c on core 0

Backtrace: 0x40083d9d:0x3ffbed3c |<-CORRUPTED

ELF file SHA256: cc2702188b671a14

E (10602) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (708) esp_core_dump_flash: No core dum���ѥѥ���found!
E (708) esp_core_dump_flash: No core dump partition found!

Starting OpenDTU
Initialize FS... done
Reading configuration... Check for default DTU serial... done
done
Reading language pack... done
Reading PinMapping... [   202][E][vfs_api.cpp:105] open(): /littlefs/pin_mapping.json does not exist, no permits for creation
using default config done
Initialize Network... done
Setting Hostname... Configuring WiFi STA using new credentials... done
Initialize NTP... done
Initialize SunPosition... done
Initialize MqTT... done
Initialize WebApi... done
Initialize Display... done
Initialize LEDs... done
Initialize Hoymiles interface... NRF: Connection successful
  Setting radio PA level... 
  Setting DTU serial... 
  Setting poll interval... 
  Adding inverter: 116185122930 - xxx
  Adding inverter: 116183773950 - xxx
  Adding inverter: 116184895235 - xxx
  Adding inverter: 116183775742 - xxx
done
Switch to WiFi mode
Setting Hostname... done
Configuring WiFi STA using new credentials... done
Configuring WiFi STA DHCP IP... done
WiFi connected
WiFi got ip: 192.168.178.128
Network connected
`

plieven, 2025-02-14:
@tbnobody its scraping of /api/prometheus/metrics that causes the reset. I will try to figure out whats going wrong.

plieven, 2025-02-14:
I bisected the issue down to this:

28788070b287f5ef6af25d9310480d8bef6ccc3e is the first bad commit
commit 28788070b287f5ef6af25d9310480d8bef6ccc3e
Author: Thomas Basler <thomas@familie-basler.net>
Date:   Fri Nov 1 21:56:41 2024 +0100

    Upgrade ESPAsyncWebServer from 3.3.17 to 3.3.21

 platformio.ini | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

plieven, 2025-02-14:
The internal buffer for the AsyncWebServerResponse was changed from a standard char buffer to StringStream between those releases. I will first try 3.7.0 and then if reverting from StringStream back to a char buf the next days.

nakott, 2025-02-17:
In the past, I also saw unexpected openDTU restarts. I'm using one HM-1500. I recorded my observation since mid of last year:

opendtu-generic_240930.bin ->sometimes unexpected restart during night
opendtu-generic_241015.bin ->so far so good
opendtu-generic_241107.bin ->sometimes unexpected restart during night
opendtu-generic_250114.bin ->so far so good
opendtu-generic_250203.bin ->sometimes unexpected restart during night

Currently I'm back on 250114. I'm using a LAN OpenDTU HW.

plieven, 2025-02-27:
@nakott are you scraping /api/prometheus/metrics ?

nakott, 2025-02-27:
@plieven 

> are you scraping /api/prometheus/metrics ?

no. I'm using the WebGUI of OpenDTU and in batch api/livedata/status?inv=xxxxxxx

plieven, 2025-02-27:
Okay, but this also involves the webserver.

Can you check if your setup is stable with 24.9.30?

Unfortunately I had not time to analyse the root cause for my issue in the ESPAsyncWebserver yet.

nakott, 2025-02-28:
@plieven
Please see above https://github.com/tbnobody/OpenDTU/issues/2535#issuecomment-2662322537
But, I installed to 24.9.30 for you and will check, if I see the problem or if my recordings are wrong. 
I will come back to you in a couple of days.

plieven, 2025-02-28:
Sorry, i missed that.

The issue I found was introduced after 24.9.30,
but it is perfectly possible that we have different issues.

Pantastisch, 2025-03-05:
> [@plieven](https://github.com/plieven)
> 
> > are you scraping /api/prometheus/metrics ?
> 
> no. I'm using the WebGUI of OpenDTU and in batch api/livedata/status?inv=xxxxxxx

I can confirm, that scraping the prometheus endpoint causes the system to get unstable as soon as the inverters start to produce in the morning. 
My observing also showed, that with three inverters, the system seems to be stable while scraping the metrics endpoint. With four inverters, it gets unstable and reboots every few seconds with `task_wdt: - async_tcp (CPU 1)`. 

During night, the uptime increases, thus there must also be a correlation between the number of producing inverters while scraping.

plieven, 2025-03-05:
@Pantastisch As mentioned earlier there was a subtle change in the ESPAsyncWebServer after 3.3.17 regarding the internal buffer from a char buffer to StringStream. I learned in another project that this can cause issues. I hope I soon find the time to revert this change on top of the current master of ESPAsyncWebServer to confirm my assumed root cause.

Even if scraping the metrics does not cause a wdt reset I have seen that the processing time for the metrics page has dramatically increased after opendtu 24.9.30.

nakott, 2025-03-08:
> The issue I found was introduced after 24.9.30,
> but it is perfectly possible that we have different issues.

As promised, I ran my OpenDTU with 240930. Within one week, I did not see a restart. I corrected my recordings above.

nakott, 2025-04-07:
I have tried the new version 250404 and I have the same problem as described above: 
https://github.com/tbnobody/OpenDTU/issues/2535#issuecomment-2662322537

opendtu-generic_250404.bin ->sometimes unexpected restart during night (values are zero afterwards)

HW:
Chip-Modell = ESP32-D0WD
Chip-Kerne = 2
CPU-Frequenz = 240 MHz
Flash-Speichergröße = 4M
LAN Port for communication

plieven, 2025-04-07:
I‘m sorry I still hadn‘t the time to test my potential patch idea for this. Maybe this week. Are you as to compile the binary yourself or would you need a precompiled version?

nakott, 2025-04-07:
I took the precompiled version opendtu-generic.bin. I have no environment to compile by my own.

plieven, 2025-04-07:
@nakott can you give this a try: https://github.com/plieven/OpenDTU/releases/tag/v25.4.4-with-AsyncResponse-fix

plieven, 2025-04-07:
@Pantastisch  could you also try my version with reverted StringStream usage for buffers in ESPAsyncWebserver please.

tbnobody, 2025-04-07:
Seems like the promeutheus output gets so large that the memory on the ESP exhausts.. Even if it works with an older version of the espasyncwebserver it's not the solution. to get rid of this problems in the web interface we splitted the response per inverter. Have to think if this is possible in some way for the prometheus response as well.

plieven, 2025-04-07:
@tbnobody I really do not think that this is a memory issue - at least in my case. Here with 4x HM1500 the prometheus output is just 22kB. I have learned the hard way in another project that StringStream is horribly slow. Just benchmark the time until the result is completely loaded for a case where the ESP does not trigger a WDT reset and you will notice that there is a huge difference between your master and the version with the patched ESP Webserver version. You sometimes watch the seperated chunks coming in like on an old modem connection.
Just to be clear, its not an old ESPWebserver version I just turned back the contect buffer in AsyncWebserverResponse back to a char buf from StringStream.

tbnobody, 2025-04-07:
> @tbnobody Maybe the OOM is just a symptom of the slow response speed?

Thats maybe possible. But then the question is why the wdt was not triggered here (I tested it by using a code modification which outputs always 10 inverters)?

Maybe a lot of concurrent calls which where handled only very slowly caused the OOM.

plieven, 2025-04-07:
@mathieucarbou the rationale was that I bisected the issues in OpenDTU down to the change in ESPAsyncWebServer that introduced StringStream instead of cbuf. If I use both master of OpenDTU and master of ESPAsyncWebServer the responses are really slow and often cause WDT resets. If I just put the patch that reverts StringStream back to cbuf as content buffer inside AsyncResponseStream on top of ESPAsyncWeberServer master the issue is gone.

mathieucarbou, 2025-04-07:
Ok.

Were you able to write a MRE with/without the patch to showcase the memory / slow issue ? 
Because there's nothing to look or fix on our side without an issue that cannot be reproduced  outside of opendtu app.

How can you tell that nothing else on opendtu impacted your test and these where identical tests ?

What part of cbug makes it faster, or which part of StringStream makes it slower ?

I am looking at the code:
- WebApiPrometheusClass::addField
- WebApiPrometheusClass::addPanelInfo
- WebApiPrometheusClass::onPrometheusMetricsGet

IMO his endpoint is problematic because:
1) fragmenting the heap
2) using a lot of heap on each prometheus poll
3) slow

maybe it would be better to move it to a chunk response ? Or use request continuation to prepare the response in the background, on flash, on heap or on a pre-allocated fixed and reused buffer, and then send the content once done.

schlimmchen, 2025-04-07:
> If you have some callbacks dealing with mutexes (like config lock), do not lock (because it can wait indefinitely).
> But like the above answer: async_tcp callback MUST BE FAST.

[Noted](https://github.com/hoylabs/OpenDTU-OnBattery/issues/1833). Sound like a lot of work 😔

mathieucarbou, 2025-04-07:
@plieven @schlimmchen @tbnobody I think I might have found the issue.

```
size_t AsyncResponseStream::_fillBuffer(uint8_t *buf, size_t maxLen) {
  return _content.readBytes((char *)buf, maxLen);
}
```

`readBytes()` is doing a timed read: meaning that if maxLen is more than the content buffer, it will wait for the default timeout of Stream class (1 sec).

If this issue happens, it means that maxLen is more. I will need to check why...

Could you please try to add in the constructor of `AsyncResponseStream`:

`_content.setTimeout(0)`

To confirm if this is because of the timeout ?

That's one diff between the StringStream and cbuf.

Please let me know. Thanks!

tbnobody, 2025-04-07:
Will have a look at it tomorrow if the inverter is online again. maybe @plieven you can also doublecheck it?

mathieucarbou, 2025-04-07:
I suspect maxLen is more than the remaining bytes to read because the buffer you send is huge (40k), which is more than the MTU size and probably more than the PCB client size, which means bytes would be flushed on the network, before reading new bytes from the buffer...

plieven, 2025-04-07:
@mathieucarbou StringStream seems like the obvious choice. I already compiled the setTimeout(0) in, but I have to wait until the inverters are producing again.

@mathieucarbou would you also mind including a getter for _contentLength: https://github.com/plieven/ESPAsyncWebServer/commit/3270df2cef901dcf8552ae9a990950ace2e37751
I can send a PR for that.

@tbnobody this would allow us to add some logic like this: https://github.com/plieven/OpenDTU/commit/bf19d5ce4dbde70e6aefcd182ca00daceaa8782e

mathieucarbou, 2025-04-07:
> [@mathieucarbou](https://github.com/mathieucarbou) would you also mind including a getter for _contentLength: [plieven/ESPAsyncWebServer@3270df2](https://github.com/plieven/ESPAsyncWebServer/commit/3270df2cef901dcf8552ae9a990950ace2e37751)
> I can send a PR for that.

maybe more something like `available()` which returns he available bytes to read in the buffer. Content length being conventionally reserved to the http header sent, and is computed and set when response is sent.

I will raise some PRs in the project for the team to review.

mathieucarbou, 2025-04-07:
- https://github.com/ESP32Async/ESPAsyncWebServer/pull/149: `AsyncResponseStream::available()`
- https://github.com/ESP32Async/ESPAsyncWebServer/pull/148: Revert to cbuf

plieven, 2025-04-08:
That was fast, thank you.

Even with setTimeout(0) I still see WDT resets as soon as the endpoint is scraped.

mathieucarbou, 2025-04-08:
FYI, if you are interested into why StreamString is slow, you can read here: https://github.com/ESP32Async/ESPAsyncWebServer/pull/148#issuecomment-2786069763

tbnobody, 2025-04-08:
> I would suggest to cut a new release with the new ESPAsyncWebServer version maybe with the patch included to limit the initial buffer.

I agree. Lets fix this now with the current approach and implement a better solution later.

plieven, 2025-04-08:
PR #2633 opened.

closed 2025-04-08 by plieven