tbnobody/OpenDTU #2543: openDTU won't connect to Unifi AP AC Pro

plieven, 2025-02-12:
### What happened?

I am unable to connect the openDTU to a network that is based on Unifi AP AC Pro. The network just stays in status inactive. Unfortunately, I am not able to have a look at the debug console as the openDTU device is several hunded kilometers away. But with same credentials it worked flawlessly at my office.

What I tried so far:
 - downgraded from 25.2.3 to 24.6.29 (as this was mentioned to solve stability issues in another thread)
 - disabled all advanced Features including BSS Transition, Band Steering, Minimum RSSI, PMF, Fast Roaming...
 - removed the SSID from all APs but one
 - enabled only 2.4GHz
 - enabled Enhanced IoT compatiblity
 - switched security to Open
 - setting a static IP

What I've not tried so far is any code modifications. I found a suggestion to basically use the following code:
`WiFi.persistent(false);
WiFi.mode(WIFI_STA);
WiFi.setSleep(false);
WiFi.begin(ssid, password);`

which conflicts with the WIFI_AP_STA approach to keep an AP as fallback and might lock me out of the device. But this might also be a cause of the issue.

Other ESP32 devices e.g. a Shelly Plus Plug S works flawlessly in the network so I suspect its a software issue.

Any suggestions?

### To Reproduce Bug

Try to connect openDTU 25.2.3 on a Unifi AP AC Pro network.

### Expected Behavior

The openDTU should connected to a Unfi Network.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

25.2.3

### What firmware variant (PIO Environment) are you using?

generic

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

ms1958, 2025-02-13:
@plieven : My Unifi AP AC Pro work with my OpenDTU (V24.6.29)
connect with WPA2

plieven, 2025-02-13:
Could your share the AP firmware version and the exact Wifi Settings?

Am I missing something or is there no Wifi Settings in openDTU except for Wlan-SSID and Wlan-Password?

WPA2 is selected in the Unifi Wifi Settings.

plieven, 2025-02-13:
Does anyone know if the ESP32 Wifi Stack can handle multiple SSIDs on the same BSSID properly?

ms1958, 2025-02-13:
@plieven wifi settings:
[image]
AP-Firmware is 6.6.77.15402 , unifi Network Software 9.0.114

plieven, 2025-02-13:
@tbnobody regarding the Wifi code I have a few concerns:

a) the common approach for a fall back AP i have seen is that you try to connect to a wifi (if one is configured). If this does not succeed you fire up a fallback AP. If noone connects to this AP within a given time you restart scanning and try to connect. The same logic could apply if one is disconnected from the wifi.

What was the reasoning to have both in parallel? I'm concerned that this might cause issues. The fallback AP starts at channel 1 and as soon as the network is connected the AP will change to the channel of the network.

b) I have seen that there is a WiFi.disconnect(true, false) in the reconnect path. Won't this also disconnect clients from the fallback AP?

c) If you clear the Wifi Config from the ESP at startup with WiFi.disconnect(true, true) why don't you set WiFi.persistent(false); to no write to the NVRAM at all?

plieven, 2025-02-14:
I bisected the issue down to his:

28788070b287f5ef6af25d9310480d8bef6ccc3e is the first bad commit
commit 28788070b287f5ef6af25d9310480d8bef6ccc3e
Author: Thomas Basler <thomas@familie-basler.net>
Date:   Fri Nov 1 21:56:41 2024 +0100

    Upgrade ESPAsyncWebServer from 3.3.17 to 3.3.21

 platformio.ini | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

meckiemac, 2025-02-16:
> Does anyone know if the ESP32 Wifi Stack can handle multiple SSIDs on the same BSSID properly?

yes. I have the same setup with few AC Pros running successfully. What could be worth checking is the 2.4GHz channel bandwidth. ESPs may have an issue above 40MHz, to be save stay at 20MHz.

plieven, 2025-03-10:
I managed to get on site and found out that it was a very stupid config mistake in the Unifi Network Manager. The SSID name had a hidden space in the config. The Unifi Network Manager didn't warn about this and all other devices except my MacBook obviously don't care and don't show the space, but OSX does. Sorry for the noise!

closed 2025-03-10 by plieven