tbnobody/OpenDTU #2554: Unexpected reboots

ms49434, 2025-02-22:
### What happened?

I am encountering unexpected reboots with the latest version of OpenDTU (v25.2.3) on a Fusion board. I was able to log the panic messages from the serial console. The event happened at 5:08, just in case it is important.

### To Reproduce Bug

There's no specific way of reproducing the bug. The reboot randomly occurs.

### Expected Behavior

OpenDTU should run without panicing.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v25.2.3

### What firmware variant (PIO Environment) are you using?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x42079752  PS      : 0x00060b34  A0      : 0x820799f5  A1      : 0x3fca81c0
A2      : 0x00c19081  A3      : 0x01000000  A4      : 0x3fca119c  A5      : 0x00060b23
A6      : 0x00060b20  A7      : 0x00000001  A8      : 0x60008800  A9      : 0x00000001
A10     : 0x3fcef940  A11     : 0x3fcef904  A12     : 0x00000000  A13     : 0x420faef8
A14     : 0x02c95bfc  A15     : 0x00ffffff  SAR     : 0x0000001c  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace: 0x4207974f:0x3fca81c0 0x420799f2:0x3fca81e0 0x42057c05:0x3fca8200 0x420cff4a:0x3fca8220 0x420cfe0f:0x3fca8240 0x420cb991:0x3fca8260 0x420c2d23:0x3fca8280 0x420faef8:0x3fca82c0

Core  1 register dump:
PC      : 0x403806ef  PS      : 0x00060c34  A0      : 0x82079901  A1      : 0x3fcebe30
A2      : 0x3fc95bfc  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060c23
A6      : 0x00060c20  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0xffffffff
A10     : 0x00060a23  A11     : 0x00000000  A12     : 0x00060a20  A13     : 0x0000002f
A14     : 0x02c95bfc  A15     : 0x00ffffff  SAR     : 0x00000019  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0x00000000

Backtrace: 0x403806ec:0x3fcebe30 0x420798fe:0x3fcebe70 0x42053110:0x3fcebe90 0x42046dec:0x3fcebeb0 0x4200d1cc:0x3fcebef0 0x420f4b61:0x3fcebf50 0x4202a9ce:0x3fcebf70 0x4202ab49:0x3fcebf90 0x4202ab62:0x3fcebfb0 0x4204c775:0x3fcebfd0

ELF file SHA256: f5504a548b374d56

E (5719) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

ms49434, 2025-02-22:
Another capture from 5:10pm today:
```
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x403806f1  PS      : 0x00060634  A0      : 0x82079901  A1      : 0x3fcebe30  
A2      : 0x3fc95bfc  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060623  
A6      : 0x00060620  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0xffffffff  
A10     : 0x00060423  A11     : 0x00000000  A12     : 0x00060420  A13     : 0x0000002f  
A14     : 0x02c95bfc  A15     : 0x00ffffff  SAR     : 0x00000019  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0x00000000  

Backtrace: 0x403806ee:0x3fcebe30 0x420798fe:0x3fcebe70 0x42053110:0x3fcebe90 0x42046dec:0x3fcebeb0 0x4200d1cc:0x3fcebef0 0x420f4b61:0x3fcebf50 0x4202a9ce:0x3fcebf70 0x4202ab49:0x3fcebf90 0x4202ab62:0x3fcebfb0 0x4204c775:0x3fcebfd0

Core  0 register dump:
PC      : 0x42079752  PS      : 0x00060f34  A0      : 0x820799f5  A1      : 0x3fca81c0  
A2      : 0x00c19084  A3      : 0x01000000  A4      : 0x3fca119c  A5      : 0x00060f23  
A6      : 0x00060f20  A7      : 0x00000001  A8      : 0x60008800  A9      : 0x00000001  
A10     : 0x3fcef940  A11     : 0x3fcef904  A12     : 0x00000000  A13     : 0x420faef8  
A14     : 0x02c95bfc  A15     : 0x00ffffff  SAR     : 0x00000008  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x4207974f:0x3fca81c0 0x420799f2:0x3fca81e0 0x42057c05:0x3fca8200 0x420cff4a:0x3fca8220 0x420cfe0f:0x3fca8240 0x420cb991:0x3fca8260 0x420c2d23:0x3fca8280 0x420faef8:0x3fca82c0

ELF file SHA256: f5504a548b374d56

E (8025) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

nakott, 2025-02-24:
Maybe related: https://github.com/tbnobody/OpenDTU/issues/2535#issuecomment-2662322537

schlimmchen, 2025-02-26:
Regarding the backtrace(s) in the issue description:
```
0x4207974f: temperature_sensor_get_raw_value at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32s3/sar_periph_ctrl.c:170
0x420799f2: temp_sensor_get_raw_value at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32s3/sar_periph_ctrl.c:187 (discriminator 2)
0x42057c05: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/freertos_hooks.c:48
0x420cff4a: ram_temp_to_power at ??:?
0x420cfe0f: ram_tx_pwctrl_background at ??:?
0x420cb991: get_phy_version_str at ??:?
0x420c2d23: lmacProcessTxComplete at ??:?
0x420faef8: ppTask at ??:?

0x403806ec: compare_and_set_native at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
 (inlined by) spinlock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
 (inlined by) xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:301
0x420798fe: temperature_sensor_power_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32s3/sar_periph_ctrl.c:140
0x42053110: temp_sensor_start at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/esp32s3/rtc_tempsensor.c:96
0x42046dec: CpuTemperatureClass::read() at /home/beki/Documents/OpenDTU-OnBattery/lib/CpuTemperature/src/CpuTemperature.cpp:41
0x4200d1cc: MqttHandleDtuClass::loop() at /home/beki/Documents/OpenDTU-OnBattery/src/MqttHandleDtu.cpp:47
0x420f4b61: void std::__invoke_impl<void, void (MqttHandleDtuClass::*&)(), MqttHandleDtuClass*&>(std::__invoke_memfun_deref, void (MqttHandleDtuClass::*&)(), MqttHandleDtuClass*&) at /home/beki/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/invoke.h:70
 (inlined by) std::__invoke_result<void (MqttHandleDtuClass::*&)(), MqttHandleDtuClass*&>::type std::__invoke<void (MqttHandleDtuClass::*&)(), MqttHandleDtuClass*&>(void (MqttHandleDtuClass::*&)(), MqttHandleDtuClass*&) at /home/beki/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/invoke.h:95
 (inlined by) void std::_Bind<void (MqttHandleDtuClass::*(MqttHandleDtuClass*))()>::__call<void, , 0u>(std::tuple<>&&, std::_Index_tuple<0u>) at /home/beki/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/functional:400
 (inlined by) void std::_Bind<void (MqttHandleDtuClass::*(MqttHandleDtuClass*))()>::operator()<, void>() at /home/beki/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/functional:484
 (inlined by) std::_Function_handler<void (), std::_Bind<void (MqttHandleDtuClass::*(MqttHandleDtuClass*))()> >::_M_invoke(std::_Any_data const&) at /home/beki/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:297
0x4202a9ce: std::function<void ()>::operator()() const at /home/beki/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:687
0x4202ab49: Scheduler::execute() at /home/beki/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/TaskScheduler/src/TaskScheduler.h:1552
 (inlined by) Scheduler::execute() at /home/beki/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/TaskScheduler/src/TaskScheduler.h:1351
0x4202ab62: loop() at /home/beki/Documents/OpenDTU-OnBattery/src/main.cpp:160
0x4204c775: app_main at /home/beki/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:56
```

Regarding the backtrace(s) in the first comment:
```
0x403806ee: compare_and_set_native at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
 (inlined by) spinlock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
 (inlined by) xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:301
0x420798fe: temperature_sensor_power_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32s3/sar_periph_ctrl.c:140
0x42053110: temp_sensor_start at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/esp32s3/rtc_tempsensor.c:96
0x42046dec: CpuTemperatureClass::read() at /home/beki/Documents/OpenDTU-OnBattery/lib/CpuTemperature/src/CpuTemperature.cpp:41
0x4200d1cc: MqttHandleDtuClass::loop() at /home/beki/Documents/OpenDTU-OnBattery/src/MqttHandleDtu.cpp:47
0x420f4b61: void std::__invoke_impl<void, void (MqttHandleDtuClass::*&)(), MqttHandleDtuClass*&>(std::__invoke_memfun_deref, void (MqttHandleDtuClass::*&)(), MqttHandleDtuClass*&) at /home/beki/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/invoke.h:70
[... cut ...]
0x420cfe0f: ram_tx_pwctrl_background at ??:?
0x420cb991: get_phy_version_str at ??:?
0x420c2d23: lmacProcessTxComplete at ??:?
0x420faef8: ppTask at ??:?
```

So the difference is only that the dumps for core1 and core0 are ordered differently between the two instances the problem occurred,

* Are you absolutely sure that you are using a proper power supply and that 5.0V actually reach your board at all loads?
* Can you replace your ESP32 easily to see if the issue goes away?
* What board are you using anyways?

ms49434, 2025-02-26:
> Regarding the backtrace(s) in the issue description:
>
> ```
> 0x4207974f: temperature_sensor_get_raw_value at [...]

For regular operation, a genuine RaspBerry Pi 5 power supply (5.1 V, 25.5 W) is used. For the capturing of the serial console log, the unit is connected to a USB 3.0 port of a Dell T20 running Ubuntu > Minicom. The issue doesn't seem to be related to the power supply.
I am running a Fusion Board from AllianceApps and it is managing several HM and HMS inverters. I don't have a replacement board.

MartinHL, 2025-03-02:
I can confirm that there reboots at frequent intervals. When I have a look the device usually has a runtime of some few hours although it runs 24/7. I'm also using v25.2.3.
I also have to mention that I have attached an Ethernet adapter to my ESP. There were also issues in the past with crashes when the device was accessed via Ethernet rather than WLAN. There was a fix back then and the device ran quite well after it until now. Maybe there is a similar issue now?

JWRu, 2025-03-11:
I also experienced a total hang-up some days after updating my Fusion Board to v25.2.3. 
I had to power cycle the board, so no logs are available.
I didn't experience any hang-ups with v25.1.14. Therefore I stepped down to 25.1.14, and the board ist running without problems for the last 2 weeks.

ms49434, 2025-04-03:
I've updated OpenDTU to version 25.4.2, no unexpected reboots in the last 21 hours.

schlimmchen, 2025-04-03:
Ah, yes, that makes sense! I disabled temperature readings on ESP32-S3, Thomas cherry-picked that change from downstream, and your backtraces are those that look exactly like the ones I was targeting. Hence, I am confident that your are good now. Thanks for your feedback! If you feel confident that your problem was resolved, you may close this issue.

ms49434, 2025-04-04:
No reboots in the last 39 hours, it seems  that https://github.com/tbnobody/OpenDTU/commit/0b798056f33e8dad4e5b14cca4eadb9f891ca074 fixed my issue.
Many thanks!

closed 2025-04-04 by ms49434