tbnobody/OpenDTU #2561: [Request] Stop inverter data retrieval for GUI (per Inverter)

4ae38G, 2025-02-25:
### Is your feature request related to a problem? Please describe.

Currently Iam running a "zero grid input" setup where I limit inverter to the amount I need. The amount needed is measured with smart meters, also the overall output of all inverter is measured with a smart meter. All data is feed into iobroker where a script is limiting the inverter. I do this quite often, and found the inverter freezing after some time without accepting new limits anymore.
So I did a trace on the problem and what I found is, that while the DTU is gathering data from the inverter for the GUI and I send a new limit, the whole inverter freezes.

### Describe the solution you'd like

It would be very helpful to have a function to stop gathering data from the inverter for the GUI as per inverter, so you still have the possibility to retrieve data from new ones or for troubleshooting old ones.

tbnobody, 2025-02-25:
Please tell use the version you are using...

schlimmchen, 2025-02-25:
I don't understand your request, but I think I understand your problem. Have you considered using a minimum limit value? You should not set very low limits, as the Hoymiles inverters seems to have trouble with very low limits. Oftentimes they start to oscillate (AC power output) and then they shut off internally. A good rule of thumb seems to be ~13W per input. Maybe that is something that can help you avoid your issue altogether.

ms49434, 2025-02-25:
Please ensure that you check the result of your set limit API call (api/limit/status > limit_set_status) before you submit a new set limit API call. If you don't do so, your requests will be queued until the queue is full. As a result, OpenDTU will become unresponsive.

4ae38G, 2025-04-09:
Ive got another workaround in the meantime. Im doing this for each inverter:

on({ id: [].concat(['opendtu.0.1164a006afe0.ac.phase_1.power']), change: 'any' }, async (obj) => {
  let value = obj.state.val;
  let oldValue = obj.oldState.val;
  if (getState('opendtu.0.1164a0122605.available').val == true && getState('opendtu.0.1164a0122605.producing').val == true) {
    setState('opendtu.0.1164a006afe0.power_control.limit_persistent_absolute' /* Set limit as a absolute value */, Math.round((getState('javascript.0.P1VerbrauchSumme').val / 3) * 10) / 10);
  }

so as soon as a new value is reported to iobroker by dtu, it is setting a new limit right after fetch from this specific inverter. it keeps the spaming of the inverter a little lower and its stable now