tbnobody/OpenDTU #2592: DTU needs permanently reboots - faulty system memory handling?

SolarDwarf, 2025-03-17:
### What happened?

Since July 2023 I'm using openDTU with the WebGUI on ESP32-D0WD-V3 (4 MB, nRF24L01+, no display) to monitor 2 HM-300 inverters.

The openDTU system was working stable until mid of 2024. 
From that time on, with every update the system became increasingly unstable and the GUI started more and more delayed. And after a undefined period of time the openDTU looses WiFi connection, an access by GUI or AP is not possible any more, and a system reboot is required. 

Finally after the last update to v25.2.3, I had to reboot the openDTU several times a day.

To narrow down the problem, I downgraded the firmware back to former versions step by step. The actual stable version looks to be v24.6.29 which I am running now. 

However, also with version v24.6.29 the GUI starts often delayed and/or you need multible tries to open the GUI in a web browser.
-----------------------------------------------------------------------------
Additionally 'ping' monitors the increasingly slower reaction of the openDTU system:

C:\>ping 192.168.178.110

Ping wird ausgeführt für 192.168.178.110 mit 32 Bytes Daten:
Antwort von 192.168.178.110: Bytes=32 Zeit=195ms TTL=64
Antwort von 192.168.178.110: Bytes=32 Zeit=35ms TTL=64
Zeitüberschreitung der Anforderung.
Antwort von 192.168.178.110: Bytes=32 Zeit=69ms TTL=64

Ping-Statistik für 192.168.178.110:
    Pakete: Gesendet = 4, Empfangen = 3, Verloren = 1
    (25% Verlust),
Ca. Zeitangaben in Millisek.:
    Minimum = 35ms, Maximum = 195ms, Mittelwert = 99ms

Some time later:

C:\>ping 192.168.178.110

Ping wird ausgeführt für 192.168.178.110 mit 32 Bytes Daten:
Antwort von 192.168.178.110: Bytes=32 Zeit=35ms TTL=64
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.

Ping-Statistik für 192.168.178.110:
    Pakete: Gesendet = 4, Empfangen = 1, Verloren = 3
    (75% Verlust),
Ca. Zeitangaben in Millisek.:
    Minimum = 35ms, Maximum = 35ms, Mittelwert = 35ms
--------------------------------------------------------------

So my guess is, there might be a problem with memory handling/usage of openDTU system - leeding finally to a slow or freezed openDTU because of memory overflow.

### To Reproduce Bug

Install openDTU v25.2.3 on ESP32-D0WD-V3 (4 MB, nRF24L01+, no display)

### Expected Behavior

Please make the openDTU system stable again.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v25.2.3

### What firmware variant (PIO Environment) are you using?

generic

### Relevant log/trace output

```Shell

```

### Anything else?

Probably related issues with similiar cause of problem (memory overflow):
 #2554 Unexpected reboots
 #2551 OpenDTU freezes about once a day 
 #2535 openDTU crash with 4x HM-1500 from releases after 24.9.30
 #2505 viele Zeitabgleiche mit v25.1.14 & v25.2.3
 #2437 WLAN Verbindungsverluste
 #2400 ESP32 does not reconnect to Wifi
 #2366 OpenDTU-Wifi does not reconnect after a network WiFi failure

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-03-17:
What do you see in the serial console when the esp gets unresponsive or looses it's wifi connection?

SolarDwarf, 2025-03-18:
Neither WiFi access nor AP access is possible, the esp is 'frozen' and requires a new start / reboot (off-on of power supply).

tbnobody, 2025-03-18:
Thats why I asked for a [serial log](https://www.opendtu.solar/firmware/howto/serial_console/) to determine whats going on.

tbnobody, 2025-03-18:
> But the usual issues are

Might be. But I cannot reproduce without the console log. My ESP has a uptime of > 20 days (and thats just because I unplugged it by intention)

arekpietrzyk, 2025-03-18:
One more thing. Recently I've enabled MQTT integration to send out the data. Today I changed the settings for intervals 15s, not 5s which was by default. For now it works :-)

arekpietrzyk, 2025-03-19:
This is a small update from me. It still works. So, the only change I made for now is to switch Settins -> DTU Settings -> Poll Intervals.

I'll continue observing.

arekpietrzyk, 2025-03-20:
Here is another update from my side. Increasing the publish interval for MQTT ... did the trick. OpenDTU is still running without any issues.

@SolarDwarf do you have MQTT enabled?

norgoe, 2025-04-06:
And I thought my OpenDTU device is getting faulty - thank you for this thread.

ESP32-D0WD-V3
4MB
CMT2300A + 1x HMS-1600-4T

For me it always starts with OpenDTU getting no longer updates from the HMS and than it takes some time until the complete DTU no longer is available through the network and display is off.
If I happen to recognize it beforehand, rebooting the OpenDTU through the webinterface does the trick.

I today stumbled upon this thread and changed the mqtt update from 5 to 15 seconds.
2 hours later the opendtu lost connection to the HMS (2m away from DTU).

I now downgraded to 24.6.29 fingers crossed to get a stable system again.

andifr84, 2025-04-26:
> And I thought my OpenDTU device is getting faulty - thank you for this thread.
>
> ESP32-D0WD-V3
> 4MB
> CMT2300A + 1x HMS-1600-4T
>
> For me it always starts with OpenDTU getting no longer updates from the [...]

same problems here

downgrade was a solution? since last update my dtu hangs twice a day.

arekpietrzyk, 2025-04-26:
@andifr84 No, it wasn't. I did some more changes regarding my configuration, like cleaning the Event Log at midnight on each inverter setting. Again hope that will solve the issue, but too early to confirm.

It would be good if you guys would start describing your setup with more details, like the number of inverters, for example, or whether you are using any external integrations that are reading data from OpenDTU or MQTT to export data.

Because I think my issues started with adding a couple more to reach 7 devices currently.

My setup is 7 inverters, an MQTT test setup, and Victron reading data through the web api.

andifr84, 2025-04-26:
@arekpietrzyk 

only 1 inverter
HM 300 
mqtt activated
poll interval 15s
iobroker module

sometimes it runs for weeks without any problem. but yesterday i need 2 hard resets (power cable) to get it working again

ms49434, 2025-04-26:
> [@andifr84](https://github.com/andifr84) No, it wasn't. I did some more changes regarding my configuration, like cleaning the Event Log at midnight on each inverter setting. Again hope that will [...]

I'm running a  similar configuration. Fusion board, 1 HM & 7 HMS inverters, MQTT (10s) & API for zero export. I haven't seen any issues with 25.4.10:

[image]

arekpietrzyk, 2025-04-26:
Good to hear. I'm at 25.4.10 for a couple of days too, but I had issues with wi-fi interference from neighbouring, so it is hard to say whether it is solved or not yet in my case, because I see some gaps in the data logs.

stefan123t, 2025-05-14:
> Thats why I asked for a [serial log](https://www.opendtu.solar/firmware/howto/serial_console/) to determine whats going on.

@SolarDwarf @arekpietrzyk Please send the serial log or the developers will have to pretend that it just did not happen 😜

closed 2025-06-29 by github-actions[bot]