tbnobody/OpenDTU #2597: WebIf Overview inverter different

RealNBB, 2025-03-23:
### What happened?

Hello, the overview page of the Hm-1500 does not match the design of the HM-800 and/or HMS-1600.

[image]

[image]

[image]

### To Reproduce Bug

every time

### Expected Behavior

same design webif for all inverters

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

25.2.3

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-03-23:
Yes thats right. The 1500 has more channels and therefor it needs more space... the line break has to be occour in same place. If you say now, then just let the general section always on top and the channels below, then what  to do with the inverters with 6 channels? they would also wrap around. 
So based on the different inverter types, its not possible to have the same layout for all.

closed 2025-03-23 by tbnobody

RealNBB, 2025-03-23:
`Yes thats right. The 1500 has more channels and therefor it needs more space...`

The HM-1500 have like HMS-1600 but different view...

tbnobody, 2025-03-23:
Thats right, but your HMS has less "Gesamtertrag".. Therefor the text is shorter and therefor more space for all the sections is available. It is always a tradeof of different text sizes, display resolutions and text to display..