tbnobody/OpenDTU #2609: OpenDTU falsely shows power production with all inverters offline

DjLex2021, 2025-03-27:
### What happened?

OpenDTU occasionally indicates that power is being produced, even though all inverters are offline. This incorrect status is displayed only on the display, the Web UI, and the app. However, the correct data is transmitted via MQTT. Steps to reproduce are unclear, but the issue persists intermittently.

### To Reproduce Bug

Steps to reproduce are unclear, but the issue persists intermittently.

### Expected Behavior

OpenDTU should accurately reflect the power production status across all interfaces, showing no production when all inverters are offline.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v25.2.3

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Relevant log/trace output

```Shell

```

### Anything else?

[image]

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-03-27:
Please Provide a Screenshot of the Web Interface when the values are wrong. 
What exactly do you mean With wrong values?

DjLex2021, 2025-03-27:
Ok, then I have to wait until it happens again

In the screenshot you can see that openDTU reports 187.8W power out but the invertver is offline since 17h
At this time, the display also shows 189W

MQTT will report the correct value 0W
So I think this problem is only a issue in the displayed value

tm-107, 2025-03-31:
The same here ... all inverters offline (it was 11:00 PM) and false production on every inverter.

[image]

tbnobody, 2025-03-31:
Did you both activate the [Zero runtime data](https://www.opendtu.solar/firmware/configuration/inverter_settings/#zero-runtime-data) setting in the inverter options? Otherwise it will only show the last received value.

tm-107, 2025-04-01:
> Did you both activate the [Zero runtime data](https://www.opendtu.solar/firmware/configuration/inverter_settings/#zero-runtime-data) setting in the inverter options? Otherwise it will only show the last received value.

Yes, it's activated.
The problem doesn't occur every day. Not today, for example.

tm-107, 2025-04-02:
Night mode is activated.
But when they are disabled, it should be set to 0 W.

closed 2025-05-17 by github-actions[bot]