tbnobody/OpenDTU #2611: HMS-800W-2T detected HM-800 bug

Didierp31, 2025-03-28:
### What happened?

Hello,
When enter serial number, opendtu detect HM-600/700/800-2T , but I have an HM-800W-2T
On the micro onduleur,  There are two serials numbers , one with password that I use and detect HM-600....
The second serial number message sans unknow and dont detect type.
Is it a bug, HM et HMS not same frequency If I understant.
Thanks

### To Reproduce Bug

Stay red

### Expected Behavior

Sorry I dont understand what I nedd to say here

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

V25.2.3

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Didierp31, 2025-03-28:
First serial number is like 1412.... but unknow by opendtu, second serial number start by  4143 but detected like HM-

ms49434, 2025-03-28:
HMS-800W-2T is not a supported model.

closed 2025-03-28 by tbnobody