tbnobody/OpenDTU #2625: 25.4.4 - Live view: no more data is displayed.

fila612, 2025-04-06:
### What happened?

after updating to 25.4.4 the data in live view is no more available.
all data seems to be received (grab them via REST & mqtt to HA) but not shown in the webapp.

the notification that data will be displayed after 10s does not disappear.

[image]

### To Reproduce Bug

open live view

### Expected Behavior

data should be displayed

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

25.4.4

### What firmware variant (PIO Environment) are you using?

ESP32S3_USB (FusionBoard)

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-04-06:
Are you usiung any kind of reverse proxy? 
It seems that your browser does not support websockets...

fila612, 2025-04-06:
thanks @tbnobody: that's the reason. I'm using a nginx to access all of my local systems via dns name. last week I had to change the IP of of the opendtu and therefore I also changed the nginx-entry.
!!but, I forgot to check the box "support websockets". just changed them and now its working.
I wouldn't have been able to solve this without your advice - thank you

closed 2025-04-06 by fila612