tbnobody/OpenDTU #2641: Permanent limit is not permanent anymore

broth-itk, 2025-04-20:
### What happened?

I can't tell since when but with my HMS-2000 and HMS-1800 inverters, running with Firmware 2.0.4, I can't set the permanent limit anymore.
The behaviour is exactly as with the "non-persistent" setting:

- Setting 20% will show in GUI shortly as 400W
- Some time later the GUI shows 2000W but limit stays active (can't explain but have read other users with that issue)
- Next day the limit is gone, setting is at 100% as well as power output...

Is anyone with 2.0.4 or similar having the same issue?

This might be a hoymiles firmware issue or maybe something related to the power distribution logic.

Offtopic: Is there any plan to implement a dynamic power limiter like then one from the OpenDTU on battery project? This would be very handy since we would not need the "permanent" inverter setting anymore.

### To Reproduce Bug

Set permanent limit and observe system.

### Expected Behavior

Setting permanent limit being actually permanent.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

25.4.10

### What firmware variant (PIO Environment)?

opendtu-generic_esp32.bin

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-04-20:
Just let me explain what happens when you set the limit:

* A limit change command is issued to the inverter. If the ESP receives a receive acknowlege from the inverter it _assumes_ the the limit was applied and sets the display in the WebUI etc. to the new value
* After a few minutes (it's not possible earlier as the inverter would generate log entries otherwise) the ESP queries the inverter for the new limit. This might be the time when your view swiches back to 100%

That means two things:
1. The query command seem to have the current limit on either a different position or has changed completly
2. The type of the set command (persistant vs. temporary) might also different.

To analyze the first issue it would be helpfull to have a log output from the moment you've set the new limit until the occourance of the `SystemConfigPara` command.

broth-itk, 2025-04-21:
Thanks for the insights! I'm going to connect my OpenDTU unit to USB and capture logs once I'm back home.

In the meantime I created a script which sets the limits via MQTT. Cron will take care to set them daily at 10am.

tbnobody, 2025-04-21:
A output of the web-console should also be sufficient as we  are only looking for raw sent and received data. 
This issue  seem to be related to #2523
And it's maybe also related to https://github.com/hoylabs/OpenDTU-OnBattery/commit/539f5898089030b226070cfa7dd18244c9da8075

broth-itk, 2025-05-12:
@tbnobody with some delay and thanks to the new syslog feature, here is an example command trace:

```
May 12 05:53:23 OpenDTU-645198 OpenDTU[4f1131a9] D (31608329) hoymiles: TX ActivePowerControl (101) 868.00 MHz --> 51 80 72 97 85 80 17 44 88 81 0B 00 00 C8 01 01 CE 40 26
May 12 05:53:23 OpenDTU-645198 OpenDTU[4f1131a9] V (31608414) hoymiles: Interrupt received
May 12 05:53:23 OpenDTU-645198 OpenDTU[4f1131a9] D (31608424) hoymiles: RX 868.00 MHz --> D1 80 72 97 85 80 17 44 88 81 00 00 0B 00 14 07 F3 | -24 dBm
May 12 05:53:24 OpenDTU-645198 OpenDTU[4f1131a9] I (31609526) hoymiles: Fetch inverter: 116481313070
May 12 05:53:24 OpenDTU-645198 OpenDTU[4f1131a9] I (31609533) hoymiles: Queue size - NRF: 0 CMT: 3
May 12 05:53:25 OpenDTU-645198 OpenDTU[4f1131a9] I (31610376) hoymiles: RX Period End
May 12 05:53:25 OpenDTU-645198 OpenDTU[4f1131a9] I (31610381) hoymiles: Success
```

This was executed on that inverter:

[image]

Non-persistent looks like that:

```
May 12 05:58:01 OpenDTU-645198 OpenDTU[4f1131a9] D (31886696) hoymiles: TX ActivePowerControl (01) 868.00 MHz --> 51 81 31 30 70 80 17 44 88 81 0B 00 00 C8 00 01 5E 41 A6
May 12 05:58:01 OpenDTU-645198 OpenDTU[4f1131a9] V (31886779) hoymiles: Interrupt received
May 12 05:58:01 OpenDTU-645198 OpenDTU[4f1131a9] D (31886788) hoymiles: RX 868.00 MHz --> D1 81 31 30 70 80 17 44 88 81 00 00 0B 00 14 07 E3 | -40 dBm
May 12 05:58:03 OpenDTU-645198 OpenDTU[4f1131a9] I (31888744) hoymiles: RX Period End
May 12 05:58:03 OpenDTU-645198 OpenDTU[4f1131a9] I (31888748) hoymiles: Success
May 12 05:58:04 OpenDTU-645198 OpenDTU[4f1131a9] I (31890232) hoymiles: Fetch inverter: 116480729785
May 12 05:58:04 OpenDTU-645198 OpenDTU[4f1131a9] I (31890238) hoymiles: Queue size - NRF: 0 CMT: 1

```

I can confirm the observation that the limit is shown only for a short term on GUI but reverts to default a while after.
Nevertheless the inverters are actually still limiting their output to the configured limit.

readonly24, 2025-05-13:
I would also like to update my HMS-1600 to the latest firmware to use the Power Distribution Logic. So it it save to upgrade because this bug is only a GUI problem or should I wait until OpenDTU is fixed?

stefan123t, 2025-05-14:
@broth-itk thanks for tracing, here are your commands / responses manually decoded

But @tbnobody was asking for the logs from the `ActivePowerControl` Command until the `SystemConfigPara` response.
The `SystemConfigPara` will tell what the inverter has actually set as a limit and what will be displayed in the OpenDTU view.

Can you provide these extended logs from your syslog too ?

```
51 <80729785> <80174488> 81 0B00 00C8 0101 CE40 26 0x0B00 ActivePowerControl 0x00C8 = 20.0%, type RelativPersistent = 0x0101 
D1 <80729785> <80174488> 81 0000 0B00 1407 F3 DevControlResponse ActivePowerControl OK
```

```
51 <81313070> <80174488> 81 0B00 00C8 0001 5E41 A6 0x0B00 ActivePowerControl 0x00C8 = 20.0%, type RelativNonPersistent = 0x0001
D1 <81313070> <80174488> 81 0000 0B00 1407 E3 DevControlResponse ActivePowerControl OK
```

broth-itk, 2025-05-15:
@stefan123t 

```
May 15 10:15:08 OpenDTU-645198 OpenDTU[4f1131a9] D (306521530) hoymiles: TX ActivePowerControl (01) 868.00 MHz --> 51 80 72 97 85 80 17 44 88 81 0B 00 01 2C 00 01 95 00 D9
May 15 10:15:08 OpenDTU-645198 OpenDTU[4f1131a9] V (306521614) hoymiles: Interrupt received
May 15 10:15:08 OpenDTU-645198 OpenDTU[4f1131a9] D (306521623) hoymiles: RX 868.00 MHz --> D1 80 72 97 85 80 17 44 88 81 00 00 0B 00 14 07 F3 | -24 dBm
May 15 10:15:08 OpenDTU-645198 OpenDTU[4f1131a9] I (306521724) hoymiles: Fetch inverter: 116480729785
May 15 10:15:08 OpenDTU-645198 OpenDTU[4f1131a9] I (306521731) hoymiles: Queue size - NRF: 0 CMT: 2
May 15 10:15:10 OpenDTU-645198 OpenDTU[4f1131a9] I (306523581) hoymiles: RX Period End
May 15 10:15:10 OpenDTU-645198 OpenDTU[4f1131a9] I (306523586) hoymiles: Success

```
some time later

```
May 15 10:15:28 OpenDTU-645198 OpenDTU[4f1131a9] I (306541779) hoymiles: Request SystemConfigPara
May 15 10:15:28 OpenDTU-645198 OpenDTU[4f1131a9] I (306541784) hoymiles: Queue size - NRF: 0 CMT: 2
May 15 10:15:28 OpenDTU-645198 OpenDTU[4f1131a9] D (306541801) hoymiles: TX RealTimeRunData 868.00 MHz --> 15 81 31 30 70 80 17 44 88 80 0B 00 68 25 A2 A0 00 00 00 00 00 00 00 00 D6 25 89
May 15 10:15:28 OpenDTU-645198 OpenDTU[4f1131a9] V (306541903) hoymiles: Interrupt received
May 15 10:15:28 OpenDTU-645198 OpenDTU[4f1131a9] D (306541913) hoymiles: RX 868.00 MHz --> 95 81 31 30 70 80 17 44 88 01 00 01 01 78 01 49 00 41 01 72 00 F7 04 C2 00 03 0F | -39 dBm
May 15 10:15:28 OpenDTU-645198 OpenDTU[4f1131a9] V (306541952) hoymiles: Interrupt received
May 15 10:15:28 OpenDTU-645198 OpenDTU[4f1131a9] D (306541962) hoymiles: RX 868.00 MHz --> 95 81 31 30 70 80 17 44 88 02 22 19 00 03 2C 9F 00 2E 00 AE 02 08 01 45 00 65 1C | -39 dBm
May 15 10:15:28 OpenDTU-645198 OpenDTU[4f1131a9] V (306542002) hoymiles: Interrupt received
May 15 10:15:28 OpenDTU-645198 OpenDTU[4f1131a9] D (306542011) hoymiles: RX 868.00 MHz --> 95 81 31 30 70 80 17 44 88 03 01 7C 02 0F 04 D4 00 04 86 16 00 03 70 BD 00 75 B2 | -40 dBm
May 15 10:15:28 OpenDTU-645198 OpenDTU[4f1131a9] V (306542052) hoymiles: Interrupt received
May 15 10:15:28 OpenDTU-645198 OpenDTU[4f1131a9] D (306542062) hoymiles: RX 868.00 MHz --> 95 81 31 30 70 80 17 44 88 04 00 8A 09 22 13 85 0B FD 00 00 00 83 03 E8 01 1B 89 | -40 dBm
May 15 10:15:28 OpenDTU-645198 OpenDTU[4f1131a9] V (306542093) hoymiles: Interrupt received
May 15 10:15:28 OpenDTU-645198 OpenDTU[4f1131a9] D (306542101) hoymiles: RX 868.00 MHz --> 95 81 31 30 70 80 17 44 88 85 00 05 9B 39 1C | -39 dBm
May 15 10:15:29 OpenDTU-645198 OpenDTU[4f1131a9] I (306542356) hoymiles: RX Period End
May 15 10:15:29 OpenDTU-645198 OpenDTU[4f1131a9] I (306542361) hoymiles: Success
May 15 10:15:29 OpenDTU-645198 OpenDTU[4f1131a9] D (306542381) hoymiles: TX SystemConfigPara 868.00 MHz --> 15 81 31 30 70 80 17 44 88 80 05 00 68 25 A2 A0 00 00 00 00 00 00 00 00 18 2A 46
May 15 10:15:29 OpenDTU-645198 OpenDTU[4f1131a9] V (306542483) hoymiles: Interrupt received
May 15 10:15:29 OpenDTU-645198 OpenDTU[4f1131a9] D (306542494) hoymiles: RX 868.00 MHz --> 95 81 31 30 70 80 17 44 88 01 00 03 03 E8 00 00 03 E8 00 00 00 00 00 00 3C F8 F8 | -39 dBm
May 15 10:15:29 OpenDTU-645198 OpenDTU[4f1131a9] V (306542532) hoymiles: Interrupt received
May 15 10:15:29 OpenDTU-645198 OpenDTU[4f1131a9] D (306542542) hoymiles: RX 868.00 MHz --> 95 81 31 30 70 80 17 44 88 02 00 00 00 00 00 00 00 01 00 64 00 00 00 00 00 00 59 | -39 dBm
May 15 10:15:29 OpenDTU-645198 OpenDTU[4f1131a9] V (306542574) hoymiles: Interrupt received
May 15 10:15:29 OpenDTU-645198 OpenDTU[4f1131a9] D (306542583) hoymiles: RX 868.00 MHz --> 95 81 31 30 70 80 17 44 88 83 01 F4 00 64 B2 EE 70 | -39 dBm
May 15 10:15:29 OpenDTU-645198 OpenDTU[4f1131a9] I (306542675) hoymiles: RX Period End
May 15 10:15:29 OpenDTU-645198 OpenDTU[4f1131a9] E (306542681) hoymiles: (lib/Hoymiles/src/parser/SystemConfigParaParser.cpp, 41) stats packet too large for buffer
May 15 10:15:29 OpenDTU-645198 OpenDTU[4f1131a9] E (306542691) hoymiles: (lib/Hoymiles/src/parser/SystemConfigParaParser.cpp, 41) stats packet too large for buffer
May 15 10:15:29 OpenDTU-645198 OpenDTU[4f1131a9] I (306542702) hoymiles: Success

```

I wonder why there is an error followed by "Success" but I hope this is what you are looing for :)

Thanks!

stefan123t, 2025-05-16:
Thanks for the SystemConfigPara Request / Response!

```
51 80729785 80174488 81 0B00 012C 0001 9500 D9 <- 0x0B00 ActivePowerControl 0x012C = 30.0%, type RelativNonPersistent = 0x0001 
D1 80729785 80174488 81 0000 0B00 1407 F3 <- DevControlResponse ActivePowerControl OK

15 81313070 80174488 80 0500 6825A2A0 0000 0000 0000 0000 182A 46 <- SystemConfigPara Command 0x6825A2A0 timestamp 
95 81313070 80174488 01 0003 03E8 0000 03E8 0000 0000 0000 3CF8 F8 <- SystemConfigPara Response > 16 Byte
95 81313070 80174488 02 0000 0000 0000 0001 0064 0000 0000 0000 59
95 81313070 80174488 83 01F4 0064 B2EE 70
```

```
0003 03E8 0000 03E8 0000 0000 <- 0x3E8 = 100.0% !
0000 3CF8 0000 0000 0000 0001 <- 0x3CF8 = 15608 ?
0064 0000 0000 0000 01F4 0064 <- 0x0064 = 100, 0x01F4 = 500, 0x0064 = 100
B2EE <- CRC16
```
So this is a somewhat different SystemConfigPara Response than the one we received in earlier versions.
The Inverter responds indeed with 100.0%.
Before that we can see probably SystemConfigPara Version 0x0003 as the response format.
This used to be 0x0001 for previous firmware versions / models.

The error in line 41 is thrown somewhere here in the `appendFragment()` method:

[SystemConfigParaParser.cpp](https://github.com/tbnobody/OpenDTU/blob/3f01b8d4b7534fd5c09e67b32d0783206e88f4c5/lib/Hoymiles/src/parser/SystemConfigParaParser.cpp#L32C1-L46C2)
```c++
void SystemConfigParaParser::clearBuffer()
{
    memset(_payload, 0, SYSTEM_CONFIG_PARA_SIZE);
    _payloadLength = 0;
}

void SystemConfigParaParser::appendFragment(const uint8_t offset, const uint8_t* payload, const uint8_t len)
{
    if (offset + len > (SYSTEM_CONFIG_PARA_SIZE)) {
        ESP_LOGE(TAG, "(%s, %d) stats packet too large for buffer", __FILE__, __LINE__);
        return;
    }
    memcpy(&_payload[offset], payload, len);
    _payloadLength += len;
}
```

@tbnobody which is preset to max. 16 bytes in [SystemConfigParaParser.h](https://github.com/tbnobody/OpenDTU/blob/3f01b8d4b7534fd5c09e67b32d0783206e88f4c5/lib/Hoymiles/src/parser/SystemConfigParaParser.h#L5C1-L5C35) while in fact it is 36 bytes plus 2 bytes CRC16 in your response.

```c++
#define SYSTEM_CONFIG_PARA_SIZE 16
```

stefan123t, 2025-05-17:
@tbnobody this is the new `SystemConfigPara` response version 3 which was introduced with HMS Firmware version 2.0.4 or earlier having the **Power Distribution Logic PDL** feature.

In contrast to earlier `SystemConfigPara` responses this is now three packets long and starts with 0x0003:

```
15 81313070 80174488 80 0500 6825A2A0 0000 0000 0000 0000 182A 46 <- SystemConfigPara Command 0x6825A2A0 timestamp 
95 81313070 80174488 01 0003 03E8 0000 03E8 0000 0000 0000 3CF8 F8 <- SystemConfigPara Response 36 >> 16 Byte
95 81313070 80174488 02 0000 0000 0000 0001 0064 0000 0000 0000 59
95 81313070 80174488 83 01F4 0064 B2EE 70

0003 03E8 0000 03E8 0000 0000 0000 3CF8 <- 0x0003 version 3, 0x3E8 = 100.0% Limit ! 0x3E8 = 100.0% ?, 0x3CF8 = old CRC16 ??? this may be something else
0000 0000 0000 0001 0064 <- 0x0001 = 1, 0x0064 = 100
0000 0000 0000 01F4 0064 <- 0x01F4 = 500, 0x0064 = 100
B2EE <- CRC16
```

Could we automatically notify the user to upload such an "unknown" response based on the version mismatch or at least the packets being too large for our buffer and the two errors being logged ?

DOliana, 2025-05-24:
Hi @tbnobody . I see there is a branch that seems to fix the issue (at least according to the name of the branch). I would be happy to test if you are looking for beta-testers.

AndreasBoehm, 2025-05-28:
@alexciven the only solution right now is to downgrade the firmware of your inverter.

GitKranich, 2025-06-18:
Is there any update on a solution? I’m facing the same problem.

OutdoorRob, 2025-07-26:
@stefan123t @broth-itk 
I think there is an issues with those [logs by broth-itk](https://github.com/tbnobody/OpenDTU/issues/2641#issuecomment-2882981936) - could you confirm this please?!

See stefan123t's [post](https://github.com/tbnobody/OpenDTU/issues/2641#issuecomment-2886433841) for the already 'decoded' messages and better formatting. 

The 'ActivePowerControl is send to inverter S/N **80729785** but the 'RealTimeRunData' and the 'SystemConfigPara' belong to an inverter with the S/N **81313070**.

stefan123t, 2025-07-27:
@OutdoorRob thanks for looking into this. I was simply taking the logs from Bernhard and manual / mental decoding the transmissions going forth and back between the DTU and inverters.

You are right the ActivePowerControl is for a different inverter than the SystemConfigPara Command and response. But it shows the 0x0003 "version" (?) identifier of the SystemConfigPara response.

What / where is the error in Bernhards logs you mention, as I am unable to see it ?

OutdoorRob, 2025-07-27:
@stefan123t No worries.

The 'error' / issue is, that the two messages don't belong to the same inverter.
Maybe there are inverters that send a new `SystemConfigPara` response version 3 **but** it might not be related to the firmware update / 'Power Distribution Logic' function.
So we are looking at the 'ActivePowerControl' from one inverter that was updated to support PDL but the response might be from an inverter that runs a completely different firmware.

I just wanted to say we should be careful and really keep facts separated :)

@broth-itk 
Could you tell us what inverters + firmware version belongs to the S/N 80729785 and 81313070?

stefan123t, 2025-08-08:
@OutdoorRob do you still need the answer from Bernhard ?

We have answered this some time ago here https://github.com/tbnobody/OpenDTU/issues/1650#issuecomment-2480015944

---

schau mal hier sind die beiden Firmware Updates von @broth-itk

https://github.com/tbnobody/OpenDTU/issues/1421#issuecomment-2425232860

Die sind für die beiden Inverter mit den Serial IDs 80729785 und 81313070.

Er hatte hier und weiter oben in diesem issue seine Inverter genauer beschrieben:

https://github.com/tbnobody/OpenDTU/issues/2137#issuecomment-2428138345

These are my IDs and SW

DTU: 80164633 (v00.02.23)
HMS 1: 80729785 (1800-4T, v1.00.27)
HMS 2: 81313070 (2000-4T, v1.00.27)

[update of two -4T inverters to 1.0.27](https://github.com/tbnobody/OpenDTU/issues/1650#issuecomment-1928827458) (Firmware Build Date 2023-06-05 10:24:00)
HMS 3: 83199047 (350-1T, v1.00.14)
Anbei ein großer Trace von heute morgen.
Darin sind [folgende Updates](https://github.com/tbnobody/OpenDTU/issues/1650#issuecomment-2428650265) enthalten:

HMS-1800-4T/HMS-2000-4T von V01.00.27 auf V01.01.12
HMS-350-1T von V01.00.14 auf V01.01.00

https://github.com/tbnobody/OpenDTU/issues/2137#issuecomment-2260425094

Serial 1164807xxxxx
Production Year 2022
Production Week 7
Model HMS-1800-4T
Detected max. Power 1,800 W
Bootloader Version 0.0.1
Firmware Version 1.0.27
Firmware Build Date 2023-06-05 10:24:00
Hardware Part Number 269635841
Hardware Version 01.00
Serial 1164813xxxxx
Production Year 2022
Production Week 13
Model HMS-2000-4T
Detected max. Power 2,000 W
Bootloader Version 0.0.1
Firmware Version 1.0.27
Firmware Build Date 2023-06-05 10:24:00
Hardware Part Number 269644033
Hardware Version 01.00
Serial 1124831xxxxx
Production Year 2022
Production Week 31
Model HMS-350-1T
Detected max. Power 350 W
Bootloader Version 0.1.2
Firmware Version 1.0.14
Firmware Build Date 2021-12-09 12:46:00
Hardware Part Number 270541056
Hardware Version 02.00

Micha12323, 2025-09-09:
I have the same issue. 

I found out, that you can set a "fixed" limit with the dtu-lite from hoymiles. With that you can at least set the limit to 800 W or what you want and have it again after a restart at another day or after some minutes.