tbnobody/OpenDTU #2654: SSD1309 not working with CMT

njordan77, 2025-05-01:
### What happened?

I have several NRF+SSD1309 setups, work great. Started with a new CMT+SSD1309 Setup - but here i only can make CMT work, but the display never goes Online. I even thought it would be a brand new but broken display. so i bought another one, but the same problem. not starting at all.

Using the same pin layout and pinmapping config as for the other workign displays.
I only can imagine that the CMT does not co-work, but its not written anywhere, so i'm not sure if that is really the issue. does this work for other people here? I spent much time but i'm at the end of what is possible to try next for myself.

Therefore i hope for some wisdom from the community!
Thanks
Jordan

### To Reproduce Bug

NodeMCU32 (ESP32-S) with latest firmware v25.4.10.
pinout looks as follows

`   {
        "name": "NodeMCU32 with CMT2300A & SSD1309",
        "nrf24": {
            "miso": -1,
            "mosi": -1,
            "clk": -1,
            "irq": -1,
            "en": -1,
            "cs": -1
        },
        "cmt": {
            "clk": 12,
            "cs": 27,
            "fcs": 26,
            "sdio": 14,
            "gpio2": -1,
            "gpio3": -1
        },
        "eth": {
            "enabled": false,
            "phy_addr": -1,
            "power": -1,
            "mdc": -1,
            "mdio": -1,
            "type": 0,
            "clk_mode": 0
        },
        "display": {
            "type": 4,
            "data": 21,
            "clk": 22,
            "reset": 32
        }
    },
`

### Expected Behavior

that not just CMT works and does receive data from the inverter, but also the display presents the data

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

f0db2a5

### What firmware variant (PIO Environment)?

opendtu-generic_esp32.bin

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-05-01:
I don't have a ESP32-S avialble here. But I can confirm that display type 4 with CMT module works perfectly great on an ESP-WROOM-32..

Some things to mention:
* You  can remove the reset pin mapping `"reset": 32` as no reset pin will be used for this type of display. 
* Did you also choose the right hardware profile in the device configuration dialog?
* Please provide a output of the serial console during the startup phase (not the web console as it does not show the startup phase) 
* To eliminate any wiring issues a photo of your setup might also help

njordan77, 2025-05-01:
Thanks a lot for your feedback.

* did remove now the reset line, so it is set to -1 now, but did not change anything, also removed the RST wire, same behavior.
* sure i choose the hardware profile, else your webGUI would immediately complain about the missing prpfile selection. The CMT itself is recognized and works...

here is the console outcome.....

`Starting OpenDTU
Initialize FS... done
Reading configuration... Check for default DTU serial... done
done
Reading language pack... done
Reading PinMapping... found valid mapping done
Initialize Network... done
Setting Hostname... Configuring WiFi STA using new credentials... done
Initialize NTP... done
Initialize SunPosition... done
Initialize MqTT... done
Initialize WebApi... done
Initialize Display... done
Initialize LEDs... done
Initialize Hoymiles interface... CMT: Connection successful
  Setting country mode... 
  Setting CMT target frequency... 
  Setting radio PA level... 
CMT TX power set to 16 dBm
  Setting DTU serial... 
  Setting poll interval... 
  Adding inverter: 1164XXXXXX - oh63Hoymiles done
done
Switch to WiFi mode
Setting Hostname... done
Configuring WiFi STA using new credentials... done
Configuring WiFi STA DHCP IP... done
Fetch inverter: 1164XXXXXX
Queue size - NRF: 0 CMT: 0
WiFi connected
WiFi got ip: 192.168.1.68
Network connected
Fetch inverter: 1164XXXXXX
Queue size - NRF: 0 CMT: 0
Fetch inverter: 1164XXXXXX
Queue size - NRF: 0 CMT: 0
Admin AP remaining seconds: 10 / 60
Fetch inverter: 1164A00CA6F0
Queue size - NRF: 0 CMT: 0
Fetch inverter: 1164XXXXXX
Queue size - NRF: 0 CMT: 0
Admin AP remaining seconds: 20 / 60`

Regarding the wiring i will concentate on the SSD1309 as the CMT itself works.
As its I2C it will use 21 for data and 22 for the clock.....
connected to the display and used 3,3V.....never tried 5V, some say its  too high.

Kind Regards
Norbert

tbnobody, 2025-05-01:
> also removed the RST wire

What do you mean you removed the reset wire? There should only be 4 wires. VCC, GND, SCL and SDA. There shouldn't even more connections on the display itself

njordan77, 2025-05-01:
initially i used a setup with RST configured to 32 and hard-wired to GPIO32. it it works with 2 other OpenDTU units. based on your feedback i removed now RST wire and the RST line in pinmapping.config, so its -1 now.

My SSD1309 is a 2,4inch OLED display with SPI and I2C Support. i jumperred it according to I2C wiring.

njordan77, 2025-05-07:
Any other idea i could try next to get my issue forward? Thanks Norbert

njordan77, 2025-05-19:
i fixed it myself. info just in case someone else running into this issue. i have SSD1309 version that supports both, SPI and I2C. different from what was mentioned above, a RESET handling is mandatory for my display. i moved the RST Display pin to the ESP32 RST pin, which worked most of the time, but not when plugged in the usb-port for the first time....when elko's have been with no charge at all.

what works all the time, is giving a certain ESP32 pin this function and set it up in the pinmapping config. My MISTAKE was that i once used GPIO36, which also is ADC0, BUT works only AS an output. so i guess as RST is an output in never worked. moved on to GPIO4 and now it all works fine.

Thanks for the discussion

closed 2025-05-19 by njordan77