tbnobody/OpenDTU #2656: [Request] Publish Inverter Log to HomeAssistant

Dani1802, 2025-05-03:
### Is your feature request related to a problem? Please describe.

No, not related to a problem.
Would it be possible to make the event log of the inverter available in HomeAssisant?
Or at least the most recent entry?
That would be a good way to get actively informed of unexpected events.

### Describe the solution you'd like

Inverter Log is available in HA

podly, 2025-08-29:
It would be very useful to have inverter log notifications sent via MQTT. My inverter, after brief grid issues (e.g., overvoltage, underfrequency, etc.), can take up to 45 minutes to come back online, even though a simple reset brings it back instantly. If Home Assistant could detect these errors and automatically reset the inverter, it would solve this problem. It would also be great to have HA push notifications about faults, issues, or other events directly to phone. Is there any chance this feature could be implemented in the near future?

Dani1802, 2025-08-30:
In the meantime, I made myself a workaround.
I query the Eventlog periodically by API (/eventlog/status?inv=$InverterSerial) and

- send all Events to my EventLog Topic
- send Events (excl. Time Sync) to my ErrorLog Topic

I've added both Topics to my HA and made an automation which send me a push every time the ErrorLog Message changes

[image]

PS: "Daily Log Reset" are Messages I publish every night to clear the Error-Log

podly, 2025-09-01:
> Yes sure, I'm running this script by cron in a Debian VM:
> 
> [getInverterLog.sh](https://github.com/user-attachments/files/22065993/getInverterLog.sh)

Nice idea, works for me. Thx!