tbnobody/OpenDTU #2664: OpenDTU not more working with latest Firefox, but works with Microsoft Edge

capiman, 2025-05-07:
### What happened?

Since a few days when I open OpenDTU page to check values of my PV, the above part works, but detailed info (lower part) is loaded forever and not displaying any content.
I yesterday updated software of OpenDTU, but problem remain the same.
I just tried Microsoft Edge, and there it is working without problem.
Going back to Firefox problem occurs again.
When opening the console on Firefox (F12 on Windows11), I can see

"Firefox kann keine Verbindung zu dem Server unter ws://admin:...@192.168.177.23/livedata aufbauen."

Firefox cannot establish connection to ...

### To Reproduce Bug

At the moment it occurs every time on Firefox and work (at least at time of testing) with Microsoft Edge

### Expected Behavior

Works with Firefox as well as Microsoft Edge

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

f0db2a5

### What firmware variant (PIO Environment)?

generic

### Relevant log/trace output

```Shell

```

### Anything else?

Perhaps some access problem? That certain address is now forbidden?
I also have ABP Adblocker in Firefox, I will try if this makes a difference...

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

capiman, 2025-05-07:
It was the addblocker! When disabling addblocker on local OpenDTU page, it works again!

closed 2025-05-07 by capiman

schlimmchen, 2025-05-09:
What Adblocker? A particular add-on?

capiman, 2025-05-09:
Adblock Plus

Versionsnummer 4.21.1

Copyright © 2025 [eyeo GmbH]

https://adblockplus.org/