tbnobody/OpenDTU #2668: [Request] Zweiten SML Zähler einlesen

pillepalle127, 2025-05-09:
### Is your feature request related to a problem? Please describe.

Mein OpenDTU läuft fantastisch. Ich besitze eine Wärmepumpe mit einem seperaten Zähler und würde den gerne aus dem selben Heimspeicher ausregeln. Leider finde ich dazu keinerlei Infos. 
Gibt es eine möglichkeit ein zweites SML Interface einzulesen?

### Describe the solution you'd like

Gibt es eine möglichkeit ein zweites SML Interface einzulesen? In der pin_mapping.json Sysntax finde ich dazu keine Definition.

schlimmchen, 2025-05-09:
> Ich besitze eine Wärmepumpe mit einem seperaten Zähler und würde den gerne aus dem selben Heimspeicher ausregeln.

Oder wir klären gleich hier: Einer von uns hat etwas missverstanden. Entweder begreife ich deine Anforderung nicht, oder du hast nicht verstanden, dass sich die Produktion von Strom in deinem Hausnetz niemals positiv (im Sinne von "Geld gespart") auf dem Wärmepumpenzähler auswirkt.

Wenn du Strom produzierst und ihn einspeist in dein Hausnetz, dann sieht dein Hauszähler das und wegen der Summierung senkst du bestenfalls deinen berechneten Verbrauch auf Null. Der Zähler der Wärmepumpe wird aber immer genau die Leistung messen, die die Wärmepumpe verbraucht, egal ob der Strom vom Zähler nebenan durch den Wärmepumpenzähler fließt, oder aus dem Netz. Du müsstest ja in den Teil des Stromnetztes zwischen deinem Wärmepumpenzähler und deiner Wärmepumpe einspeisen, damit sich das auf den Wärmepumpenzähler auswirkt.

Is klar, was ich meine? Vielleicht habe ich aber auch etwas nicht verstanden, dann öffne mal ein Issue bei OpenDTU-OnBattery und wir besprechen das.

pillepalle127, 2025-05-09:
Okay, weitere Erläuterung. 
Ich werde die beiden Kreise über den Speicher DC seitig koppeln. Alle PV (2 BKW mit Option auf mehr) geht in den Speicher und von da aus fahre ich simultan Nulleinspeisung mit 2 WR.
Zwei unabhängige OpenDTU sind technisch überflüssig, ineffizient und schwer zu regeln. Daher die Frage nach einer zweiten SML Schnittstelle. 
Aktuell fahre ich was ähnliches mit Tasmota. Mir fehlt sich die DTU Schnittstelle. 
Ich bin überrascht, dass das nicht vorgesehen ist.
Lässt sich da was tricksen?

pillepalle127, 2025-05-09:
Oh Mist, du hast Recht. Mein Opfer ist tatsächlich OpenDTUonbattery. 
Trotzdem danke und super Projekt!

closed 2025-05-09 by pillepalle127

schlimmchen, 2025-05-09:
> Lässt sich da was tricksen?

Nicht mit OpenDTU-OnBattery. Da musst du dann Ahoy-ZE (non-public Beta) anschauen. Die können bereits WR in Gruppen einteilen und die Gruppen nach unterschiedlichen Zählerwerten regeln.

Mit OpenDTU-OnBattery kannst du lediglich zwei DTUs bauen, je eine an einen Zähler, und dann die WR entsprechend aufteilen.

pillepalle127, 2025-05-09:
Danke für Tip! Das schaue ich mir an.