tbnobody/OpenDTU #2672: Seit FW 24.10.15 nur Probleme mit Hoymiles WR

azazul1980, 2025-05-12:
### What happened?

Hallo liebe Gemeinde,

mir ist aufgefallen das alle Versionen von mir nach 24.10.15 nicht mehr sauber laufen. Wie äußert sich das bei mir ? Ich habe 2 WR HM-1500 und HM-1200 welche über openDTU und HomeAsisstant eine Nulleinspeisung realisieren. Leider habe ich mit neueren Versionen das Problem das die Datenübertragung ziemlich zäh wird. Es kommt weniger an und es kommt auch weniger zurück an Daten. Die Funkstatistik habe ich auch schon verglichen konnte hier aber keine extremen Abweichungen feststellen. Ich würde sogar fast sagen diese sind nahezu gleich. Aber die Realität ist leider eine andere. Auch das Hoch und Herunterstellen der Sendeleistung hat mir leider bei den neueren FW Versionen nichts gebracht. Aber so bald ich auf 24.10.15 gehe läuft das System wieder ziemlich Sauber. 

Auch das Thema Zeitabgleich ID2 ist bei den neuen Versionen viel häufiger zu lesen wie bei der alten FW. 

Nun meine Frage hat jemand anderes dieses Problem und wie konnte er dies in Griff bekommen ? Was ist überhaupt hier am schief laufen ? 

### To Reproduce Bug

Vergleich bei der Anpassung und Geschwindigkeit beider FW Varianten 24.10.15 - aktueller

### Expected Behavior

Langsame Anpassung der Werte auf den WR 30-90 sek 

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

24.10.15 - Neuer

### What firmware variant (PIO Environment)?

ESP32

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

lueralba, 2025-05-16:
Hello.
The same problem here: I've been using OpenDTU on an "AllianceApps Fusion" (generic esp32s3 usb) with an HMS1600-4T since March 25, 2025.
Since my first update to 25.4.10, the query "hanged" for the first time (always the same values) on May 7, 2025.
The MQTT transmission continues to run smoothly during this time.
(Updated to 25.5.10 the day before yesterday, and today's second "hang")
Restarting the OpenDTU software no longer establishes a connection to the inverter!

lueralba, 2025-05-30:
Today (1:00 PM) the same problem again! 
It just stopped at 1088W (fine sunshine).

I just (3:30 PM) changed the frequency from 865.25MHz to 865.00MHz.
The connection to the inverter was restored almost immediately.

Perhaps the development guys here can add something to the software so that if the inverter freezes and the yield is good, the reconnection is tested at around +/-0.25MHz.

Best regards
from Berlin
lueralba

iomax, 2025-07-03:
> Auch das Thema Zeitabgleich ID2 ist bei den neuen Versionen viel häufiger zu lesen wie bei der alten FW.

Not the same inverter (HM-350 and HM-400 here), but regarding ID 2, I’ve verified in my case that all versions after 24.11.7 will raise many 'Time Calibration' errors/warnings. https://github.com/tbnobody/OpenDTU/issues/2505#issuecomment-2897846745