tbnobody/OpenDTU #2673: Bug mise à jour openDTU

BJC31600, 2025-05-12:
### What happened?

Bonjour,
Depuis la version 25.2.3, je n'arrive plus à faire la mise à jour des nouveaux firmware sur mon open DTU branché en ethernet? La mise à jour reste bloquée à 7%.
Que faire?
Merci d'avance pour votre réponse

### To Reproduce Bug

Pendant la mise à jour

### Expected Behavior

Comment faire cette mise à jour?

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v25.2.3

### What firmware variant (PIO Environment)?

generic_esp32

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-05-12:
> Hello,
Since version 25.2.3, I can no longer update the new firmware on my ethernet-connected open DTU? The update remains blocked at 7%.
What can I do?
Thank you in advance for your reply.

stefan123t, 2025-05-14:
@BJC31600 have you tried to flash the `factory` firmware binary via USB cable ?
It may be that the old version you are trying to upgrade requires such a fresh start.
You should backup your old config and use the json file to fill out the new config after the factory.
Bonne chance!

BJC31600, 2025-05-15:
Bonjour,
Je n'ai pas essayé de flasher un factory firmware car j'ai acheté mon boitier à une personne qui fabrique l'openDTU.

stefan123t, 2025-05-16:
Salut @BJC31600 you will have to either contact your vendor / manufacturer to help you on this or do it yourself.
When persuing this yourself you should [make a backup](https://www.opendtu.solar/firmware/configuration/config_settings/#backup-configuration-file-backup) of your current configuration files (pin_mapping.json, config.json) just in case you have to look up something later.
The [factory firmware installation](https://www.opendtu.solar/firmware/) process is described on the OpenDTU page and can be carried out with a USB cable connected between your PC and OpenDTU board.
You may want to join the Discord channel to get some hands-on advice, as we may not help you via a github issue as quickly.

closed 2025-06-30 by github-actions[bot]