tbnobody/OpenDTU #2677: Verwirrend: Loglevel "Ausführlich" steht hinter Loglevel "Debug"

Gitsarry, 2025-05-15:
### What happened?

v25.5.10

Herkömmlich (z.B. syslog) ist die Loglevel-Reihenfolge:
Fehler, Warnung, Info, Ausführlich(Engl. "verbose"), Debug

In der OpenDTU-Protokollierungseinstellung (zumindest im Auswahlmenü, siehe Bildschirmfoto) ist die Reihenfolge jedoch:
Fehler, Warnung, Info, Debug, Ausführlich

Das finde ich verwirrend, bzw. ich vermute einen Bug. "Debug" sollte doch immer die höchste Loglevel-Stufe sein?

### To Reproduce Bug

Gehe zu Einstellungen->Protokllierung->Öffne Menü "Standard-Protokollierungsstufe"

Angezeigt Reheinfolge im Menü:
Fehler, Warnung, Info, Debug, Ausführlich

[image]

### Expected Behavior

Reihenfolge: Fehler, Warnung, Info, Ausführlich(Engl. "verbose"), Debug

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v25.5.10

### What firmware variant (PIO Environment)?

generic

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-05-15:
> Das finde ich verwirrend, bzw. ich vermute einen Bug. "Debug" sollte doch immer die höchste Loglevel-Stufe sein?

Die Log-Level sind exakt so abgebildet wie es das ESP-IDF vorgibt. Ob es verwirrend oder nicht ist kann ich nicht beurteilen.

closed 2025-05-15 by tbnobody

Gitsarry, 2025-05-15:
"wie es das ESP-IDF vorgibt." Meinst Du [das](https://esphome.io/components/logger.html) hier?

Das ist wirklich kurios, habe ich in 20 Jahren Linux-Admin/Open Source-Zeit noch nie gesehen. Ich würde sagen beim CCC Hacker Jeopardy würde auch nahezu jeder auf die Frage ob Loglevel verbose vor oder nach debug kommt, sagen "davor" (also weniger geschwätzig).

Mit Trace kann ich mitgehen, jedoch wäre die richtige Übersetzung "Spurensuche" oder "Aufspüren" (trace=Spur), um zu verdeutlichen dass es wirklich noch feiner ist als Debuggen.

Man müsste einen Upstream Bug bei ESP aufmachen, aber die werden das wohl jetzt nicht mehr ändern wollen.

Also muss man mit diesem Kuriosum leben, Schwamm drüber.